john30/ebusd #374: "invalid configpath without scanconfig"

RhineDelta, 2020-11-11:
Installation:
Rasberry Pi 3B+/Buster with working MQTT broker (already in use for connecting Sonoff/Node Red/Domoticz)
Brink Allure B16 air heater and Brink Thermostat communicating over ebus

Tapping into ebus works fine:
[image]

Guess there are no configuration files for my heater. Am I correct in assuming that missing a configuration file will have any attempt to configure ebusd run into "invalid configpath without scanconfig"? So I don't seem to be able to configure the ebusd MQTT function.

For now I have a limited objective of simply listening in to the communication between Thermostat and airheater and get both the "set" and the actual temperature from the thermostat.

ebusd.log is filling up continuously but somehow I can't read the data because it is not UTF coded. Guess the raw data bytes are causing this.

Is there a way to circumvent the "invalid configpath without scanconfig" blockade and to get the raw data to MQTT so I can explore what's coming in?

RhineDelta, 2020-11-11:
This circuit is used to tap into the ebus:
[image]
 Circuit obtained from:
https://gitlab.com/fromeijn/ebuzzz-adapter/-/tree/master

RhineDelta, 2020-11-12:
With Filezilla ebusd.log to PC and opened in EXCELL. 
Reboot Raspberry gives this record:
[Ebusd log at restart.xlsx](https://github.com/john30/ebusd/files/5531461/Ebusd.log.at.restart.xlsx)

My guess:
"Econ"=air heater
"Kromschroeder"=thermostat

So it works, but how to circumvent scanconfig blocking MQTT configuration?
[image]

john30, 2020-11-16:
you need to configure the path to the config files using the ebusd "-c" parameter

RhineDelta, 2020-11-17:
Thanks for the hint!

Guess you mean like this:
[image]

This doesn't seem to work. Alas.

Please correct me if I am wrong:
- ebusd performs an initial scan to identify the type of equipment it is connect to and subsequently loads the applicable configuration file from the path specified.
- So if the equipment Brink Allure air heater (https://www.brinkclimatesystems.de/Warmluftheizung.html) is **not known**, what will ebusd do?
- Looking at the log file it seems any unknown command is logged. 

So if I use a rudimentary logfile **which at least will enable me to have the equipment recognized**, then I may be able:
1) to configure MQTT?
2) use the logfile to start constructing a more extensive configuration file by adding unrecognized commands from the logfile to the configuration file?

RhineDelta, 2020-11-20:
Note: I am fully aware that importing in Excell the data is wrongly interpreted as characters while the data actually has to be interpreted in line with the coding formats supplied in the ebus specification. Anyway the file only demonstrates the regular ebus communication is intercepted.

RhineDelta, 2020-11-20:
If my assumptions are correct you may have a rudimentary config file template?

RhineDelta, 2020-12-02:
Guess I may have to develop my own csv file.
The spec_prot_7_v1_6_3_d.pdf page 38: Abfrage der unterstutzten Befehle (07h 02h).

Could that be useful to identify which commands are in use?
How can I ebusd order to issue such command?

john30, 2021-01-10:
as you can see [here](https://github.com/john30/ebusd-configuration), we currently don't have any Kromschroeder config, so you'd have to create it yourself
using message 0702 will only reveal those that are specified in the eBUS spec, but not the vendor specific ones. But most of the configs are vendor specific

closed 2021-01-10 by john30