john30/ebusd #379: VWZ00

kodidu, 2020-12-13:
Hello!

Here is my setup:
`
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0308;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0308;HW=0403"
`

I can data from the hmu, 700, vr71.
But I could observer these cyclic errors:
`
2020-08-26 17:09:24.823 [update error] unable to parse scan-read scan.76 id from 3176b5090124 / 00: ERR: invalid position
2020-08-26 17:09:24.923 [update error] unable to parse scan-read scan.76 id from 3176b5090125 / 00: ERR: invalid position
2020-08-26 17:09:25.022 [update error] unable to parse scan-read scan.76 id from 3176b5090126 / 00: ERR: invalid position
2020-08-26 17:09:25.122 [update error] unable to parse scan-read scan.76 id from 3176b5090127 / 00: ERR: invalid position
2020-08-26 17:09:25.122 [main error] scan config 76: ERR: invalid position
`
VWZ00 seems to be the freshwater module (Aguaflow).
Other have this issue too:
https://forum.fhem.de/index.php/topic,84636.msg1081006.html#msg1081006
https://github.com/john30/ebusd/issues?q=is%3Aissue+VWZ00+is%3Aclosed

It is correct that there is no VWZ00 csv config.
It looks like ebusd fails parsing the answer and keeps scanning/asking the slave.
I think it is no config error and needs to be fixed in ebusd.
Which logs will you need to parse the answer of the module VWZ00?
So I can try to decode its messages.

Otherwise a Retry only x times to parse the answer and then stop will also work.
But it wont be possible to apply a csv for the VWZ00, because it fails to parse the config.
Or is it possible to blacklist some ebus addresses/ids?

kodidu, 2020-12-14:
With additional debug output here:
https://github.com/john30/ebusd/blob/3926e9cbf81457cd8cbb3b5985aae6445560d8e2/src/lib/ebus/data.cpp#L529

In error case:
offset = 0; remainder = 0; m_length = 1;  data.getDataSize()=0
=> 1 > 0 => RESULT_ERR_INVALID_POS
`  
if (offset + (remainder?1:m_length) > data.getDataSize()) {
    return RESULT_ERR_INVALID_POS;
  }
`

john30, 2021-01-10:
the result of the product ID request is just too short, so the device does not reveal it.
it is possible to provide a CSV for the device anyway, as the product info is only used if available. just use a name built from the address and identifier alone, i.e. "76.vwz.csv".

closed 2021-01-10 by john30