john30/ebusd #380: Mosquitto version 2

mozzarelka, 2020-12-14:
Hi, 
after updating mosquitto version from 1.6 to 2.0 on ubuntu server I lost mqtt connection with ebusd.

[mqtt error] invalid mosquitto version 2 instead of 1
[main error] error registering data handlers

Rafal

john30, 2021-01-24:
can you try commenting lines 302-305 in src/ebusd/mqtthandler.cpp and see if this works?

kyle-mge, 2021-03-12:
> can you try commenting lines 302-305 in src/ebusd/mqtthandler.cpp and see if this works?

Hi @john30,

had the same issue and tried your suggestion of commenting the mentioned lines. Without checking the major version it works. mqtt messages are flying in.

Thanks
Matthias.

closed 2021-04-05 by john30

dreamdancer27, 2022-03-10:
had the same issue today, everything is up to date, anny suggestions for debian packages?

john30, 2022-03-12:
try upgrading libmosquitto

commit db524b9d (2021-04-05): expect newer major versions to work as well (fixes #380)