john30/ebusd #382: Possible wrong data type in broadcast.csv

WildRat, 2020-12-20:
In **broadcast.csv** data type for "**date**" field is **BDA**:

`b,,vdatetime,date/time,,,,"00",time,,BTI,,,,date,,BDA,,,,,,,,,`

**BDA** data type is "_day first, including weekday, **Sunday=0x06**_". But in my system, and in others log files I see on internet data type of this field is more likely "_day first, including weekday, **Sunday=0x07**_".

Please add data type "_day first, including weekday, **Sunday=0x07**_".

john30, 2021-01-10:
sure this isn't "HDA" as described [here](https://github.com/john30/ebusd/wiki/4.3.-Builtin-data-types)?

WildRat, 2021-01-10:
No, I checked its BCD-encoded date with weekday, but weekday starts from 1:
Here is recent example 10.01.2020 14:04:
2021-01-10 12:04:07.391 [bus notice] <10feb51608000804141001072103

commit 3e81e4af (2021-02-20): corrected BDA type according to eBUS spec (fixes #382), added BDZ
closed 2021-02-20 by john30