john30/ebusd #383: No UDP connection / No Signal after update to 3.4

kjumybit, 2020-12-28:
Hi,
I just upgraded from release 3.1 to release 3.4. I build and deployed the new version on an Ubuntu 18.04 LTS system with MQTT support. But the new version of the ebusd didn't get an **UDP network connection** to my local Ebus Ethernet adapter. There is also no MQTT connection established.

I did't see neither any error message nor a try to establish a UDP network connection.

ebusd options (without MQTT):

`--configpath=/etc/ebusd --scanconfig -d udp:192.168.2.13:5000 --initsend --latency=20000 --nodevicecheck --updatecheck=off -f --loglevel=debug
`
Logfile with debug level:
`2020-12-28 16:42:56.482 [main notice] ebusd 3.4.v3.4-20-gedfe09a started with auto scan
2020-12-28 16:42:56.482 [main info] loading configuration files from /etc/ebusd
2020-12-28 16:42:56.482 [main info] reading templates /
2020-12-28 16:42:56.483 [main info] read templates in /
2020-12-28 16:42:56.483 [main info] reading file broadcast.csv
2020-12-28 16:42:56.484 [main info] successfully read file broadcast.csv
2020-12-28 16:42:56.484 [main info] reading file memory.csv
2020-12-28 16:42:56.484 [main info] successfully read file memory.csv
2020-12-28 16:42:56.484 [main info] read config files
2020-12-28 16:42:56.485 [bus notice] bus started with own address 31/36
2020-12-28 16:42:56.485 [main info] registering data handlers
2020-12-28 16:42:56.485 [main info] registered data handlers
2020-12-28 16:43:06.486 [main debug] performing regular tasks
2020-12-28 16:43:16.486 [main debug] performing regular tasks
2020-12-28 16:43:26.487 [main debug] performing regular tasks
2020-12-28 16:43:36.487 [main debug] performing regular tasks
`

Output of ebusctl:
`localhost: info
version: ebusd 3.4.v3.4-20-gedfe09a
signal: no signal
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
`

john30, 2021-01-10:
check with netcat if you can establish a UDP connection at all

kjumybit, 2021-01-10:
Thanks, for the advice. I observe, the my older ebusd version get connected, the current version not.

commit 2f3f0422 (2021-01-10): fix for UDP connection introduced 20200411 with fix for #336 (fixes #383
closed 2021-01-10 by john30