john30/ebusd #384: Read-only adapter: csv not loaded and fields are not accessible

frahemmi, 2021-01-04:
Hello, 
first of all I’d like to thank all of you for the great work on this topic. It looks so promising and makes the eBus available to many people. 

I guess I need some help, since I am trying hard to get ebusd working since some day. It seems that I turn in circles and I am not even sure if it is possible at al what I am trying to achieve. 

My problem: 
I have a read only adapter (based on an Arduino) connected to a raspberry to read the ebus of a Vaillant heating system. 
I did all the necessary steps:

- ebusd installed
- I checked that the get the `<aa` when logging raw data as bytes
- I installed the config files locally 
- When logging I can see that the csv files are loaded successfully
- I stripped the config files down to the once I need (since I use a read only adapter I can’t use `–scanconfig`)
- I restarted the heating system while ebusd was running to force the system to send the required information about the slaves (required due to the read only adapter)
- I get a reasonable `ebusctl info` result
- When I use `ebusd find` I get a reasonable output

The problems I am facing are:

- I can’t the that csv files are loaded for each slave (maybe due to the read only adapter and no `–scanconfig`)
- When I try to read a filed with `ebusctl read –c hmu EnergyHc` I get a `ERR: arbitration lost`
- On other fields I get `ERR: element not found`

I am not sure if ebusd works at all with a read only adapter. Is there something I forgot?
Are people around which use ebusd in combination with a read only adapter?

It would be great if I can get some help. 
Frank

Ps: Here are some of the results/log files. This might help to see what’s wrong here at my end.

```
ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available, vaillant/15.700.csv: different version available, vaillant/hcmode.inc: different version available
signal: acquired
symbol rate: 23
max symbol rate: 125
reconnects: 0
masters: 5
messages: 704
conditional: 64
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0307;HW=0403"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"
address 10: master #2
address 12: slave
address 15: slave #2
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0307;HW=0403"
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0131;HW=5202"
address f7: master #20
```

Some rows of the log file:
```
2021-01-04 20:11:14.394 [main notice] ebusd 3.4.v3.3-51-g57eae05 started
2021-01-04 20:11:14.478 [bus notice] bus started with own address 31/36
2021-01-04 20:11:14.506 [bus notice] signal acquired
2021-01-04 20:11:21.863 [bus notice] new master 10, master count 2
2021-01-04 20:11:21.921 [bus notice] new master 03, master count 3
2021-01-04 20:11:21.922 [update notice] received unknown MS cmd: 1008b5110101 / 096c634003ffff0100ff
2021-01-04 20:11:22.188 [bus notice] new master 71, master count 4
2021-01-04 20:11:22.188 [update notice] received unknown MS cmd: 1076b5110101 / 09ff634003ffff0000ff
[... 163 lines cut ...]
2021-01-04 21:12:14.965 [bus notice] <1026b5230107d1000f78030080008000800000800080fc058900
2021-01-04 21:12:15.234 [bus notice] <1008b51009000000ffffff070000520001019a00
2021-01-04 21:12:15.501 [bus notice] <1076b51009000000ffffff050000dd0001019a00
2021-01-04 21:12:15.787 [bus notice] <1026b5230f05ff0000000000ffff0000000000001b0001019a00
2021-01-04 21:12:16.045 [bus notice] <1026b52304020001495a00020100b700
2021-01-04 21:12:23.880 [bus notice] <1008b511010189000961574003ffff0000ff8c00
2021-01-04 21:12:24.148 [bus notice] <1076b5110101160009ff574003ffff0000ffb200
2021-01-04 21:12:24.414 [bus notice] <1076b512030f0001fc0007008000008012058900
2021-01-04 21:12:24.711 [bus notice] <1026b5230106d00010c50246020080008027038e0297035000e400
2021-01-04 21:12:25.003 [bus notice] <1026b5230107d1000f78030080008000800000800080fc058900
2021-01-04 21:12:25.269 [bus notice] <1008b51009000000ffffff070000520001019a00
2021-01-04 21:12:25.537 [bus notice] <1076b51009000000ffffff050000dd0001019a00
2021-01-04 21:12:25.824 [bus notice] <1026b5230f05ff0000000000ffff0000000000001b0001019a00
2021-01-04 21:12:26.069 [bus notice] <1026b52304020001495a00020100b700
2021-01-04 21:12:27.013 [bus notice] <03e0b521090008221ee4ffff460028000b0002f9ff110007000008002700
2021-01-04 21:12:33.926 [bus notice] <1008b511010189000961574003ffff0000ff8c00
2021-01-04 21:12:34.193 [bus notice] <1076b5110101160009ff574003ffff0000ffb200
2021-01-04 21:12:34.460 [bus notice] <1076b512030f0001fc0007008000008012058900
2021-01-04 21:12:34.758 [bus notice] <1026b5230106d00010c40246020080008025038d0297035000a300
2021-01-04 21:12:35.050 [bus notice] <1026b5230107d1000f78030080008000800000800080fc058900
2021-01-04 21:12:35.316 [bus notice] <1008b51009000000ffffff070000520001019a00
2021-01-04 21:12:35.583 [bus notice] <1076b51009000000ffffff050000dd0001019a00
2021-01-04 21:12:35.870 [bus notice] <1026b5230f05ff0000000000ffff0000000000001b0001019a00
2021-01-04 21:12:36.116 [bus notice] <1026b52304020001495a00020100b700
```

mwildbolz, 2021-01-04:
Hi!
I'm also working with an Vaillant heating system (VWF117/4 and VRC700) and an ESERA USB-eBus-Adapter.
As far is I understand the topology of the eBus and how the Vaillant heating system works, it is **not possible to get most of the data without write-access to the bus.**
Many data-points have to be polled (like hmu EnergyHc) and are not transmitted over the bus automatically.
Just a few values are transmitted regularly (e.g. OutsideTemperature broadcast, VR_71 sensor values) but most of them have to be polled...

So, in my opinion, write-access to the bus is a MUST...

BR

frahemmi, 2021-01-07:
Hi @mwildbolz ,
thanks for sharing your experience with me. 
Based on this I had a deeper look and what you wrote make sense to me know. I can see some values in OutsideTemp and VR_71 SensorData1 as well. When I use `ebusctl read -c  vr_71 SensorData1` get these values as result. 
I guess this is because, as you have mentioned as well, they are broadcasted and ebusd stores them in the cache. 
All the once which are reported with "No data stored" by `ebusctl find` do not have a value stored in the ebusd cache, since they are not broadcasted, and an active "pull" is required and this requires read and write access to the ebus.
This makes at least some sense to me :-) 

@mwildbolz You mentioned that you are working with an ESERA Adapter. Can I ask you how you "consume" the ebus values? I guess you have a script in place that reads the values, but how often do you do that? Do you have any delays defined between the ebus calls? 

Regards
Frank

mwildbolz, 2021-01-07:
> @mwildbolz You mentioned that you are working with an ESERA Adapter. Can I ask you how you "consume" the ebus values? I guess you have a script in place that reads the values, but how often do you do that? Do you have any delays defined between the ebus calls?

I work with openHAB (and ebus-Binding) and have polling intervals - depending on the subsystem (water, hc, energy values, etc.) - of 1 Minute (e.g. most of the temperatures) up to 12h (e.g. for energy values).

john30, 2021-01-10:
as stated above: a read-only adapter will not be able to reveal more than what is sent on the bus by other devices and in addition to that ebusd does not really have a chance to find the best fitting CSVs. If you know which CSVs are right, you can still not use the scanconfig feature and instead put the matching files only to a local folder and point ebusd to that one

frahemmi, 2021-01-12:
Hi @john30 , hi @mwildbolz.
Thanks for your quick answers. I tried it with a read and write adapter and it all works perfectly fine. It is maybe worth to mention, as @mwildbolz already did, that at least with Vaillant systems a read-only adapter does not make much sense. Other people can save a good amount of time if they know that right from the start :-).
Thanks for your help and even more for the great work you all did on this topic. It works so well and makes a lot of fun to use it :-)

Frank

closed 2021-01-12 by frahemmi