john30/ebusd #385: Saunier Duval as Vaillant - To many write polls

Dexx-electronic, 2021-01-05:
Hi 
I'm trying to connect and read data from my Saunier Duval GeniaAir Split. It is a rebranded Vaillant Heat Pump.
I build adapter according Prof. Dr. Peter A. Henning schematic with some minor changes to get rid of poti but after all it works.
But now i have problem because ebusd is blocking the bus with write-polls and in end effect that caused error in my Heat pump
(communication error)

**Is there a possibility to add a interval that ebusd must wait before it will send next poll write ?**
Or how should I omit this problematic Variables 

When i read info i got:

```
sudo ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
signal: acquired
symbol rate: 88
max symbol rate: 156
min arbitration micros: 2104
max arbitration micros: 3374
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 4
messages: 75
conditional: 0
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0405;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=E7000;SW=0607;HW=7003"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0517;HW=5103"
```
But i see that some values are incorrect and because of that anther poll request is sanded

for example like here:

```
2021-01-05 22:32:20.475 [update error] unable to parse read hmu errorhistory from 7108b50303010105 / 0a010000000000ffffffff: ERR: argument value out of valid range
2021-01-05 22:32:20.754 [update notice] received unknown MS cmd: 7108b51a0405183200 / 0a18082f14000000000000
2021-01-05 22:32:21.043 [update notice] received unknown MS cmd: 7108b5120e110202fe019501ea020eb3070000 / 020632
2021-01-05 22:32:21.324 [update notice] received unknown MS cmd: 7108b509022802 / 0c020004055f0100484d553031
2021-01-05 22:32:21.604 [update notice] received unknown MS cmd: 7108b509022802 / 0c020004055f0100484d553031
2021-01-05 22:32:21.868 [update notice] received unknown MS cmd: 7108b5110100 / 0902020e000008010000
2021-01-05 22:32:22.133 [update notice] received read hmu Status01 QQ=71: 32.0;31.5;-;-;-;off
2021-01-05 22:32:22.402 [update notice] received read hmu State QQ=71: 0;20;on;2
2021-01-05 22:32:22.682 [update notice] received unknown MS cmd: 7108b509022802 / 0c020004055f0100484d553031
2021-01-05 22:32:23.701 [update notice] received read hmu currenterror QQ=71: 820;-;-;-;-
2021-01-05 22:32:23.970 [update notice] received read hmu State QQ=71: 0;20;on;2
2021-01-05 22:32:24.497 [update notice] received unknown MS cmd: 7108b5110100 / 0902020e000008010000
2021-01-05 22:32:24.982 [update notice] received read hmu Status01 QQ=71: 32.0;31.5;-;-;-;off
2021-01-05 22:32:25.535 [mqtt error] decode hmu errorhistory: ERR: argument value out of valid range
2021-01-05 22:32:25.700 [update notice] received unknown MS cmd: 7108b51a0405193201 / 0a19082f00000000000000
2021-01-05 22:32:26.800 [update notice] received read hmu Status01 QQ=10: 32.0;31.5;-;-;-;off
2021-01-05 22:32:26.980 [update notice] received read hmu State QQ=71: 0;20;on;2
2021-01-05 22:32:27.112 [update notice] received unknown MS cmd: 1076b5110101 / 093f3f6801ff4b0000ff
2021-01-05 22:32:27.376 [update notice] received unknown MS cmd: 1076b512030f0001 / 075f0200fd010eff
2021-01-05 22:32:27.509 [update notice] received unknown MS cmd: 7108b5110100 / 0902020e000008010000
2021-01-05 22:32:27.722 [update notice] received read vr_71 SensorData1 QQ=10: 31.44;32.19;32.06;-;-;-;-;40 3d
2021-01-05 22:32:28.019 [update notice] received read vr_71 SensorData2 QQ=10: -;-;-;-;0.00;8.00;80 54 05
2021-01-05 22:32:28.152 [update notice] received read hmu Status01 QQ=71: 32.0;31.5;-;-;-;off
2021-01-05 22:32:28.328 [update notice] received update-write hmu SetMode QQ=10: off;0.0;-;-;0;1;1;1;0;0
2021-01-05 22:32:28.592 [update notice] received unknown MS cmd: 1076b51009000032ffffff040000 / 0101
2021-01-05 22:32:28.733 [update notice] received read hmu currenterror QQ=71: 820;-;-;-;-
2021-01-05 22:32:28.937 [update notice] received write vr_71 SetActorState QQ=10: -;-;off;off;off;off;-;-;-;-;off;off;00 00
2021-01-05 22:32:29.093 [update notice] received unknown MS cmd: 7108b5120e110202fe019601e6020e87070000 / 020632
[... 80 lines cut ...]
2021-01-05 22:32:57.008 [update notice] received read hmu Status01 QQ=10: 32.0;31.5;-;-;-;off
2021-01-05 22:32:57.190 [update notice] received read hmu State QQ=71: 0;20;on;2
2021-01-05 22:32:57.321 [update notice] received unknown MS cmd: 1076b5110101 / 093f3f6801ff4b0000ff
2021-01-05 22:32:57.585 [update notice] received unknown MS cmd: 1076b512030f0001 / 075f0200fb010eff
2021-01-05 22:32:57.718 [update notice] received unknown MS cmd: 7108b5110100 / 0901020e000008010000
2021-01-05 22:32:57.932 [update notice] received read vr_71 SensorData1 QQ=10: 31.06;31.69;31.62;-;-;-;-;40 3d
2021-01-05 22:32:58.229 [update notice] received read vr_71 SensorData2 QQ=10: -;-;-;-;0.00;8.00;80 54 05
2021-01-05 22:32:58.494 [update notice] received update-write hmu SetMode QQ=10: off;0.0;-;-;0;1;1;1;0;0
2021-01-05 22:32:58.758 [update notice] received unknown MS cmd: 1076b51009000032ffffff040000 / 0101
2021-01-05 22:32:58.846 [update notice] received read hmu Status01 QQ=71: 32.0;31.5;-;-;-;off
2021-01-05 22:32:59.094 [update notice] received write vr_71 SetActorState QQ=10: -;-;off;off;off;off;-;-;-;-;off;off;00 00
2021-01-05 22:32:59.190 [update notice] received read hmu currenterror QQ=71: 820;-;-;-;-
2021-01-05 22:32:59.397 [update notice] received write vr_71 Mc1FlowTempDesired QQ=10: on;40.0;on;65
2021-01-05 22:32:59.646 [update notice] received write vr_71 Mc2FlowTempDesired QQ=10: on;37.5;on;45
2021-01-05 22:33:00.222 [update notice] received read hmu State QQ=71: 0;20;on;2
2021-01-05 22:33:00.705 [update notice] received unknown MS cmd: 7108b5110100 / 0901020e000008010000
^C2021-01-05 22:33:00.775 [main notice] SIGINT received
2021-01-05 22:33:00.994 [update notice] received unknown MS cmd: 7108b5120e110002fd019d01d7020e39070000 / 020632
2021-01-05 22:33:01.259 [update notice] received read hmu Status01 QQ=71: 32.0;31.5;-;-;-;off
2021-01-05 22:33:01.540 [update notice] received unknown MS cmd: 7108b51a040520323d / 0a20082c5c000000000000
2021-01-05 22:33:03.257 [update notice] received read hmu State QQ=71: 0;20;on;2
2021-01-05 22:33:03.740 [update notice] received unknown MS cmd: 7108b5110100 / 0900020e000008010000
2021-01-05 22:33:04.058 [update notice] received read hmu currenterror QQ=71: 820;-;-;-;-
2021-01-05 22:33:04.322 [update notice] received read hmu Status01 QQ=71: 32.0;31.5;-;-;-;off
```

john30, 2021-01-10:
see the ebusd command line arguments [here](https://github.com/john30/ebusd/wiki/2.-Run), especially "--pollinterval", "--sendretries", and "--acquireretries"

Dexx-electronic, 2021-01-20:
I've tested a lot in the meantime, but it seems that the main cause of the problem was a defective transistor.

closed 2021-01-20 by Dexx-electronic

aldert-b, 2021-12-08:
Hi Dexx-electronic,

I have a Saunier Duval GeniaAir Split as well and would like to read/write from it as well to, for example, warm up water when the solar panels produce more electricity than needed in the house. Do you think this is possible? 
Thanks for a reply.
Regards,
Aldert

Dexx-electronic, 2021-12-21:
Hi, I think that it is possible to set up this unit as you want but I don't use currently any active writes. I'm using only "build in" write command to read parameters. 
The biggest problem is that Saunier Duval commands are not fully recognized, and we can only base on this commands that are common with Vailant.