john30/ebusd #386: Unable to install with docker on alpine linux

Melkor82, 2021-01-11:
I am using ebusd on a homeassistant  machine that is running alpine 3.12.
I always used this command `docker run --name ebusd -p 8888 --mount type=bind,source=/mnt/data/supervisor/homeassistant/ebusd,target=/var/ebusd/ \
--restart unless-stopped  john30/ebusd:latest -f --scanconfig -d tcp:192.168.1.84:9999 --latency=80000 \
--mqtthost=192.168.1.99 --mqttport=1883 --mqttuser=melkor --mqttpass=332490Ms -c /var/ebusd/ -l /var/ebusd/ebusd.log`  but now i get two errors, glibc version 2.29 and glibcxx version 3.4.26 needed. 
For now i am sticking to the ebusd:v3.4 image tag, was wondering if there was the possibility to have a versione for alpine

mozzarelka, 2021-01-11:
same issue here after update to latest docker build, on ubuntu 20.04
ebusd: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by ebusd)
ebusd: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by ebusd)

john30, 2021-01-11:
can you try again please (docker pull ...)?

Melkor82, 2021-01-11:
Worked perfectly, thank you so much, you can set this issue as solved,
thanks!

Melkor82, 2021-01-11:
Out of topic, i am using it for a Brink Flair 325, if you want i can
provide you my modified csv file

mozzarelka, 2021-01-12:
Thank you john30, working fine now.

closed 2021-01-12 by john30