john30/ebusd #387: ebusctl state (info, find..) show no output whatsoever

mrowka912, 2021-01-13:
I use ebus HW 1.6, installed ebusd on Rpi4, did potentiometer calibration and I'm receiving some data after ebusd -f (also in daemon with mqtt)

`2021-01-13 10:25:03.550 [main notice] ebusd 3.4.v3.4-20-gedfe09a started with auto scan
2021-01-13 10:25:04.057 [bus notice] bus started with own address 31/36
2021-01-13 10:25:04.065 [mqtt notice] connection established
2021-01-13 10:25:04.080 [bus notice] signal acquired
2021-01-13 10:25:05.781 [bus notice] new master 10, master count 2
2021-01-13 10:25:05.849 [bus notice] new master 03, master count 3
2021-01-13 10:25:05.849 [update notice] received unknown MS cmd: 1008b5110101 / 0954520001ff6e0000ff
2021-01-13 10:25:07.812 [update notice] received unknown BC cmd: 10feb516080006241013010321
2021-01-13 10:25:10.867 [update notice] received unknown MS cmd: 1008b5100900006e64ffff04ff00 / 0101
2021-01-13 10:25:14.204 [bus notice] scan 08: ;Vaillant;BAI00;0407;7401
2021-01-13 10:25:14.204 [update notice] store 08 ident: done
2021-01-13 10:25:14.205 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0407;7401
2021-01-13 10:25:14.205 [bus notice] scan 08: ;Vaillant;BAI00;0407;7401
2021-01-13 10:25:14.734 [main notice] read common config file vaillant/scan.csv
2021-01-13 10:25:14.858 [main notice] read common config file vaillant/general.csv
2021-01-13 10:25:14.993 [main notice] read common config file vaillant/broadcast.csv
2021-01-13 10:25:15.346 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0407, HW7401
2021-01-13 10:25:15.511 [update notice] sent scan-read scan.08 id QQ=31:
2021-01-13 10:25:15.694 [update notice] sent scan-read scan.08 id QQ=31:
2021-01-13 10:25:15.879 [update notice] sent scan-read scan.08 id QQ=31:
2021-01-13 10:25:16.122 [update notice] sent scan-read scan.08 id QQ=31: ;;;;;;
2021-01-13 10:25:16.592 [main notice] found messages: 229 (0 conditional on 23 conditions, 0 poll, 9 update)
2021-01-13 10:25:16.757 [update notice] sent scan-read scan.08 id QQ=31: ;;;;;;
2021-01-13 10:25:16.942 [update notice] sent scan-read scan.08 id QQ=31: ;;;;;;
2021-01-13 10:25:17.128 [update notice] sent scan-read scan.08 id QQ=31: ;;;;;;
2021-01-13 10:25:17.311 [update notice] sent scan-read scan.08 id QQ=31: ;;;;;;
2021-01-13 10:25:17.312 [bus notice] scan 08: ;;;;;;;
2021-01-13 10:25:17.961 [update notice] received read bai DateTime QQ=10: nosignal;00:00:00;-.-.-;1.000
2021-01-13 10:25:18.230 [update notice] received read bai Status02 QQ=10: auto;60;55.0;80;50.0
2021-01-13 10:25:19.442 [bus notice] scan 15: ;Vaillant;40000;0139;7301
2021-01-13 10:25:19.443 [update notice] store 15 ident: done
2021-01-13 10:25:19.444 [update notice] sent scan-read scan.15  QQ=31: Vaillant;40000;0139;7301
2021-01-13 10:25:19.444 [bus notice] scan 15: ;Vaillant;40000;0139;7301
2021-01-13 10:25:19.647 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231303632363330
2021-01-13 10:25:19.845 [update notice] sent scan-read scan.15 id QQ=31:
2021-01-13 10:25:20.045 [update notice] sent scan-read scan.15 id QQ=31:
2021-01-13 10:25:20.247 [update notice] sent scan-read scan.15 id QQ=31: 21;06;26;307420<<<<;0008;005037;N6
2021-01-13 10:25:20.247 [bus notice] scan 15: ;21;06;26;307420<<<<;0008;005037;N6
2021-01-13 10:25:20.530 [main notice] read scan config file vaillant/15.400.csv for ID "40000", SW0139, HW7301
2021-01-13 10:25:20.531 [main notice] found messages: 429 (4 conditional on 26 conditions, 0 poll, 9 update)
2021-01-13 10:25:20.948 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:25:25.979 [update notice] received read bai Status01 QQ=10: 41.0;41.0;1.000;-;55.0;off
2021-01-13 10:25:27.875 [update notice] received update-read broadcast outsidetemp QQ=10: -1.000
2021-01-13 10:25:31.008 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:25:35.950 [update notice] received read bai Status01 QQ=10: 41.0;41.0;1.000;-;55.0;off
2021-01-13 10:25:37.944 [update notice] received read bai Status02 QQ=10: auto;60;55.0;80;50.0
2021-01-13 10:25:41.076 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:25:46.114 [update notice] received read bai Status01 QQ=10: 41.0;40.0;1.000;-;55.0;off
2021-01-13 10:25:48.073 [update notice] received read bai DateTime QQ=10: nosignal;00:00:00;-.-.-;1.000
2021-01-13 10:25:51.070 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:25:56.051 [update notice] received read bai Status01 QQ=10: 41.0;40.0;1.000;-;55.0;off
2021-01-13 10:26:01.131 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:26:06.073 [update notice] received read bai Status01 QQ=10: 41.0;40.0;1.000;-;55.0;off
2021-01-13 10:26:08.084 [update notice] received update-read broadcast vdatetime QQ=10: 10:25:06;13.01.2021
2021-01-13 10:26:11.177 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:26:16.170 [update notice] received read bai Status01 QQ=10: 40.0;40.0;1.000;-;55.0;off
2021-01-13 10:26:18.178 [update notice] received read bai DateTime QQ=10: nosignal;00:00:00;-.-.-;1.000
2021-01-13 10:26:18.442 [update notice] received read bai Status02 QQ=10: auto;60;55.0;80;50.0
2021-01-13 10:26:21.169 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:26:26.161 [update notice] received read bai Status01 QQ=10: 40.0;40.0;1.000;-;55.0;off
2021-01-13 10:26:28.103 [update notice] received update-read broadcast outsidetemp QQ=10: -1.000
2021-01-13 10:26:31.194 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:26:36.181 [update notice] received read bai Status01 QQ=10: 40.0;39.0;1.000;-;55.0;off
2021-01-13 10:26:38.129 [update notice] received read bai Status02 QQ=10: auto;60;55.0;80;50.0
2021-01-13 10:26:41.269 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:26:46.258 [update notice] received read bai Status01 QQ=10: 40.0;39.0;1.000;-;55.0;off
2021-01-13 10:26:48.269 [update notice] received read bai DateTime QQ=10: nosignal;00:00:00;-.-.-;1.000
2021-01-13 10:26:51.270 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
[... 41 lines cut ...]
2021-01-13 10:29:19.032 [update notice] received read bai Status02 QQ=10: auto;60;55.0;80;50.0
2021-01-13 10:29:21.766 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:29:26.752 [update notice] received read bai Status01 QQ=10: 40.0;39.0;1.000;-;55.0;off
2021-01-13 10:29:28.786 [update notice] received update-read broadcast outsidetemp QQ=10: -1.000
2021-01-13 10:29:32.011 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:29:37.365 [update notice] received read bai Status01 QQ=10: 40.0;39.0;1.000;-;55.0;off
2021-01-13 10:29:38.855 [update notice] received read bai Status02 QQ=10: auto;60;55.0;80;50.0
2021-01-13 10:29:41.891 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:29:46.838 [update notice] received read bai Status01 QQ=10: 40.0;39.0;1.000;-;55.0;off
2021-01-13 10:29:48.879 [update notice] received read bai DateTime QQ=10: nosignal;00:00:00;-.-.-;1.000
2021-01-13 10:29:51.873 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:29:56.996 [update notice] received read bai Status01 QQ=10: 40.0;39.0;1.000;-;55.0;off
2021-01-13 10:30:01.998 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:30:06.981 [update notice] received read bai Status01 QQ=10: 40.0;39.0;1.000;-;55.0;off
2021-01-13 10:30:08.950 [update notice] received update-read broadcast vdatetime QQ=10: 10:29:07;13.01.2021
2021-01-13 10:30:12.042 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:30:16.984 [update notice] received read bai Status01 QQ=10: 40.0;39.0;1.000;-;55.0;off
2021-01-13 10:30:19.023 [update notice] received read bai DateTime QQ=10: nosignal;00:00:00;-.-.-;1.000
2021-01-13 10:30:19.293 [update notice] received read bai Status02 QQ=10: auto;60;55.0;80;50.0
2021-01-13 10:30:21.982 [update notice] received update-write bai SetMode QQ=10: auto;55.0;50.0;-;0;0;1;0;0;0
2021-01-13 10:30:26.962 [update notice] received read bai Status01 QQ=10: 40.0;39.0;1.000;-;55.0;off
2021-01-13 10:30:28.990 [update notice] received update-read broadcast outsidetemp QQ=10: -1.000
`

but when I use command 'ebusctl info' it does not produce any output? What am I missing?

john30, 2021-01-24:
which version of ebusd are you running?

mrowka912, 2021-01-24:
it is
```ebusd --version
ebusd 21.1.v21.1-12-gccfc025
```
by the way, all mqtt commands and queries work fine

mrowka912, 2021-02-11:
'ebusd 21.2.v21.2'
same issue

john30, 2021-02-20:
do you have selinux enabled or is a firewall active?

mrowka912, 2021-02-20:
neither of those are installed
```
root@raspberrypi:/home/pi# sudo ufw status
sudo: ufw: command not found
root@raspberrypi:/home/pi# getenforce
bash: getenforce: command not found
```

only login to SSH is with RSA key.

john30, 2021-02-20:
weird. please provide some minutes of verbose logging (start with `--log all:debug'` with the newest code. make sure this log covers an attempt to issue a read command through ebusctl

mrowka912, 2021-02-20:
did 
`sudo ebusd -d /dev/ttyUSB0 -p 8888 --log=all:debug --logfile=/var/log/ebusd_log.log --httpport=8889 --accesslevel=* --scanconfig=full --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/en/
`

attached logfile from this, [ebusd_log.log](https://github.com/john30/ebusd/files/6015678/ebusd_log.log)

during and at the end called couple of times breaking with ctrl+c

```pi@raspberrypi:~ $ ebusctl info
^C
pi@raspberrypi:~ $ ebusctl info
^C
pi@raspberrypi:~ $ ebusctl info
^C
pi@raspberrypi:~ $ ebusctl info
^C
```

john30, 2021-02-20:
are you sure there is no other daemon listening on port 8888?
please check the output of `sudo lsof -n |grep :8888`

mrowka912, 2021-02-20:
of course it was ! 
```
pi@raspberrypi:~ $ sudo lsof -n |grep :8888
pigpiod     402                             root    0u     IPv6      14890       0t0        TCP *:8888 (LISTEN)
pigpiod     402   413 pigpiod               root    0u     IPv6      14890       0t0        TCP *:8888 (LISTEN)
pigpiod     402   414 pigpiod               root    0u     IPv6      14890       0t0        TCP *:8888 (LISTEN)
pigpiod     402   435 pigpiod               root    0u     IPv6      14890       0t0        TCP *:8888 (LISTEN)
```

now, all works fine! THANK YOU!
```
pi@raspberrypi:~ $ ebusctl -p 8899 info
version: ebusd 21.2.v21.2
access: *
signal: acquired
symbol rate: 29
max symbol rate: 41
min arbitration micros: 1063
max arbitration micros: 2717
min symbol latency: 5
max symbol latency: 14
reconnects: 0
masters: 3
messages: 12
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0407;HW=7401"
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

closed 2021-02-20 by mrowka912