john30/ebusd #389: Docker docker: Error response from daemon: OCI runtime create failed: container_linux.go:370:

vancako, 2021-01-16:
Hello John,
I am trying to migrate from working ebusd configuration to Docker.
I am running on PI4 and docker image was build from your config for pi. (name image is ebus)
There is my command line for launch Docker

docker run -d --name=ebusd --device=/dev/ttyUSB0 -v /opt/ebusd/conf/ebusd:/etc/ebusd -p 8888 ebus --scanconfig --device=/dev/ttyUSB0 --configpath=/etc/ebusd --mqttport=1883 --mqtthost=172.17.0.3
9bdc94fe1e31a1ebc931734162af06562880212f996a907aa31a4adc68b9fd4b

docker: Error response from daemon: OCI runtime create failed: container_linux.go:370: starting container process caused: exec: "--scanconfig": executable file not found in $PATH: unknown

Have you any idea what should be wrong?
Thanks Ondrej

john30, 2021-01-17:
please try with `docker pull john30/ebusd:devel`, as this is now multi-arch.

vancako, 2021-01-17:
Hi John, I can confirm that you proposed solution working.
Btw. I was rebuilt pi4 image under "john30/ebusd" and this working too.
Thanks for support and your perfect job!!!!!!

closed 2021-01-17 by vancako

stevoh6, 2021-01-17:
> Hi John, I can confirm that you proposed solution working.
> Btw. I was rebuilt pi4 image under "john30/ebusd" and this working too.
> Thanks for support and your perfect job!!!!!!

Hi, could you please share your Dockerfile? Thx

john30, 2021-01-18:
it is in the repo: https://github.com/john30/ebusd/tree/master/contrib/docker