john30/ebusd #390: Docker john30/ebusd:rpi-latest cannot start - `GLIBC_2.28' not found

stevoh6, 2021-01-16:
HI, I was trying to run latest ebusd on "NanoPi neo plus2" with  "Ubuntu 16.04.6 LTS", means "armhf" architecture. I got following error. 
> root@NanoPi-NEO-Plus2:/var/docker-containers/ebusd-dev# docker run --rm -it --device=/dev/ttyUSB0:/dev/ttyUSB0 -p 8888 john30/ebusd:rpi-latest
> Unable to find image 'john30/ebusd:rpi-latest' locally
> rpi-latest: Pulling from john30/ebusd
> cc11be355171: Pull complete
> c7765c2405e2: Pull complete
> 346e83cd4d1d: Pull complete
> 342ba6097142: Pull complete
> bb03a62953be: Pull complete
> bb71ab366af0: Pull complete
> 786acf2cba9a: Pull complete
> d85572022cf0: Pull complete
> 254bf62f105b: Pull complete
> f02532a41be1: Pull complete
> b0fcb50c1236: Pull complete
> d854d78db1f6: Pull complete
> 8a69a0ad93e8: Pull complete
> a32d7ccb782b: Pull complete
> 83daa2a0e8d8: Pull complete
> 1033c1c851d8: Pull complete
> cf97a4313546: Pull complete
> 858f06134773: Pull complete
> 26af9360b377: Pull complete
> 53e803aab872: Pull complete
> 6e7f6cebc795: Pull complete
> 0a7854975dfb: Pull complete
> 747467271911: Pull complete
> 12332f715246: Pull complete
> Digest: sha256:8401bfa975e26c6bb639c868878b2b983390b3899017a0f5aaa9f89c9fc10c69
> Status: Downloaded newer image for john30/ebusd:rpi-latest
> ebusd: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by ebusd)

Problem is that "libc6 with version 2.28" is only on "balenalib/raspberrypi3:buster" version. 
Logs from docker build  with changed Dockerfile: "FROM balenalib/raspberrypi3:buster" 

> root@NanoPi-NEO-Plus2:/var/docker-containers/ebusd-dev# docker build -t ebusd:rpi-latest-fix-buster-v0.1 .
> Sending build context to Docker daemon  4.096kB
> Step 1/13 : FROM balenalib/raspberrypi3:buster
>  ---> 9b5d8655c5d4
> Step 2/13 : RUN install_packages     logrotate libmosquitto1 libstdc++6 libc6 libgcc1     curl
>  ---> Running in eac8614d3588
> Here are a few details about this Docker image (For more information please visit https://www.balena.io/docs/reference/base-images/base-images/):
> Architecture: ARM v7
> OS: Debian Buster
> Variant: run variant
> Default variable(s): UDEV=off
> Extra features:
>  Easy way to install packages with `install_packages <package-name>` command
>  Run anywhere with cross-build feature  (for ARM only)
>  Keep the container idling with `balena-idle` command
>  Show base image details with `balena-info` command
> Reading package lists...
> Building dependency tree...
> Reading state information...
> curl is already the newest version (7.64.0-4+deb10u1).
> The following package was automatically installed and is no longer required:
>   libidn11
> Use 'apt autoremove' to remove it.
> The following additional packages will be installed:
>   cron gcc-8-base libpopt0 sensible-utils
> Suggested packages:
>   anacron checksecurity glibc-doc libc-l10n locales
> Recommended packages:
>   default-mta | mail-transport-agent bsd-mailx | mailx
> The following NEW packages will be installed:
>   cron libmosquitto1 libpopt0 logrotate sensible-utils
> The following packages will be upgraded:
>   gcc-8-base libc6 libgcc1 libstdc++6
> 4 upgraded, 5 newly installed, 0 to remove and 31 not upgraded.
> Need to get 3184 kB of archives.
[... 231 lines cut ...]
>                                  Dload  Upload   Total   Spent    Left  Speed
> 100   657  100   657    0     0   1261      0 --:--:-- --:--:-- --:--:--  1263
> 100  190k  100  190k    0     0  96126      0  0:00:02  0:00:02 --:--:--  132k
> Removing intermediate container 196b90b602d9
>  ---> 2368811fb0a2
> Step 8/13 : RUN dpkg -i ebusd.deb
>  ---> Running in 63e5ec981ac7
> Selecting previously unselected package ebusd.
> (Reading database ... 9854 files and directories currently installed.)
> Preparing to unpack ebusd.deb ...
> Unpacking ebusd (3.4) ...
> Setting up ebusd (3.4) ...
> Instructions:
>  Edit /etc/default/ebusd if necessary
>    (especially if your device is not /dev/ttyUSB0)
>  Start the daemon with 'systemctl start ebusd'
>  Check the log file /var/log/ebusd.log
>  Make the daemon autostart with 'systemctl enable ebusd'
> root@NanoPi-NEO-Plus2:/var/docker-containers/ebusd-dev# docker run -it --device=/dev/ttyUSB0 -v /var/docker-containers/ebusd/data:/etc/ebusd -p 8888 ebusd:rpi-latest-fix-stretch-v0.1 --scanconfig --device=/dev/ttyUSB0
> root@NanoPi-NEO-Plus2:/var/docker-containers/ebusd-dev#
 
Coul you please check it?
I hope, that those logs will help, if you ned some testing with docker I can help. 

Thank you.

MichlW, 2021-01-22:
I think rpi-latest tag is configured wrong as it states the platform as `amd64`:

[image]

I actually use `john30/ebusd:devel` on Raspberry Pi 4 and it is working great, because it is a multi-platform tag for `amd64` and `armv7`. You should try out that tag or `john30/ebusd:devel-arm32v7` directly...

Good luck ;-)

MichlW, 2021-01-22:
The actual documentation also states that `latest` tag is multi-platform too. Thats not true...
Only with tag `devel` or `devel-arm32v7` it is working on arm.

stevoh6, 2021-01-24:
HI, 

> I think rpi-latest tag is configured wrong as it states the platform as `amd64`:
>
> [image]
>
> I actually use `john30/ebusd:devel` on Raspberry Pi 4 and it is working great, because it is a [...]

Thank you, it worked :-)

closed 2021-01-24 by stevoh6

john30, 2021-01-24:
the latest tag will be functional again with the next release of ebusd. sorry for any inconvenience in between