john30/ebusd #391: Scan State via TCP

c0fec0de, 2021-01-19:
I know the `scan` and `info` TCP commands. But is there any possibility to see, if the scanning still running?

I am currently writing a python ebus TCP client for programmatic and CLI access. It reads all defined messages at the beginning and needs to know, that no new messages will be added.
I.e. the command `ebt observe` will produce this output:
[image]

john30, 2021-01-24:
`scan result` reveals if a scan is still running and also the HTTP port JSON includes scan running info

c0fec0de, 2021-01-24:
Thanks.

Is it valid to say:
The name of the CSV file name in the `info` statement i.e. `rcc.3` in `vaillant/f5.rcc.3.csv` ALWAYS relates to the circuit name?

c0fec0de, 2021-01-24:
After starting EBUSD, the scan runs by default (as --scanconfig is set).
But the magic line `1 scan(s) still running` is missing in `scan result`, which would be my hookup to see if all devices are known to EBUDS.

On `scan full` the line `1 scan(s) still running` shows up reliable.

john30, 2021-02-13:
the auto scan runs on demand, i.e. whenever a new device is seen via other messages on the bus, it will be scanned some time later.
so you can't simply determine scan completion that way

john30, 2021-10-06:
closed due to inactivity

closed 2021-10-06 by john30