john30/ebusd #392: Error message reading config file -

nori0077, 2021-01-23:
Hello.
I try to work with the ebus plugin. I used a CSV file for the Vaillant Recovair and place the file to folder at a loxberry. 

The startup arguments are: --scanconfig -d 192.168.1.64:8889 --accesslevel=* --configpath=/opt/loxberry/ --latency=50000

The error message from the log is:
2021-01-23 02:11:28.633 [main error] error reading config files from /opt/loxberry/: ERR: element not found, last error: 08.recov.csv:25: ERR: element not found, field type TEMP in field 0

More screenshot from the traffic:
2021-01-23 02:11:28.627 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with auto scan
2021-01-23 02:11:28.633 [main error] error reading config files from /opt/loxberry/: ERR: element not found, last error: 08.recov.csv:25: ERR: element not found, field type TEMP in field 0
2021-01-23 02:11:28.720 [bus notice] bus started with own address 31/36
2021-01-23 02:11:28.720 [bus notice] signal acquired
2021-01-23 02:12:12.774 [bus notice] new master 10, master count 2
2021-01-23 02:12:12.775 [update notice] received unknown BC cmd: 10feb516080012120223010621
2021-01-23 02:12:13.757 [bus notice] new master 03, master count 3
2021-01-23 02:12:13.757 [update notice] received unknown MS cmd: 1008b522050001012802 / 0100
2021-01-23 02:12:14.100 [update notice] received unknown MS cmd: 1008b50903298503 / 048503bc02
2021-01-23 02:12:14.442 [update notice] received unknown MS cmd: 1008b50903298603 / 0486030080
2021-01-23 02:12:14.784 [update notice] received unknown MS cmd: 1008b50903298703 / 0487030080
2021-01-23 02:12:15.135 [update notice] received unknown MS cmd: 1008b50903298803 / 0488030080
2021-01-23 02:12:15.502 [update notice] received unknown MS cmd: 1008b51303048903 / 0a8903b0042003d007b004
2021-01-23 02:12:15.912 [update notice] received unknown MS cmd: 1008b51303048a03 / 0a8a030300010006000400
2021-01-23 02:12:16.410 [update notice] received unknown MS cmd: 1008b51303048b03 / 0a8b030500010006000100
2021-01-23 02:12:16.907 [update notice] received unknown MS cmd: 1008b51303048c03 / 0a8c030000000002000100
2021-01-23 02:12:17.254 [update notice] received unknown BC cmd: 10feb51603018002
2021-01-23 02:12:20.400 [bus notice] scan 08: ;Vaillant;recoV;0225;0203
2021-01-23 02:12:20.400 [update notice] store 08 ident: done
2021-01-23 02:12:20.400 [update notice] sent scan-read scan.08  QQ=31: Vaillant;recoV;0225;0203
2021-01-23 02:12:20.400 [bus notice] scan 08: ;Vaillant;recoV;0225;0203
2021-01-23 02:12:20.400 [main error] unable to load scan config 08: list files in vaillant ERR: element not found
2021-01-23 02:12:20.400 [main error] scan config 08: ERR: element not found
2021-01-23 02:12:22.580 [main error] scan config 15: ERR: read timeout
2021-01-23 02:13:12.933 [update notice] received unknown BC cmd: 10feb516080012130223010621
2021-01-23 02:13:14.087 [update notice] received unknown MS cmd: 1008b522050001012802 / 0100
2021-01-23 02:13:14.429 [update notice] received unknown MS cmd: 1008b50903298503 / 048503bc02
2021-01-23 02:13:14.771 [update notice] received unknown MS cmd: 1008b50903298603 / 0486030080
2021-01-23 02:13:15.114 [update notice] received unknown MS cmd: 1008b50903298703 / 0487030080
2021-01-23 02:13:15.464 [update notice] received unknown MS cmd: 1008b50903298803 / 0488030080
2021-01-23 02:13:15.831 [update notice] received unknown MS cmd: 1008b51303048903 / 0a8903b0042003d007b004
2021-01-23 02:13:16.240 [update notice] received unknown MS cmd: 1008b51303048a03 / 0a8a030300010006000400
2021-01-23 02:13:16.738 [update notice] received unknown MS cmd: 1008b51303048b03 / 0a8b030500010006000100
2021-01-23 02:13:17.236 [update notice] received unknown MS cmd: 1008b51303048c03 / 0a8c030000000002000100
2021-01-23 02:13:17.584 [update notice] received unknown BC cmd: 10feb51603018002
2021-01-23 02:14:13.240 [update notice] received unknown BC cmd: 10feb516080012140223010621
2021-01-23 02:14:14.181 [update notice] received unknown MS cmd: 1008b522050001012802 / 0100
2021-01-23 02:14:14.524 [update notice] received unknown MS cmd: 1008b50903298503 / 048503bc02
2021-01-23 02:14:14.866 [update notice] received unknown MS cmd: 1008b50903298603 / 0486030080
2021-01-23 02:14:15.208 [update notice] received unknown MS cmd: 1008b50903298703 / 0487030080
2021-01-23 02:14:15.558 [update notice] received unknown MS cmd: 1008b50903298803 / 0488030080
2021-01-23 02:14:15.929 [update notice] received unknown MS cmd: 1008b51303048903 / 0a8903b0042003d007b004
2021-01-23 02:14:16.339 [update notice] received unknown MS cmd: 1008b51303048a03 / 0a8a030300010006000400
2021-01-23 02:14:16.837 [update notice] received unknown MS cmd: 1008b51303048b03 / 0a8b030500010006000100
2021-01-23 02:14:17.334 [update notice] received unknown MS cmd: 1008b51303048c03 / 0a8c030000000002000100
2021-01-23 02:14:17.682 [update notice] received unknown BC cmd: 10feb51603018002
2021-01-23 02:14:24.617 [main notice] update check: revision v3.4 available
2021-01-23 02:15:13.400 [update notice] received unknown BC cmd: 10feb516080013150223010621
2021-01-23 02:15:14.511 [update notice] received unknown MS cmd: 1008b522050001012802 / 0100
2021-01-23 02:15:14.853 [update notice] received unknown MS cmd: 1008b50903298503 / 048503ba02
2021-01-23 02:15:15.200 [update notice] received unknown MS cmd: 1008b50903298603 / 0486030080
2021-01-23 02:15:15.546 [update notice] received unknown MS cmd: 1008b50903298703 / 0487030080
2021-01-23 02:15:15.897 [update notice] received unknown MS cmd: 1008b50903298803 / 0488030080
2021-01-23 02:15:16.268 [update notice] received unknown MS cmd: 1008b51303048903 / 0a8903b0042003d007b004
2021-01-23 02:15:16.678 [update notice] received unknown MS cmd: 1008b51303048a03 / 0a8a030300010006000400
2021-01-23 02:15:17.175 [update notice] received unknown MS cmd: 1008b51303048b03 / 0a8b030500010006000100
2021-01-23 02:15:17.677 [update notice] received unknown MS cmd: 1008b51303048c03 / 0a8c030000000002000100
2021-01-23 02:15:18.108 [update notice] received unknown BC cmd: 10feb51603018002
2021-01-23 02:16:13.660 [update notice] received unknown BC cmd: 10feb516080013160223010621
2021-01-23 02:16:14.771 [update notice] received unknown MS cmd: 1008b522050001012802 / 0100
2021-01-23 02:16:15.113 [update notice] received unknown MS cmd: 1008b50903298503 / 048503ba02
2021-01-23 02:16:15.460 [update notice] received unknown MS cmd: 1008b50903298603 / 0486030080
2021-01-23 02:16:15.806 [update notice] received unknown MS cmd: 1008b50903298703 / 0487030080
2021-01-23 02:16:16.153 [update notice] received unknown MS cmd: 1008b50903298803 / 0488030080
2021-01-23 02:16:16.520 [update notice] received unknown MS cmd: 1008b51303048903 / 0a8903b0042003d007b004
2021-01-23 02:16:16.930 [update notice] received unknown MS cmd: 1008b51303048a03 / 0a8a030300010006000400
2021-01-23 02:16:17.427 [update notice] received unknown MS cmd: 1008b51303048b03 / 0a8b030500010006000100
2021-01-23 02:16:17.925 [update notice] received unknown MS cmd: 1008b51303048c03 / 0a8c030000000002000100
2021-01-23 02:16:18.271 [update notice] received unknown BC cmd: 10feb51603018002
2021-01-23 02:16:19.425 [update notice] received unknown BC cmd: 10feb510020601

The communication is working and the broadcastmessages come back, but not anything more.  Hope you can help me out.

john30, 2021-01-24:
you probably missed to copy all the _templates.csv files:
* [_templates.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/_templates.csv)
* [vaillant/_templates.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/_templates.csv)

nori0077, 2021-01-24:
Hello.
Thanks for quick response. During the time waiting for an answer, I copied the files memory.csv, broadcast.csv and _templates in the folder /etc/ebusd/. I also copied the folder "vaillant" to this directory. I copied also the csv file for a Recovair (got from another page) to the vaillant folder. 
It seems, that changes something, but it still doesn`t work. Seems that the csv file I got worked with another soft- hardware combination. Am I right?

 2021-01-24 21:45:32.019 [main notice] ebusd stopped
2021-01-24 21:45:36.392 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with auto scan
2021-01-24 21:45:36.506 [bus notice] bus started with own address 31/36
2021-01-24 21:45:36.513 [bus notice] signal acquired
2021-01-24 21:45:44.541 [bus notice] new master 10, master count 2
2021-01-24 21:45:44.541 [update notice] received unknown BC cmd: 10feb516080043452124010721
2021-01-24 21:45:45.480 [bus notice] new master 03, master count 3
2021-01-24 21:45:45.480 [update notice] received unknown MS cmd: 1008b522050001012802 / 0100
2021-01-24 21:45:45.823 [update notice] received unknown MS cmd: 1008b50903298503 / 048503bc02
2021-01-24 21:45:46.169 [update notice] received unknown MS cmd: 1008b50903298603 / 0486030080
2021-01-24 21:45:46.511 [update notice] received unknown MS cmd: 1008b50903298703 / 0487030080
2021-01-24 21:45:46.618 [bus notice] scan 08: ;Vaillant;recoV;0225;0203
2021-01-24 21:45:46.618 [update notice] store 08 ident: done
2021-01-24 21:45:46.618 [update notice] sent scan-read scan.08  QQ=31: Vaillant;recoV;0225;0203
2021-01-24 21:45:46.618 [bus notice] scan 08: ;Vaillant;recoV;0225;0203
2021-01-24 21:45:46.621 [main error] unable to load scan config 08: no file from vaillant with prefix 08 matches ID "recov", SW0225, HW0203
2021-01-24 21:45:46.621 [main error] scan config 08: ERR: element not found
2021-01-24 21:45:46.882 [update notice] received unknown MS cmd: 1008b50903298803 / 0488030080
2021-01-24 21:45:47.249 [update notice] received unknown MS cmd: 1008b51303048903 / 0a8903b0042003d007b004
2021-01-24 21:45:47.663 [update notice] received unknown MS cmd: 1008b51303048a03 / 0a8a030300010006000400
2021-01-24 21:45:48.161 [update notice] received unknown MS cmd: 1008b51303048b03 / 0a8b030500010006000100
2021-01-24 21:45:48.658 [update notice] received unknown MS cmd: 1008b51303048c03 / 0a8c030000000002000100
2021-01-24 21:45:48.825 [main error] scan config 15: ERR: read timeout
2021-01-24 21:45:49.371 [update notice] received unknown BC cmd: 10feb5160301d001

Is there an option to change the csv file that this one match with my combination? By the way, I used the DE folder, because I´m german, but think shouldn`t be a difference. Or is there something wrong with my starts urguments at the plugin? Is the way to give in the path correct?

closed 2021-01-24 by nori0077

nori0077, 2021-01-24:
The information about to copy the files to folder was the key. After that it works. It works also fine for the loxberry plugin. It`s only necessary to copie a little bit manual, if you want to use the recovair from Vaillant with this plugin for loxberry. Thanks a lot for the program and support.

hermannf1976, 2023-10-29:
Hallo John,
I hope you can help me:
The ebusd adapter works well so far when querying a Vaillant RecovAir 260/4 with a Loxberry ebusd Plugin. Except for the values ​​that should return a temperature value. If I comment out these lines then the query runs through and I get all other values. Otherwise the query hangs and I don't get all the values.

--scanconfig --localhost --accesslevel='*' --enabledefine --device=enh:192.168.1.131:9999 --configpath=/opt/loxberry/data/plugins/ebusdplugin --latency=50000

File 08.recov.csv in /opt/loxberry/data/plugins/ebusdplugin:

# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment
#,recoV,RecoVair,RecoVair,,,,,,,,,,
*r,,,,,,B509,0D,,,,,,
*w,,,,,,B509,0E,,,,,,
*wi,#install,,,,,B509,0E,,,,,,
# ##### Generell #####,,,,,,,,,,,,,
r;w,,FanLevelDay,L fterstufe Tag,,,,8a03,,,UCH,1=1;2=2;3=3;4=4;5=5;6=6,,
r;w,,FanLevelNight,L fterstufe Nacht,,,,8b03,,,UCH,1=1;2=2;3=3;4=4;5=5;6=6,,
#r;w,,Heatrecovery,W rmer ckgewinnung,,,,8c03,,,UCH,0=auto;1=on;2=off,,Aktue lle Uhrzeit
r;w,,FilterChange,Tage bisFilterwechsel,,,,350e,,,UIN,,Tage,
r;w,,maintenance,Tage bis Wartung,,,,360e,,,UIN,,Tage,
r,,DayKwh,Ertrag Heute kwh /10,,,,560e,,,ULG,10,Kwh,
r,,YesterdayKwh,Ertrag Vortag kwh /10,,,,570e,,,ULG,10,Kwh,
r,,MonthKwh,Ertrag Monat kwh /10,,,,580e,,,ULG,10,Kwh,
r,,YearKwh,Ertrag Jahr kwh,,,,590e,,,ULG,,Kwh,
r,,TotalKwh,Ertrag Gesamt kwh,,,,5a0e,,,ULG,,Kwh,
r;w,,InstallationHight,Installationsh he in m,,,,1a0e,,,UIN,,m,
r;w,,Disbalance,Dissbalance Abluft %,,,,1b0e,,,SCH,,%,
r;w,,VolumeFlow,Nennvolumenstrom NL,,,,1c0e,,,UIN,,m /h,
r;w,,RedVolumeFlow,Red.l ftung % zu NL,,,,1d0e,,,UCH,,%,
r;w,,IntensVolumeFlow,Intensivl ftung % zu NL,,,,1e0e,,,UCH,,%,
r;w,,WTTyp,WTTyp,,,,250e,,,UCH,0=Standart;1=Etalph ie,,
r;w,,FreezProtect,FrostschutzTyp,,,,260e,,,UCH,0=k ein;1=elektrisch;2=hydraulisch,,
r;w,,GroundCollector,Erdkollektor,,,,280e,,,UCH,0= nein;1=ja,,
**#r;w,,BypassDiffTemp,BypassDiffTemp,,,,2a0e,,,temp, ,°C,**
r;w,,PressureSwitch,Druckw chter,,,,2b0e,,,UCH,0=nein;1=ja,,
r;w,,StepSwitch,Stufenschalter,,,,2c0e,,,UCH,0=nei n;1=ja,,
r;w,,AirQualitySens,Luftqualit tssensor,,,,2d0e,,,UCH,0=keiner;1=1Sensor;2=2Senso ren,,
r;w,,MinCO2,min CO2 Wert,,,,2e0e,,,UIN,,ppm,
r;w,,MaxCO2,max CO2 Wert,,,,2f0e,,,UIN,,ppm,
r;w,,MinHum,min Luftfeuchte %,,,,300e,,,UCH,,,
r;w,,MaxHum,max Luftfeuchte %,,,,310e,,,UCH,,,
r,,VolumeFlowAct,aktueller Volumenstrom,,,,040e,,,UIN,,,
r,,ByPass,ByPass Klappe,,,,0a0e,,,UCH,0=on;01=mid;02=off,,
r;w,,FanStage,Luefterstufe,,,,150e,,,UCH,0=low;1=a uto;2=mid;3=high,,
**#r;w,,Desiredtemp,Wunschtemp,,,,140e,,,temp,,°C,
#r,,OutsideAirTemp,AussenTemp,,,,0c0e,,,temp,,°C,
#r,,ExhaustAirTemp,FortLuft,,,,0d0e,,,temp,,°C,
#r,,OutletAirTemp,AbLuft,,,,b40d,,,temp,,°C,
#r,,InletAirTemp,ZuLuft,,,,b30d,,,temp,,°C,**
r,,HeatElement,Frostschutzelement,,,,f10c,,,UCH,0= nein;1=ja,,
r,,BypassPos,Bypass Position,,,,0b0e,,,UCH,,,

version | ebusd 21.3.v21.3
signal | acquired
max symbol rate | 43
max arbitration micros | 3
max symbol latency | 8
masters | 2
conditional | 0
update | 0
address 08 | slave #11, scanned "MF=Vaillant;ID=recoV;SW=0225;HW=0203"
address 36 | slave #8, ebusd
args | --scanconfig --localhost --accesslevel='*' --enabledefine --device=enh:192.168.1.131:9999 --configpath=/opt/loxberry/data/plugins/ebusdplugin --latency=50000
access | *
symbol rate | 24
min arbitration micros | 2
min symbol latency | 7
reconnects | 0
messages | 30
poll | 0
address 03 | master #11
address 31 | master #8, ebusd

Please can you find out what the problem ist? Thanks a lot. Hermann