john30/ebusd #393: No data with RS232-->USB reader

tadex327, 2021-01-24:
Bus GitHub

Hey guys,

I have been trying to get ebusd working on my Raspberry Pi with my Esera Ebus RS232 Serial Reader, but I simply can’t do it. I’m a total newbie when it comes to Pi, so please excuse some lack of knowledge here and there. I’ve been reading tons of sites and forum threads for the past days, but I’m currently stuck and would greatly appreciate any help!

### Here’s a quick status quo:

Raspberry Pi 4, OS: Raspbian Buster v10
Esera eBus Reader RS232 connected to Vaillant Heater (the USB version isn’t currently available, I had to stick with the RS232)
Prolific PL2303 Serial to USB cable connected to the USB2.0 port of the Pi

ebusd was installed as per official instructions and working fine as a autostart service running as root with the following settings:
`-d /dev/ttyUSB0 --scanconfig -lograwdata=bytes`
However, I can’t get any valid signals from the device. The „Data“ LED of the reader is constantly lit and not flashing, no matter how I adjust the potentiometer.

This is ebusctl I: no Vaillant heating device is recognized
```
pi@raspberrypi:~ $ ebusctl i
version: ebusd 21.1.v21.1
signal: acquired
symbol rate: 203
max symbol rate: 232
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

When I kill the ebusd service and let ebusd run in the foreground, I get mixed results, which confuses me a lot!
Sometimes I get:

```
pi@raspberrypi:~ $ sudo ebusd -f -lograwdata=bytes --scanconfig
2021-01-24 13:17:49.209 [main notice] ebusd 21.1.v21.1 started with full scan
2021-01-24 13:17:49.360 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
2021-01-24 13:17:49.361 [bus notice] bus started with own address 31/36
2021-01-24 13:17:49.361 [bus notice] device invalid
2021-01-24 13:17:54.361 [bus error] unable to open /dev/ttyUSB0: ERR: generic device error
```

Other times I get:

```
pi@raspberrypi:~ $ sudo ebusd -f -lograwdata=bytes --scanconfig
2021-01-23 23:39:45.266 [main notice] ebusd 21.1.v21.1 started with auto scan
2021-01-23 23:39:45.429 [bus notice] bus started with own address 31/36
2021-01-23 23:39:45.430 [bus notice] signal acquired
2021-01-23 23:39:48.006 [bus notice] max. symbols per second: 207
2021-01-23 23:40:29.001 [bus notice] max. symbols per second: 227
2021-01-23 23:40:30.002 [bus notice] max. symbols per second: 231
2021-01-23 23:40:31.003 [bus notice] max. symbols per second: 232
2021-01-23 23:40:33.002 [bus notice] max. symbols per second: 233
```
With no >aa signals at all

This is dmesg | grep usb:
```
pi@raspberrypi:~ $ dmesg | grep usb
[    0.130249] usbcore: registered new interface driver usbfs
[    0.130305] usbcore: registered new interface driver hub
[    0.130425] usbcore: registered new device driver usb
[    0.130753] usb_phy_generic phy: phy supply vcc not found, using dummy regulator
[    1.163785] usbcore: registered new interface driver r8152
[    1.163853] usbcore: registered new interface driver lan78xx
[    1.163911] usbcore: registered new interface driver smsc95xx
[    1.169028] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.04
[    1.169046] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.169061] usb usb1: Product: xHCI Host Controller
[    1.169076] usb usb1: Manufacturer: Linux 5.4.83-v7l+ xhci-hcd
[    1.169091] usb usb1: SerialNumber: 0000:01:00.0
[    1.170971] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.04
[    1.170988] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.171004] usb usb2: Product: xHCI Host Controller
[    1.171018] usb usb2: Manufacturer: Linux 5.4.83-v7l+ xhci-hcd
[    1.171033] usb usb2: SerialNumber: 0000:01:00.0
[    1.174082] usbcore: registered new interface driver uas
[    1.174186] usbcore: registered new interface driver usb-storage
[    1.185232] usbcore: registered new interface driver usbhid
[    1.185244] usbhid: USB HID core driver
[    1.532756] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    1.715381] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
[    1.715400] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    1.715416] usb 1-1: Product: USB2.0 Hub
[    2.062701] usb 1-1.4: new full-speed USB device number 3 using xhci_hcd
[    2.197248] usb 1-1.4: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 4.00
[    2.197269] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.197286] usb 1-1.4: Product: USB-Serial Controller D
[    2.197304] usb 1-1.4: Manufacturer: Prolific Technology Inc. 
[    5.165156] usbcore: registered new interface driver brcmfmac
[    5.217742] usbcore: registered new interface driver usbserial_generic
[    5.217817] usbserial: USB Serial support registered for generic
[    5.233480] usbcore: registered new interface driver pl2303
[    5.233557] usbserial: USB Serial support registered for pl2303
[    5.273064] usb 1-1.4: pl2303 converter now attached to ttyUSB0
```

The device PL2303 is successfully connected to ttyUSB0.

I have configured the device to work on 2400 8N1 as required per instruction manual and tried to reach the device with CU and 
CuteCom: CU shows some random character output before a hangup signal is received and CuteCom is not able to reach the device („Device Error - Resource temporarily unavailable“)

Only one instance of ebusd is running
The ebusd.log cannot be opened („document not UTF-8 valid. Invalid byte sequence“)
My user „pi“ has been added to groups dialout tty and netdev:
```
pi@raspberrypi:~ $ groups
pi root adm tty dialout cdrom sudo audio video plugdev games users input netdev lpadmin gpio i2c spi
```

Any thoughts on how to tackle the problem would be highly appreciated!
Thank you very much in advance!

tadex327, 2021-01-30:
Hello again,

I am able to receive signals via ebusd now! I had to change the read/write permissions for the USB device:

```
pi@raspberrypi:~ $ ls -la /dev/ttyUSB0
crwxrwxrwx 1 root dialout 188, 0 Jan 29 14:17 /dev/ttyUSB0
```

Now, I am able to receive bytes, but ebusd is not detecting any device nor is it loading any configuration. Here's what ebusctl and the ebusd.log look like at the moment:

```
pi@raspberrypi:~ $ ebusctl i
version: ebusd 21.1.v21.1
update check: version 3.4 available
signal: acquired
symbol rate: 191
max symbol rate: 200
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

```
2021-01-30 12:12:18.788 [bus notice] <20
2021-01-30 12:12:18.936 [bus notice] <60
2021-01-30 12:12:18.941 [bus notice] <20
2021-01-30 12:12:18.943 [bus notice] <90
2021-01-30 12:12:18.944 [bus notice] <12
2021-01-30 12:12:18.946 [bus notice] <24
2021-01-30 12:12:18.946 [main notice] SIGHUP received
2021-01-30 12:12:18.947 [bus notice] <26
2021-01-30 12:12:18.949 [bus notice] <80
2021-01-30 12:12:18.950 [bus notice] <20
2021-01-30 12:12:18.952 [bus notice] <40
2021-01-30 12:12:18.953 [bus notice] <fb
2021-01-30 12:12:18.955 [bus notice] <22
2021-01-30 12:12:18.956 [bus notice] <00
2021-01-30 12:12:18.958 [bus notice] <6c
2021-01-30 12:12:18.959 [bus notice] <f8
2021-01-30 12:12:18.961 [bus notice] <a1
2021-01-30 12:12:18.962 [bus notice] <04
2021-01-30 12:12:18.964 [bus notice] <02
2021-01-30 12:12:18.965 [bus notice] <00
2021-01-30 12:12:18.967 [bus notice] <08
2021-01-30 12:12:18.968 [bus notice] <20
2021-01-30 12:12:18.970 [bus notice] <08
2021-01-30 12:12:18.971 [bus notice] <8c
2021-01-30 12:12:18.973 [bus notice] <60
2021-01-30 12:12:18.974 [bus notice] <0c
2021-01-30 12:12:18.976 [bus notice] <02
2021-01-30 12:12:18.977 [bus notice] <fc
2021-01-30 12:12:18.979 [bus notice] <00
2021-01-30 12:12:18.980 [bus notice] <04
2021-01-30 12:12:18.982 [bus notice] <0d
2021-01-30 12:12:18.983 [bus notice] <fc
2021-01-30 12:12:18.985 [bus notice] <20
2021-01-30 12:12:18.986 [bus notice] <33
2021-01-30 12:12:18.988 [bus notice] <20
2021-01-30 12:12:18.989 [bus notice] <40
2021-01-30 12:12:18.991 [bus notice] <44
2021-01-30 12:12:18.992 [bus notice] <00
2021-01-30 12:12:18.994 [bus notice] <f0
2021-01-30 12:12:18.995 [bus notice] <69
2021-01-30 12:12:18.997 [bus notice] <40
2021-01-30 12:12:18.998 [bus notice] <44
2021-01-30 12:12:19.002 [bus notice] <fc
2021-01-30 12:12:19.008 [bus notice] <6d
2021-01-30 12:12:19.012 [bus notice] <25
```

Any hints on how to make it work would be greatly appreciated :)

Cheers!

john30, 2021-01-30:
the raw output does not look like eBUS traffic at all. Did you adjust the potentiometer as [stated here](https://github.com/john30/ebusd/wiki/6.-Hardware#adjusting-the-potentiometer)?

closed 2021-10-06 by john30