john30/ebusd #394: segmentation fault after new release 21.1

eh-gh, 2021-01-25:
Hi

after installing the newest version I can no start ebusd on rpi zero w

root@raspberrypi:/home/pi# ebusd
Segmentation fault

root@raspberrypi:/home/pi# cat /sys/firmware/devicetree/base/model
Raspberry Pi Zero W Rev 1.1

root@raspberrypi:/home/pi# cat /etc/debian_version
9.13

root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l GNU/Linux

root@raspberrypi:/home/pi# apt-cache show ebusd
Package: ebusd
Version: 21.1
Architecture: armhf
Maintainer: John Baier <ebusd@ebusd.eu>
Homepage: https://github.com/john30/ebusd
Bugs: https://github.com/john30/ebusd/issues
Depends: logrotate, libstdc++6 (>= 4.8.1), libc6, libgcc1, libmosquitto1
Priority: required
Section: net
Filename: pool/main/e/ebusd/ebusd_21.1_armhf.deb
Size: 223670
SHA256: 50a732dbba68e63e1f4afd1fec4aa30ab03e24004d98e7ac6a46fb6cf2e735d6
SHA1: 88a1e9bd3f5ba4c149ee737f67fc5118591aec01
MD5sum: 7eb9fbd58b8ce6f4818bd39b7978a673
Description: eBUS daemon.
 ebusd is a daemon for handling communication with eBUS devices connected to a
 2-wire bus system.
Description-md5: 9fbfafb1051a5d3c5eb0b4f89ab883b3

can you check plz
thansk & br
eh-gh

john30, 2021-01-27:
just bought such a device and will check in detail once it arrives

Mike1969n, 2021-01-27:
I use an elder Raspberry Pi Model B Plus Rev 1.2

john30, 2021-01-27:
are you sure you correctly compiled it and used that compiled binary?

eh-gh, 2021-01-27:
I reinstalled the repo pkg:

root@raspberrypi:/home/pi# cat /etc/apt/sources.list.d/ebusd.list
deb https://www.ebusd.eu/repos/apt/default/stretch stretch main

root@raspberrypi:/home/pi# apt-get install --reinstall ebusd
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 0 B/224 kB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 78329 files and directories currently installed.)
Preparing to unpack .../archives/ebusd_21.1_armhf.deb ...
Unpacking ebusd (21.1) over (21.1) ...
Setting up ebusd (21.1) ...
Instructions:
1. Edit /etc/default/ebusd if necessary
   (especially if your device is not /dev/ttyUSB0)
2. Start the daemon with 'service ebusd start'
3. Check the log file /var/log/ebusd.log
4. Make the daemon autostart with 'update-rc.d ebusd enable'
Processing triggers for systemd (232-25+deb9u12) ...
root@raspberrypi:/home/pi# ebusd
Segmentation fault

manually compiled using the latest src with make_all.sh

pi@raspberrypi:~/ebusd $ git show
commit 6c5cc382db263b18a06c7db47ff2669dbd635c9b
Author: John <ebusd@ebusd.eu>
Date:   Sun Jan 24 12:28:05 2021 +0100
...

pi@raspberrypi:~/ebusd $ sudo dpkg -i ebusd-21.1_armhf_mqtt1.deb
(Reading database ... 78329 files and directories currently installed.)
Preparing to unpack ebusd-21.1_armhf_mqtt1.deb ...
Unpacking ebusd (21.1) over (21.1) ...
Setting up ebusd (21.1) ...
Instructions:
1. Edit /etc/default/ebusd if necessary
   (especially if your device is not /dev/ttyUSB0)
2. Start the daemon with 'systemctl start ebusd'
3. Check the log file /var/log/ebusd.log
4. Make the daemon autostart with 'systemctl enable ebusd'
pi@raspberrypi:~/ebusd $ ebusd
2021-01-27 20:47:42.799 [main error] invalid configpath without scanconfig

-> no seg fault

john30, 2021-01-27:
ah ok, that's what I expected. then it is a problem of the cross-compile.
for the time being please stick to the manual compile

john30, 2021-02-09:
this should be fixed with the new release [v21.2](https://github.com/john30/ebusd/releases/tag/v21.2)

eh-gh, 2021-02-09:
thanks John, but it seems to be NOK.

see strace attached

if I build the pkg ebusd-21.2_armhf_mqtt1.deb, it is still OK.

thanks

[ebusd_21.2_segfault_strace.log](https://github.com/john30/ebusd/files/5949430/ebusd_21.2_segfault_strace.log)

john30, 2021-02-09:
you need to use the armel build

eh-gh, 2021-02-09:
hi
actually the system is raspbian, which is based on armhf

I can not install armel pkg to the board:

pi@raspberrypi:/tmp $ sudo dpkg -i ebusd-21.2_armel-stretch_mqtt1.deb
dpkg: error processing archive ebusd-21.2_armel-stretch_mqtt1.deb (--install):
 package architecture (armel) does not match system (armhf)
Errors were encountered while processing:
 ebusd-21.2_armel-stretch_mqtt1.deb

john30, 2021-02-13:
can you try the following please:
`wget https://github.com/john30/ebusd/releases/download/v21.2/ebusd-21.2_armel-stretch_mqtt1.deb`
`dpkg -x ebusd-21.2_armel-stretch_mqtt1.deb /`
and then run `ebusd -V`

eh-gh, 2021-02-13:
pi@raspberrypi:/tmp $ wget https://github.com/john30/ebusd/releases/download/v21.2/ebusd-21.2_armel-stretch_mqtt1.deb
...
Connecting to github-releases.githubusercontent.com (github-releases.githubusercontent.com)|185.199.109.154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 217994 (213K) [application/octet-stream]
Saving to: ‘ebusd-21.2_armel-stretch_mqtt1.deb’

ebusd-21.2_armel-st 100%[===================>] 212.88K  --.-KB/s    in 0.1s    

2021-02-13 10:47:43 (1.53 MB/s) - ‘ebusd-21.2_armel-stretch_mqtt1.deb’ saved [217994/217994]

pi@raspberrypi:/tmp $ mkdir test 
pi@raspberrypi:/tmp $ mv ebusd-21.2_armel-stretch_mqtt1.deb test/
pi@raspberrypi:/tmp $ cd test
pi@raspberrypi:/tmp/test $ dpkg -x ebusd-21.2_armel-stretch_mqtt1.deb .
pi@raspberrypi:/tmp/test $ ls
ebusd-21.2_armel-stretch_mqtt1.deb  etc  usr
pi@raspberrypi:/tmp/test $ cd usr/bin/
pi@raspberrypi:/tmp/test/usr/bin $ ls
ebusctl  ebusd  ebuspicloader
pi@raspberrypi:/tmp/test/usr/bin $ ./ebusd  -V
ebusd 21.2.v21.2
pi@raspberrypi:/tmp/test/usr/bin $ cd ..
pi@raspberrypi:/tmp/test/usr $ cd ..
pi@raspberrypi:/tmp/test $ ls
ebusd-21.2_armel-stretch_mqtt1.deb  etc  usr
pi@raspberrypi:/tmp/test $ sudo dpkg -i ebusd-21.2_armel-stretch_mqtt1.deb
dpkg: error processing archive ebusd-21.2_armel-stretch_mqtt1.deb (--install):
 package architecture (armel) does not match system (armhf)
Errors were encountered while processing:
 ebusd-21.2_armel-stretch_mqtt1.deb

john30, 2021-02-13:
ok, so the binary works but the package is not installable due to an architecture mismatch.
does someone have an idea how to solve that?

john30, 2021-02-13:
you can force the armel installation like this:
```shell
wget https://github.com/john30/ebusd/releases/download/v21.2/ebusd-21.2_armel-stretch_mqtt1.deb
dpkg -i --force-depends ebusd-21.2_armel-stretch_mqtt1.deb
```

FritzAlfred, 2021-03-01:
Hi,
I want to install ebusd on a Raspberry zero wh.
After installation ebusd does not start with the following error message.
I tried both buster and stretch as operating system and eposd package in alternation, same result.
A Raspberry4 runs on the first try. 
(Can I communicate in German?)

Linux pizero 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Mar  1 11:39:06 2021 from 192.168.0.111
pi@pizero:~ $ sudo -s
root@pizero:/home/pi# wget -qO - https://raw.githubusercontent.com/john30/ebusd-debian/master/ebusd.gpg.key|apt-key add -
OK
root@pizero:/home/pi# wget -O /etc/apt/sources.list.d/ebusd.list https://raw.githubusercontent.com/john30/ebusd-debian/master/ebusd-default-buster.list
--2021-03-01 11:45:44--  https://raw.githubusercontent.com/john30/ebusd-debian/master/ebusd-default-buster.list
Auflösen des Hostnamens raw.githubusercontent.com (raw.githubusercontent.com)… 185.199.109.133, 185.199.108.133, 185.199.111.133, ...
Verbindungsaufbau zu raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 62 [text/plain]
Wird in »/etc/apt/sources.list.d/ebusd.list« gespeichert.

/etc/apt/sources.list.d/ebusd.list             100%[==================================================================================Linux pizero 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Mar  1 11:39:06 2021 from 192.168.0.111
pi@pizero:~ $ sudo -s
root@pizero:/home/pi# wget -qO - https://raw.githubusercontent.com/john30/ebusd-debian/master/ebusd.gpg.key|apt-key add -
OK
root@pizero:/home/pi# wget -O /etc/apt/sources.list.d/ebusd.list https://raw.githubusercontent.com/john30/ebusd-debian/master/ebusd-default-buster.list
--2021-03-01 11:45:44--  https://raw.githubusercontent.com/john30/ebusd-debian/master/ebusd-default-buster.list
Auflösen des Hostnamens raw.githubusercontent.com (raw.githubusercontent.com)… 185.199.109.133, 185.199.108.133, 185.199.111.133, ...
Verbindungsaufbau zu raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 62 [text/plain]
Wird in »/etc/apt/sources.list.d/ebusd.list« gespeichert.

/etc/apt/sources.list.d/ebusd.list             100%[=================================================================================================>]      62  --.-KB/s    in 0,002s  

2021-03-01 11:45:45 (28,4 KB/s) - »/etc/apt/sources.list.d/ebusd.list« gespeichert [62/62]

root@pizero:/home/pi# apt-get update
Holen:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15,0 kB]
OK:2 http://archive.raspberrypi.org/debian buster InRelease                                                                  
Holen:3 http://packages.microsoft.com/repos/code stable InRelease [10,4 kB]                                                  
Holen:4 https://www.ebusd.eu/repos/apt/default/buster buster InRelease [4.382 B]                           
Holen:5 http://packages.microsoft.com/repos/code stable/main amd64 Packages [15,2 kB]
Holen:6 http://packages.microsoft.com/repos/code stable/main arm64 Packages [15,8 kB]
Holen:7 http://packages.microsoft.com/repos/code stable/main armhf Packages [15,6 kB]
Holen:8 https://www.ebusd.eu/repos/apt/default/buster buster/main armhf Packages [443 B]
Es wurden 76,8 kB in 6 s geholt (13,8 kB/s).
Paketlisten werden gelesen... Fertig
root@pizero:/home/pi# apt-get install ebusd
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden zusätzlichen Pakete werden installiert:
  libmosquitto1
Die folgenden NEUEN Pakete werden installiert:
  ebusd libmosquitto1
[... 2149 lines cut ...]
Mär 01 11:51:28 pizero ebusd[991]:  failed!
Mär 01 11:51:28 pizero systemd[1]: ebusd.service: Control process exited, code=exited, status=139/n/a
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit ebusd.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 139.
Mär 01 11:51:28 pizero systemd[1]: ebusd.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit ebusd.service has entered the 'failed' state with result 'exit-code'.
Mär 01 11:51:28 pizero systemd[1]: Failed to start LSB: controls ebusd, the daemon for communication with eBUS heating systems..
-- Subject: A start job for unit ebusd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit ebusd.service has finished with a failure.
-- 
-- The job identifier is 451 and the job result is failed.
lines 1504-1555/1555 (END)
root@pizero:/home/pi#

FritzAlfred, 2021-03-01:
Sorry,
now I have probably cluttered everything!

john30, 2021-03-07:
yes, your comment is rather unreadable.
as I wrote above, it should work by forcing the armel install like this:
```sh
wget https://github.com/john30/ebusd/releases/download/v21.2/ebusd-21.2_armel-stretch_mqtt1.deb
dpkg -i --force-depends ebusd-21.2_armel-stretch_mqtt1.deb
```

AloisKlingler, 2021-03-07:
`dpkg -i --force-depends ebusd-21.2_armel-stretch_mqtt1.deb` did not work for me: `package architecture (armel) does not match system (armhf)`

I manually unpacked and copied, this worked for me:
`dpkg -x ebusd-21.2_armel-stretch_mqtt1.deb`

and afterwards:
`cp usr/bin/ebusd* /usr/bin/`

best regards
Alois

riesermauf, 2021-04-06:
Hallo
I use openhabian and had the same problem:

`openhabian@openhab:~ $ systemctl status ebusd
● ebusd.service - LSB: controls ebusd, the daemon for communication with eBUS heating systems.
   Loaded: loaded (/etc/init.d/ebusd; generated)
   Active: failed (Result: exit-code) since Tue 2021-04-06 14:36:39 CEST; 3min 40s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 531 ExecStart=/etc/init.d/ebusd start (code=exited, status=139)

Apr 06 14:36:35 openhab systemd[1]: Starting LSB: controls ebusd, the daemon for communication with eBUS heating systems....
Apr 06 14:36:38 openhab ebusd[531]: Starting ebusd: ebusd/etc/init.d/ebusd: Zeile 43:   604 Speicherzugriffsfehler  start-stop-daemo
Apr 06 14:36:39 openhab ebusd[531]:  failed!
Apr 06 14:36:39 openhab systemd[1]: ebusd.service: Control process exited, code=exited, status=139/n/a
Apr 06 14:36:39 openhab systemd[1]: ebusd.service: Failed with result 'exit-code'.
Apr 06 14:36:39 openhab systemd[1]: Failed to start LSB: controls ebusd, the daemon for communication with eBUS heating systems..
lines 1-12/12 (END)

openhabian@openhab:~ $ apt-cache show ebusd
Package: ebusd
Version: 21.2
Architecture: armhf
Maintainer: John Baier <ebusd@ebusd.eu>
Homepage: https://github.com/john30/ebusd
Bugs: https://github.com/john30/ebusd/issues
Depends: logrotate, libstdc++6 (>= 4.8.1), libc6, libgcc1, libmosquitto1
Priority: required
Section: net
Filename: pool/main/e/ebusd/ebusd_21.2_armhf.deb
Size: 213928
SHA256: e8fd7beb127a287091e543a22438db7c25fea44ec47defa548d9a0203d22ae9c
SHA1: 05badbe6b237151102ea16b9bd664675a7e01fd0
MD5sum: 90a3d247c78c28a938e97fbb6dd756d2
Description: eBUS daemon.
 ebusd is a daemon for handling communication with eBUS devices connected to a
 2-wire bus system.
Description-md5: 9fbfafb1051a5d3c5eb0b4f89ab883b3

openhabian@openhab:~ $ cat /sys/firmware/devicetree/base/model
Raspberry Pi Zero W Rev 1.1

openhabian@openhab:~ $ cat /etc/debian_version
10.9
`
No solution above works for me, what can I do.

Best regards

john30, 2021-04-11:
did you do the steps mentioned [above](https://github.com/john30/ebusd/issues/394#issuecomment-778600501)?

john30, 2021-04-11:
covered by [ebusd-debian issue](https://github.com/john30/ebusd-debian/issues/5) now

closed 2021-04-11 by john30