john30/ebusd #395: ebusd crashes with segfault on raspberrypi zero

joel-quolab, 2021-01-27:
On a raspberry pi zero the ebusd (both from the debian package management and built from source) crashes during startup:

```
root@prsc:/home/tuxx/ebusd# ebusd
Segmentation fault
root@prsc:/home/tuxx/ebusd# strace ebusd
[...]
set_robust_list(0xb6f833e0, 12)         = 0
rt_sigaction(SIGRTMIN, {sa_handler=0xb6f268e8, sa_mask=[], sa_flags=SA_RESTORER|SA_SIGINFO, sa_restorer=0xb6bd7130}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {sa_handler=0xb6f269a4, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART|SA_SIGINFO, sa_restorer=0xb6bd7130}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
ugetrlimit(RLIMIT_STACK, {rlim_cur=8192*1024, rlim_max=RLIM_INFINITY}) = 0
brk(NULL)                               = 0x22e7000
brk(0x2308000)                          = 0x2308000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x4bda56} ---
+++ killed by SIGSEGV +++
Segmentation fault
```

perhaps the pre-built kernel is lacking some functionality? Any ideas?

Mike1969n, 2021-01-27:
Ähnliches Problem bei mir

```

rt_sigprocmask(SIG_BLOCK, NULL, ~[ILL TRAP BUS FPE KILL SEGV STOP RTMIN RT_1], 8) = 0
--- SIGILL {si_signo=SIGILL, si_code=ILL_ILLOPC, si_addr=0xb68de728} ---
rt_sigprocmask(SIG_SETMASK, ~[ILL TRAP BUS FPE KILL SEGV STOP RTMIN RT_1], NULL, 8) = 0
rt_sigaction(SIGILL, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6b4b120}, NULL, 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
brk(NULL)                               = 0xc94000
brk(0xcb5000)                           = 0xcb5000
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x4d2a56} ---
+++ killed by SIGSEGV +++
Speicherzugriffsfehler

```

john30, 2021-01-27:
duplicate of #394

closed 2021-01-27 by john30