john30/ebusd #396: docker ebusd, failing to recognize system properly

garageboxnv, 2021-02-07:
Hi,

I just installed the new ebus adapter 3 to my system with an USR-ES1-module
I downloaded the latest ebusd docker and started it with:
`docker run -d --name=ebusd -p 8888 john30/ebusd -f --scanconfig -d 192.168.1.226:9999 --latency=80 --mqttport=1883 --mqtthost=172.17.0.1`
I first was trying with -d udp:192.168.1.226:9999, but then I did not get a succesfull connection

However after an hour it seems my config is not yet discovered.
I have a vaillant flexotherm exclusive heatpump with a VR71 and aguaflow module

When I go to the shell of the ebusd docker and run
`ebusctl info`
I get following output:

version: ebusd 21.1.v21.1-3-g62221bb
update check: version 3.4 available
signal: acquired
symbol rate: 46
max symbol rate: 174
min arbitration micros: 2
max arbitration micros: 2
min symbol latency: 6
max symbol latency: 6
reconnects: 0
masters: 2
messages: 12
conditional: 0
poll: 0
update: 4
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd

The command `docker logs ebusd` gives me the following output
2021-02-07 12:47:32.034 [main notice] ebusd 21.1.v21.1-3-g62221bb started with auto scan
2021-02-07 12:47:32.342 [bus notice] bus started with own address 31/36
2021-02-07 12:47:32.346 [mqtt notice] connection established
2021-02-07 12:47:32.356 [bus notice] signal acquired
2021-02-07 12:47:39.005 [bus notice] max. symbols per second: 104
2021-02-07 12:47:40.012 [bus notice] max. symbols per second: 142
2021-02-07 12:47:50.041 [bus notice] max. symbols per second: 152
2021-02-07 12:49:03.036 [bus notice] max. symbols per second: 161
2021-02-07 12:49:05.740 [bus notice] new master 10, master count 2
2021-02-07 12:49:12.224 [main error] scan config 15: ERR: read timeout
2021-02-07 12:51:14.391 [main notice] update check: version 3.4 available
2021-02-07 13:03:45.004 [bus notice] max. symbols per second: 163
2021-02-07 13:43:07.005 [bus notice] max. symbols per second: 174

When I look at the ebusd adapter v3 I see that the green led is flashing very rapidly

Not sure if it is related but when I was debugging the UDP problem I ran a temporary docker with the following command:
`docker run --rm -it -p 8888 john30/ebusd -f --scanconfig -d 192.168.33.226:9999 --latency=80 --loglevel debug`
This resulted in many log messages where the following message was repeated multiple times:
2021-02-07 12:32:16.565 [bus debug] ERR: SYN received during receive command, switching to ready
2021-02-07 12:32:16.608 [bus debug] ERR: SYN received during receive command, switching to ready
2021-02-07 12:32:16.650 [bus debug] ERR: SYN received during receive command, switching to ready
2021-02-07 12:32:16.729 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2021-02-07 12:32:16.733 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2021-02-07 12:32:16.790 [bus debug] ERR: SYN received during receive command, switching to ready
I noticed in changelog that you where fixing a problem with UDP

Wondering if I have a problem with my setup or it is related with the UDP fix.

kalledausb, 2021-02-07:
Hi,
according to the recommended settings enabling enhanced protocol is missing and using latency of 20000 (which solved my issue).
So maybe try:
`docker run --rm -it -p 8888 john30/ebusd -f --scanconfig -d enh:192.168.33.226:9999 --latency=20000 --loglevel debug`

garageboxnv, 2021-02-11:
Thanks for the information, I entered it with the above line and it works smoothly!
Just a though of me the latency isn't that very long?

Dump of configuration:

ebusctl info
version: ebusd 21.1.v21.1-3-g62221bb
update check: version 21.2 available
signal: acquired
symbol rate: 87
max symbol rate: 174
min arbitration micros: 1
max arbitration micros: 87
min symbol latency: 6
max symbol latency: 22
reconnects: 0
masters: 7
messages: 673
conditional: 15
poll: 1
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0307;HW=0403", loaded "vaillant/08.hmu.csv"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302", loaded "vaillant/0a.pmw.hwc.csv"
address 10: master #2
address 12: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0613;HW=6903", loaded "vaillant/15.700.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 70: master #4
address 71: master #9
address 75: slave #4, scanned "MF=Vaillant;ID=VR_91;SW=0415;HW=4803"
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0307;HW=0403"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0613;HW=6903", loaded "vaillant/ec.sol.sc.csv"
address f0: master #5
address f5: slave #5, scanned "MF=Vaillant;ID=VR_91;SW=0415;HW=4803"
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"

john30, 2021-02-13:
the eBUS Adapter 3 only supports TCP as well as enhanced protocol over TCP.
20 ms isn't that much and might be necessary due to network packet latency

john30, 2021-10-06:
closed due to inactivity

closed 2021-10-06 by john30