john30/ebusd #397: Massive log entries -sent poll-read ui YieldThisYear

Tranquility76, 2021-02-07:
Hi John,

I just saw in my logs the following entries:

2021-02-07 21:40:53.144 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:53.347 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:53.548 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:53.742 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:54.430 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:54.583 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:54.735 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:54.890 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:55.045 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:55.198 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:55.351 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0
2021-02-07 21:40:55.504 [update notice] sent poll-read ui YieldThisYear QQ=31: 20;0;0;0;0;0;0;0;0;0;0;0

This is flooding the log (and the bus?) as it happens continously every few milliseconds.

I am not sure if this was already there before as this is a new system (RasPi 4 with Buster and a USB ebus coppler, before it was a RasPi 3B with Buster and the same USB ebus coppler).

I am a little bit confused and scared about what this might cause on the bus...

john30, 2021-02-13:
this is a combined message consisting of 12 individual ones in order to get something similar as the 12 months graph shown on the controller UI. so whenever you want to retrieve these 12 values, all 12 messages are sent. these 12 messages are not repeated (i.e. another 12 messages being sent) within the next millis but only within the poll interval, though.
you can check by looking at subsequent blocks of these 12 messages in the log

coolex, 2021-03-02:
Hi,
I have something similar:
2021-03-02 10:19:32.138 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f00 / 00: ERR: invalid position
2021-03-02 10:19:32.273 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f08 / 00: ERR: invalid position
2021-03-02 10:19:32.409 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f10 / 00: ERR: invalid position
2021-03-02 10:19:32.544 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f18 / 00: ERR: invalid position
2021-03-02 10:19:32.679 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f20 / 00: ERR: invalid position
2021-03-02 10:19:32.816 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f28 / 00: ERR: invalid position
2021-03-02 10:19:32.951 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f30 / 00: ERR: invalid position
2021-03-02 10:19:33.087 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f38 / 00: ERR: invalid position
2021-03-02 10:19:33.221 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f40 / 00: ERR: invalid position
2021-03-02 10:19:33.356 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f48 / 00: ERR: invalid position
2021-03-02 10:19:33.492 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f50 / 00: ERR: invalid position
2021-03-02 10:19:33.627 [update error] unable to parse poll-read sc YieldThisYear from 31ecb509030d3f58 / 00: ERR: invalid position

Vaillant ecoTEC plus VUW 246/5‑5 + VRC 700

RPi3 + ESERA eBus USB
version: ebusd 21.2.v21.2
update check: OK
signal: acquired
symbol rate: 23
max symbol rate: 146
min arbitration micros: 317
max arbitration micros: 6634
min symbol latency: 0
max symbol latency: 16
reconnects: 1
masters: 3
messages: 696
conditional: 2
poll: 1
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0204;HW=9602", loaded "vaillant/bai.0010015600.inc" ([HW=9602]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0614;HW=6903", loaded "vaillant/ec.sol.sc.csv"

I hope it helps, or send me PM

john30, 2021-10-06:
it seems your solar device ec does not support the YieldThisYear message and so it answers with just a single 00

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30