john30/ebusd #398: version: ebusd 21.2.v21.2 - not recognizing all devices

NimlothPL, 2021-02-10:
**root@ebusd-raspberry:/etc/default# ebusctl info
version: ebusd 21.2.v21.2**
access: *
signal: acquired
symbol rate: 14
max symbol rate: 67
min arbitration micros: 6
max arbitration micros: 8
min symbol latency: 17
max symbol latency: 21
reconnects: 0
masters: 3
messages: 438
conditional: 0
poll: 0
update: 8
address 03: master #11
address 08: slave #11
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 26: slave
address 31: master #8, ebusd
address 36: slave #8, ebusd

After downgrade after less than 1 min:
**root@ebusd-raspberry:/etc/default# ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05**
access: *
signal: acquired
symbol rate: 25
max symbol rate: 78
min arbitration micros: 12
max arbitration micros: 46
min symbol latency: 17
max symbol latency: 76
reconnects: 0
masters: 3
messages: 631
conditional: 3
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503"
address 31: master #8, ebusd
address 36: slave #8, ebusd

john30, 2021-02-13:
are the parameter the same?
pelase provide 1 minute log of both tries

NimlothPL, 2021-02-13:
Yes, parameters are same (posted in initial message).

ebusd 3.4.v3.3-51-g57eae05
`
version: ebusd 3.4.v3.3-51-g57eae05
access: *
signal: acquired
symbol rate: 72
max symbol rate: 108
min arbitration micros: 30
max arbitration micros: 49
min symbol latency: 17
max symbol latency: 81
reconnects: 0
masters: 4
messages: 225
conditional: 3
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Vaillant;ID=VR_91;SW=0200;HW=1903"
address 36: slave #8, ebusd
`

`
2021-02-13 18:03:04.196 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with full scan
2021-02-13 18:03:04.426 [bus notice] bus started with own address 31/36
2021-02-13 18:03:04.450 [bus notice] signal acquired
2021-02-13 18:03:05.170 [mqtt notice] connection established
2021-02-13 18:03:09.114 [bus notice] new master 10, master count 2
2021-02-13 18:03:09.153 [bus notice] new master 03, master count 3
2021-02-13 18:03:09.153 [update notice] received unknown MS cmd: 1008b5110101 / 09413a20ffff510100ff
2021-02-13 18:03:09.451 [update notice] received unknown MS cmd: 1026b5230106 / 10fa01db01cc010080008000800080403d
2021-02-13 18:03:09.743 [update notice] received unknown MS cmd: 1026b5230107 / 0f008000800080008000008000805705
2021-02-13 18:03:09.997 [update notice] received unknown MS cmd: 1008b5110102 / 06033c64468c5a
2021-02-13 18:03:10.041 [bus notice] max. symbols per second: 108
2021-02-13 18:03:10.267 [update notice] received unknown MS cmd: 1008b5100900003cffffff040000 / 0101
2021-02-13 18:03:10.539 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffffe0fe
2021-02-13 18:03:10.762 [update notice] received unknown BC cmd: 10feb516080007031813020621
2021-02-13 18:03:11.037 [update notice] received unknown MS cmd: 1008b5110100 / 080f021220040f0001
2021-02-13 18:03:11.255 [update notice] received unknown BC cmd: 10feb5160301e0fe
2021-02-13 18:03:12.621 [update notice] received unknown MS cmd: 1026b5230f05ffff00000000ffffffff00000000 / 0101
2021-02-13 18:03:12.868 [update notice] received unknown MS cmd: 1026b523040200013c / 020100
2021-02-13 18:03:13.113 [update notice] received unknown MS cmd: 1026b523040201013a / 020100
2021-02-13 18:03:14.435 [main notice] starting initial full scan
2021-02-13 18:03:17.060 [bus notice] scan 08: ;Vaillant;BAI00;0604;5502
2021-02-13 18:03:17.060 [update notice] store 08 ident: done
2021-02-13 18:03:17.060 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0604;5502
2021-02-13 18:03:17.060 [bus notice] scan 08: ;Vaillant;BAI00;0604;5502
2021-02-13 18:03:19.649 [update notice] received unknown MS cmd: 1008b5110101 / 09413ae0feff510100ff
2021-02-13 18:03:21.076 [update notice] received unknown MS cmd: 1026b5230106 / 10fa01da01cb010080008000800080403d
2021-02-13 18:03:22.499 [update notice] received unknown MS cmd: 1026b5230107 / 0f008000800080008000008000805705
2021-02-13 18:03:24.687 [main notice] read common config file vaillant/scan.csv
2021-02-13 18:03:24.744 [main notice] read common config file vaillant/general.csv
2021-02-13 18:03:24.803 [main notice] read common config file vaillant/broadcast.csv
2021-02-13 18:03:24.867 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0604, HW5502
2021-02-13 18:03:26.010 [update notice] sent scan-read scan.08 id QQ=31: 
2021-02-13 18:03:26.926 [update notice] sent scan-read scan.08 id QQ=31: 
2021-02-13 18:03:27.803 [update notice] received unknown MS cmd: 1026b5230f05ffff00000000ffffffff00000000 / 0101
2021-02-13 18:03:28.695 [update notice] sent scan-read scan.08 id QQ=31: 
2021-02-13 18:03:29.592 [update notice] sent scan-read scan.08 id QQ=31: 21;14;41;0010011712;0001;007580;N7
2021-02-13 18:03:29.830 [main notice] found messages: 211 (3 conditional on 26 conditions, 0 poll, 10 update)
2021-02-13 18:03:30.334 [update notice] received unknown MS cmd: 1026b523040201013a / 020100
2021-02-13 18:03:31.226 [update notice] sent scan-read scan.08 id QQ=31: 21;14;41;0010011712;0001;007580;N7
[... 120 lines cut ...]
2021-02-13 18:04:47.039 [update notice] sent read bai Ignitor QQ=31: off
2021-02-13 18:04:47.039 [mqtt notice] read bai Ignitor: 1
2021-02-13 18:04:47.828 [update notice] received read vr_71 SensorData2 QQ=10: -;-;-;-;0.00;8.00;80 57 05
2021-02-13 18:04:50.430 [main notice] SIGTERM received
2021-02-13 18:04:51.266 [bus error] send to 08: ERR: arbitration lost, retry
2021-02-13 18:04:52.662 [update notice] received read bai Status02 QQ=10: auto;60;50.0;70;70.0
2021-02-13 18:04:53.607 [update notice] received write vr_71 SetActorState QQ=10: -;-;off;off;off;off;-;-;-;-;off;off;00 00
2021-02-13 18:04:54.417 [update notice] received write vr_71 Mc1FlowTempDesired QQ=10: on;30.0;on;0
2021-02-13 18:04:55.205 [bus error] send to 08: ERR: arbitration lost, retry
2021-02-13 18:04:55.227 [update notice] received write vr_71 Mc2FlowTempDesired QQ=10: on;29.0;on;0
2021-02-13 18:04:56.060 [update notice] received read bai Status01 QQ=10: 32.5;29.0;-0.875;-;40.5;on
2021-02-13 18:04:56.926 [update notice] received read vr_71 SensorData1 QQ=10: 31.56;29.62;28.69;-;-;-;-;40 3d
2021-02-13 18:04:57.784 [update notice] received read vr_71 SensorData2 QQ=10: -;-;-;-;0.00;8.00;80 57 05
[... cut ...]
2021-02-13 18:05:04.822 [update notice] received write vr_71 Mc2FlowTempDesired QQ=10: on;29.0;on;0
2021-02-13 18:05:05.652 [update notice] received read bai Status01 QQ=10: 32.5;29.0;-1.125;-;40.5;on
2021-02-13 18:05:06.516 [update notice] received read vr_71 SensorData1 QQ=10: 31.56;29.62;28.69;-;-;-;-;40 3d
2021-02-13 18:05:07.318 [update notice] received read vr_71 SensorData2 QQ=10: -;-;-;-;0.00;8.00;80 57 05
2021-02-13 18:05:07.603 [update notice] sent read bai WP QQ=31: on
2021-02-13 18:05:07.604 [mqtt notice] read bai WP: 1
2021-02-13 18:05:07.716 [main notice] ebusd stopped
`

NimlothPL, 2021-02-18:
And ?

john30, 2021-02-20:
please provide raw logs of the same period by starting with additional argument "--lograwdata=bytes"

> And ?

really helpful comment btw

NimlothPL, 2021-02-21:
Here you go:

> EBUSD_OPTS="--scanconfig=full -d /dev/ttyAMA0 --lograwdata=bytes --enablehex --latency=60000 --receivetimeout=50000 --sendretries=4 --accesslevel=* --mqtthost=10.62.3.197 --mqttport=1883 --mqttuser=mqttuser --mqttpass=mqttpassword -l /var/logram/ebusd.log"

[ebusd.log](https://github.com/john30/ebusd/files/6017540/ebusd.log)

john30, 2021-04-11:
the raw data is in /var/log/ebusd.log unless you add --lograwdatafile=... as well, so the log file provided does not reveal the missing info

andrewt27, 2023-01-12:
@john30 
Hi I have the same problem:

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=XXX --mqttpass=XXX --scanconfig --mqttjson --latency=10 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:192.168.1.82:9999 --log=all:error --log=main:error --log=bus:error --log=update:error --log=network:error --log=other:error
2023-01-12 11:48:59.178 [update error] unable to parse poll-read bai externalFlowTempDesired from 3108b509030d2500 / 00: ERR: invalid position
2023-01-12 11:49:02.211 [mqtt error] decode bai externalFlowTempDesired: ERR: invalid position
2023-01-12 11:49:05.201 [update error] unable to parse poll-read bai ExtFlowTempDesiredMin from 3108b509030d6e04 / 00: ERR: invalid position
2023-01-12 11:49:07.219 [mqtt error] decode bai ExtFlowTempDesiredMin: ERR: invalid position
2023-01-12 11:49:47.195 [update error] unable to parse poll-read bai HcPumpMode from 3108b509030d4904 / 00: ERR: invalid position
2023-01-12 11:49:52.528 [mqtt error] decode bai HcPumpMode: ERR: invalid position
2023-01-12 11:50:11.174 [update error] unable to parse poll-read bai HoursTillService from 3108b509030d2004 / 00: ERR: invalid position
2023-01-12 11:50:17.645 [mqtt error] decode bai HoursTillService: ERR: invalid position
2023-01-12 11:50:23.196 [update error] unable to parse poll-read bai HwcPostrunTime from 3108b509030d1104 / 00: ERR: invalid position
2023-01-12 11:50:27.161 [mqtt error] decode bai HwcPostrunTime: ERR: invalid position
2023-01-12 11:50:41.189 [update error] unable to parse poll-read bai HwcTempDesired from 3108b509030dea03 / 00: ERR: invalid position
2023-01-12 11:50:42.263 [mqtt error] decode bai HwcTempDesired: ERR: invalid position
2023-01-12 11:50:47.185 [update error] unable to parse poll-read bai HwcTempMax from 3108b509030d4304 / 00: ERR: invalid position
2023-01-12 11:50:52.272 [mqtt error] decode bai HwcTempMax: ERR: invalid position
2023-01-12 11:53:11.221 [update error] unable to parse poll-read bai SecondPumpMode from 3108b509030d0b04 / 00: ERR: invalid position
2023-01-12 11:53:12.888 [mqtt error] decode bai SecondPumpMode: ERR: invalid position
2023-01-12 11:53:35.176 [update error] unable to parse poll-read bai Status16 from 3108b5040116 / 00: ERR: invalid position
2023-01-12 11:53:37.926 [mqtt error] decode bai Status16: ERR: invalid position
2023-01-12 11:53:41.180 [update error] unable to parse poll-read bai Status from 3108b5110103 / 00: ERR: invalid position
2023-01-12 11:53:42.976 [mqtt error] decode bai Status: ERR: invalid position
2023-01-12 11:54:53.209 [update error] unable to parse poll-read bai ValveMode from 3108b509030d2a04 / 00: ERR: invalid position`

(I filtered this to only errors)

There is one minute log raw:

2023-01-12 12:03:26.754 <1008b51101018900095e5da008ff600000ffb500
2023-01-12 12:03:27.023 <1008b51009000000ffffff010000260001019a00
2023-01-12 12:03:36.749 >3108070400d1<000ab542414930300105780313>00
2023-01-12 12:03:36.859 <1008b51101018900095d5ca008ff600000ff5f00
2023-01-12 12:03:37.129 <1008b51009000000ffffff010000260001019a00
2023-01-12 12:03:37.389 <1008b51101028a0006033c96468c64e900
2023-01-12 12:03:38.618 >3108b509012448<00090032313231313130308d>00
2023-01-12 12:03:38.831 >3108b509012549<00093130303231383837307e>00
2023-01-12 12:03:39.023 >3108b50901264a<000930303130303834343816>00
2023-01-12 12:03:39.218 >3108b50901274b<00094e313c3c3c3c3c3c3c8f>00
2023-01-12 12:03:40.198 >3108b509012448<00090032313231313130308d>00
2023-01-12 12:03:40.387 >3108b509012549<00093130303231383837307e>00
2023-01-12 12:03:40.592 >3108b50901264a<000930303130303834343816>00
2023-01-12 12:03:40.789 >3108b50901274b<00094e313c3c3c3c3c3c3c8f>00
2023-01-12 12:03:42.985 >31150704008b<000ab537303030300614690350>00
2023-01-12 12:03:43.176 >3115b509012406<0009003231323130373030a7>00
2023-01-12 12:03:43.365 >3115b509012507<00093230313731333136307a>00
2023-01-12 12:03:43.550 >3115b509012604<0009303832303037353637be>00
2023-01-12 12:03:43.763 >3115b509012705<00094e3800000000000000a7>00
2023-01-12 12:03:43.945 >3108b509030d6e00c4<00010c97>00
2023-01-12 12:03:46.853 <1008b51101018900095d5ca008ff600000ff5f00
2023-01-12 12:03:47.125 <1008b51009000000ffffff010000260001019a00
2023-01-12 12:03:49.208 >3108b509030d6f005f<00010a91>00
2023-01-12 12:03:55.185 >3108b509030d81008d<00010b90>00
2023-01-12 12:03:56.910 <1008b51101018900095d5ca008ff600000ff5f00
2023-01-12 12:03:57.189 <1008b51009000000ffffff010000260001019a00
2023-01-12 12:04:01.191 >3108b509030d8200bb<00010992>00
2023-01-12 12:04:06.969 <1008b51101018900095d5cd008ff600000ffe500
2023-01-12 12:04:07.180 >3108b509030d200079<0001009b>00
2023-01-12 12:04:07.334 <1008b51009000000ffffff010000260001019a00
2023-01-12 12:04:07.618 <1008b50401003d000a02ffffffffffffffd008b600
2023-01-12 12:04:07.877 <1008b51101028a0006033c96468c64e900
2023-01-12 12:04:08.115 <10feb516080007041212010423de
2023-01-12 12:04:08.387 <1008b5110100880008ed020e001f100080d400
2023-01-12 12:04:08.606 <10feb5160301a008bb
2023-01-12 12:04:13.180 >3108b509030dda0058<00010299>00
2023-01-12 12:04:17.039 <1008b51101018900095d5ca008ff600000ff5f00
2023-01-12 12:04:17.312 <1008b51009000000ffffff010000260001019a00
2023-01-12 12:04:19.104 >3108
2023-01-12 12:04:19.163 <08
2023-01-12 12:04:25.182 >3108b509030d250023<000000>00
2023-01-12 12:04:27.995 <1008b51101018900095d5ca008ff600000ff5f00

john30, 2023-01-13:
@andrewt27 this is not related at all and the issue is closed for 2 years almost.
furthermore, the devices are recognized as otherwise you would not have a "bai" circuit.
if some messages are not being answered by the addressed device, then the config is not compatible. this is nothing unusual as the config is determined by try-and-error