john30/ebusd #399: Docker - Scan don´t work

Maddin2020, 2021-02-13:
Hello together!

I want change my runing EbusD Sytem to Docker, but the ebusd don´t work

If i want start my Docker instance with the follow command

docker run --rm -it -p 8888 john30/ebusd:v21.2 -f --scanconfig=full -d udp:192.168.178.210:5000 --latency=80 

i see only see the follow info in the console

[image]

My ebusd ethernet converter is configured with udp and port 5000. It work correctly without docker instanz since over a year.

john30, 2021-02-13:
for using UDP with docker you probably need to switch to host network, i.e. add `--network=host` to the docker run command (before the image name)

Maddin2020, 2021-02-13:
Now it works ! Thank a lot. Which ports must I route in Docker. If I only route port 5000 udp then the Daemon don´t work.

john30, 2021-02-20:
what do you mean by "route in Docker"? Are you referring to port publishing? You just need to publish all ports of ebusd that you configured and want to use, e.g.:
* TCP client as passed to the ["-p" argument (default 8888)](https://github.com/john30/ebusd/wiki/2.-Run#daemon-options)
* HTTP client as passed to the ["--httpport" argument (no default) ](https://github.com/john30/ebusd/wiki/2.-Run#daemon-options)

john30, 2021-10-06:
closed due to inactivity