john30/ebusd #401: CRC errors... only during the day

pagaille, 2021-02-25:
Hi,

I'm facing a very strange problem using ebusd. My config is the following : 

- Stand alone Vaillant Solar boiler (Aurostep Plus with a VMS/8D) 
- home made bus power supply (I used the schematic provided in the ebus specification)
- 1,5m ebus cable (twisted pair from a CAT-5 ethernet cable)
- home made ebus interface (V2, the analog, self-adjusting one)
- Raspberry pi zero W

The whole works without particular problem for some months, except one : **CRC errors, mainly during the day**.

The graph below shows it : during night everything works ok, but during the day there a lot of values missing (because of the CRC error on the bus), for no particular reason. Time span = 1 day, each graphs shows different days with various weather conditions. Between 03:00 and 06:00 one can see the solar pump running with an electric resistor to preheat the water when needed.

[image]
[image]
[image]
[image]
[image]

I can't see any direct correlation other than the temperature; it affects all the measured values. It is not linked to the solar pump running or not or anything running around the boiler.

Do you see any explanation to that ?

Thanks for sharing your thoughts.

Matthieu

john30, 2021-03-07:
is there an other circuit active on the bus? i.e. what is the result of running `ebusctl info`

pagaille, 2021-03-07:
Nope, nothing else.

# ebusctl info
version: ebusd 3.4.p20201029
update check: version 21.2 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/broadcast.csv: newer version available, vaillant/general.csv: different version available
access: *
signal: no signal
reconnects: 0
masters: 1
messages: 28
conditional: 0
poll: 0
update: 8
address 31: master #8, ebusd
address 36: slave #8, ebusd
address ed: slave, scanned "MF=Vaillant;ID=VMD01;SW=0108;HW=1703", loaded "vaillant/ed.vmd.csv"

john30, 2021-04-05:
if you add a bit of raw logging, then we can check the symbols on the bus for real CRC issues.

pagaille, 2021-04-05:
Good idea. 

```
2021-04-05 15:17:14.931 [mqtt notice] read vmd T5: 1
2021-04-05 15:17:15.058 [bus error] send to ed: ERR: CRC error, retry
```

Corresponding raw log : 

```
2021-04-05 15:17:14.933 >31edb509030d050097<00035a04009e>00
```

pagaille, 2021-04-05:
Another log (not sure which query actually gives an error, nor what means "retry" since I don't see any retry in the raw log) :

```
2021-04-05 15:35:14.960 [update notice] sent read vmd T5 QQ=31: 72.44;ok
2021-04-05 15:35:14.962 [mqtt notice] read vmd T5: 1
2021-04-05 15:35:15.088 [bus error] send to ed: ERR: CRC error, retry
2021-04-05 15:35:15.276 [update notice] sent read vmd T6 QQ=31: 34.56;ok
2021-04-05 15:35:15.276 [mqtt notice] read vmd T6: 1
2021-04-05 15:35:15.411 [update notice] sent read vmd FlowRate QQ=31: 0.00
2021-04-05 15:35:15.411 [mqtt notice] read vmd FlowRate: 1
2021-04-05 15:35:15.547 [update notice] sent read vmd T1 QQ=31: 33.94;ok
2021-04-05 15:35:15.547 [mqtt notice] read vmd T1: 1
2021-04-05 15:35:15.682 [update notice] sent read vmd T7 QQ=31: 46.62;ok
2021-04-05 15:35:15.683 [mqtt notice] read vmd T7: 1
```

```
2021-04-05 15:35:14.964 >31edb509030d050097<00038704006e>00
2021-04-05 15:35:15.201 >31edb509030da10039<000329020004
2021-04-05 15:35:15.285 >31edb509030da10039<000329020084>00
2021-04-05 15:35:15.418 >31edb509030d9e0081<000200002c>00
2021-04-05 15:35:15.552 >31edb509030d07003a<00031f020066>00
2021-04-05 15:35:15.687 >31edb509030d0c0015<0003ea02004b>00
```

pagaille, 2021-04-05:
Another case that finally fails : 

```

2021-04-05 15:50:37.882 [bus error] send to ed: ERR: CRC error, retry
2021-04-05 15:50:38.065 [bus error] send to ed: ERR: CRC error, retry
2021-04-05 15:50:38.248 [bus error] send to ed: ERR: CRC error
2021-04-05 15:50:38.250 [bus error] send message part 0: ERR: CRC error
2021-04-05 15:50:38.251 [mqtt error] read vmd T6: ERR: CRC error
```

```
2021-04-05 15:50:37.996 >31edb509030da10039<000364020080
2021-04-05 15:50:38.178 >31edb509030da10039<000364020080
2021-04-05 15:50:38.367 >31edb509030da10039<000364020080
```

pagaille, 2021-04-05:
Another one, this time realated to T5

```
2021-04-05 15:52:52.774 [bus error] send to ed: ERR: CRC error, retry
2021-04-05 15:52:52.961 [bus error] send to ed: ERR: CRC error, retry
2021-04-05 15:52:53.144 [bus error] send to ed: ERR: CRC error
2021-04-05 15:52:53.145 [bus error] send message part 0: ERR: CRC error
2021-04-05 15:52:53.146 [mqtt error] read vmd T5: ERR: CRC error
```

```
2021-04-05 15:52:52.888 >31edb509030d050097<0003000500d5
2021-04-05 15:52:53.075 >31edb509030d050097<0003000500d5
2021-04-05 15:52:53.261 >31edb509030d050097<0003000500d5
```

Some seconds later a new query is sent, this time successfully : 

   2021-04-05 15:52:57.854 >31edb509030d050097<000329050073>00

john30, 2021-04-11:
the answer from the slave "0364020080" does not carry the right CRC. since you're using an adapter with poti, I guess you need to adjust that a little to cover this

pagaille, 2021-04-11:
Thanks for the follow up @john30. Actually I’m using the V2 adapter (without any adjustment).

john30, 2021-04-11:
ah sorry, misread that. well, what can I say. the answer from the slave is invalid wrt the CRC

pagaille, 2021-04-11:
Yep. Strange. I can live with it but the correlation between the errors and the temperature readings was astonishing, I wanted to share thoughts. That might be a bug on the vmd side (strangely)

closed 2021-04-11 by pagaille