john30/ebusd #402: ebusd overwrites existing /etc/default/ebusd on upgrading

AloisKlingler, 2021-03-01:
on upgrading, /etc/default/ebusd is overwritten. normally debian-packages ask what needs to be done.

I upgraded from:
ebusd 3.4.v3.3-51-g57eae05

to ebusd 21.2.v21.2

(or: as it segfaulted to this version: https://github.com/john30/ebusd/issues/394 )

Best regards
Alois

john30, 2021-03-07:
how did you do the update?

AloisKlingler, 2021-03-07:
I did apt-get update && apt-get dist-upgrade
there the file was overwritten already

Dinth, 2021-03-07:
I'm updating wbuad by compiling it but always forgetting to back up the config. I know it's my fault, but renaming default config file to for example ebusd.orig would be a huge quality of life improvement

NimlothPL, 2021-03-29:
I can confirm, same behavior - config file is overwritten upon upgrade:
apt update && apt upgrade

john30, 2021-04-11:
this is an issue for ebusd-debian, not ebusd. copied it [there](https://github.com/john30/ebusd-debian/issues/6)

closed 2021-04-11 by john30