john30/ebusd #403: which latency with USB ebus adapter needed?

AloisKlingler, 2021-03-01:
Hi John,

I am using a USB adapter from here (https://www.mikrocontroller.net/topic/346833). The adapter has a FTDI USB converter. It is connected (and working without issues) on a raspberry pi 1, attached without any additional hub, by a short USB cable.
There is one device, a brink renovent excellent 300 cabled on the adapter.

if I use "--latency=0" as mentioned in the wiki I constantly can see "bus error signal lost" in log:

```
2021-03-01 09:01:22.111 [bus debug] ERR: read timeout during receive command ACK, switching to no signal
2021-03-01 09:01:22.112 [bus error] signal lost
2021-03-01 09:01:22.130 [bus debug] ERR: SYN received during no signal, switching to ready
2021-03-01 09:01:22.132 [bus notice] signal acquired
2021-03-01 09:01:27.137 [bus debug] ERR: read timeout during receive command ACK, switching to no signal
2021-03-01 09:01:27.139 [bus error] signal lost
2021-03-01 09:01:27.156 [bus debug] ERR: SYN received during no signal, switching to ready
2021-03-01 09:01:27.158 [bus notice] signal acquired
2021-03-01 09:01:30.402 [main debug] performing regular tasks
2021-03-01 09:01:32.168 [bus debug] ERR: read timeout during receive command ACK, switching to no signal
2021-03-01 09:01:32.170 [bus error] signal lost
```

nevertheless the communication works, so I can fetch and send data from/to the brink WTW unit.

if I use "--latency=100" (e.g. "--latency=10" does not help)  this errors are gone:

```
2021-03-01 09:07:21.398 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-03-01 09:07:25.761 [mqtt debug] publish brink/global/uptime 160
2021-03-01 09:07:26.429 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-03-01 09:07:27.330 [main debug] performing regular tasks
2021-03-01 09:07:31.462 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-03-01 09:07:36.486 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-03-01 09:07:37.334 [main debug] performing regular tasks
2021-03-01 09:07:40.474 [update info] received BC cmd: 37fe07040a400816002202ffffffff
2021-03-01 09:07:40.476 [bus notice] scan 3c: ;ENCON;  ;-;-
2021-03-01 09:07:40.478 [update notice] store broadcast ident: done
2021-03-01 09:07:40.480 [update notice] received unknown BC cmd: 37fe07040a400816002202ffffffff
2021-03-01 09:07:41.533 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-03-01 09:07:41.782 [mqtt debug] publish brink/global/uptime 176
2021-03-01 09:07:42.786 [mqtt debug] publish brink/scan.3c/ ENCON;  ;-;-
2021-03-01 09:07:46.566 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-03-01 09:07:47.337 [main debug] performing regular tasks
2021-03-01 09:07:51.589 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-03-01 09:07:56.617 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-03-01 09:07:57.341 [main debug] performing regular tasks
2021-03-01 09:07:57.807 [mqtt debug] publish brink/global/uptime 192
2021-03-01 09:08:01.600 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-03-01 09:08:06.629 [bus debug] ERR: SYN received during receive command ACK, switching to ready
```

which latency should be used by a USB connection, and if 0 needs to be used, is there anything wrong in my setup?

```
ebusctl info
version: ebusd 21.2.v21.2
update check: OK
access: '*'
signal: acquired
symbol rate: 31
max symbol rate: 162
min arbitration micros: 477
max arbitration micros: 5669
min symbol latency: 0
max symbol latency: 10
reconnects: 0
masters: 2
messages: 76
conditional: 0
poll: 0
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 37: master #18
address 3c: slave #18, scanned "MF=ENCON;ID=  ;SW=-;HW=-"
```

symbol rate is sometimes 22

Thank you!

Best regards
Alois

john30, 2021-03-07:
there might be issues with buffering done by the serial converter.
if you run ebusd with `--lograwdata=bytes` for a while, you can see the delay of incoming data

AloisKlingler, 2021-03-08:
`2021-03-08 10:35:18.946 [bus debug] send/receive symbol latency 5 ms
2021-03-08 10:35:18.948 [bus notice] >50
2021-03-08 10:35:18.953 [bus notice] <50
2021-03-08 10:35:18.953 [bus debug] send/receive symbol latency 5 ms
2021-03-08 10:35:18.955 [bus notice] >01
2021-03-08 10:35:18.962 [bus notice] <01
2021-03-08 10:35:18.962 [bus debug] send/receive symbol latency 6 ms
2021-03-08 10:35:18.963 [bus notice] >01
2021-03-08 10:35:18.969 [bus notice] <01
2021-03-08 10:35:18.969 [bus debug] send/receive symbol latency 5 ms
`

do you mean these messages?

john30, 2021-04-05:
yes. here you can see a usual latency of 4-6 ms. but you should set the "--latency" to at least 20ms