john30/ebusd #404: several values are read as "0.0" after update from ebusd 3.4.v3.3 to ebusd 21.2.v21.2

AloisKlingler, 2021-03-02:
Hello John,

since update I cannot see a lot values which are fetched flawless with old ebusd version.

same configuration file and same startup parameters are used, only /usr/bin/ebusd* are exchanged between the logs below.

start parameters in /etc/default/ebusd are:
`-d /dev/ttyUSB0 --configpath=/etc/ebusd/ --latency=100 --loglevel=error --mqtthost=localhost --mqttport=1883 --mqtttopic=brink --accesslevel='*' --httpport=8880 -l /tmp/ebusd.log`

configfile attached.

i trigger the read with ebusctl read -f <param>

ebusd 3.4 debuglog:

```
2021-03-02 08:06:01.841 [main debug] >>> read -f IstwertAbluftdruck
2021-03-02 08:06:01.844 [bus info] send message: 313c40220115
2021-03-02 08:06:01.856 [bus debug] arbitration delay 1473 micros
2021-03-02 08:06:01.857 [bus debug] switching from ready to send command
2021-03-02 08:06:01.865 [bus debug] send/receive symbol latency 5 ms
2021-03-02 08:06:01.874 [bus debug] send/receive symbol latency 5 ms
2021-03-02 08:06:01.883 [bus debug] send/receive symbol latency 5 ms
2021-03-02 08:06:01.892 [bus debug] send/receive symbol latency 5 ms
2021-03-02 08:06:01.901 [bus debug] send/receive symbol latency 5 ms
2021-03-02 08:06:01.901 [bus debug] switching from send command to send command CRC
2021-03-02 08:06:01.910 [bus debug] send/receive symbol latency 5 ms
2021-03-02 08:06:01.910 [bus debug] switching from send command CRC to receive command ACK
2021-03-02 08:06:01.914 [bus debug] switching from receive command ACK to receive response
2021-03-02 08:06:01.927 [bus debug] switching from receive response to receive response CRC
2021-03-02 08:06:01.931 [bus debug] switching from receive response CRC to send response ACK
2021-03-02 08:06:01.939 [bus debug] send/receive symbol latency 0 ms
2021-03-02 08:06:01.941 [bus info] send/receive symbol latency 0 - 9 ms
2021-03-02 08:06:01.943 [update info] sent MS cmd: 313c40220115 / 02008c
2021-03-02 08:06:01.949 [update notice] sent read kwl IstwertAbluftdruck QQ=31: 14.0
2021-03-02 08:06:01.953 [bus debug] notify request: done
2021-03-02 08:06:01.954 [bus debug] read res: 02008c
2021-03-02 08:06:01.955 [main info] read kwl IstwertAbluftdruck: 14.0
2021-03-02 08:06:01.959 [main debug] <<< 14.0
```

and the result:
```
   "IstwertAbluftdruck": {
    "name": "IstwertAbluftdruck",
    "passive": false,
    "write": false,
    "lastup": 1614669175,
    "zz": 60,
    "fields": {
     "0": {"name": "", "value": 14.0}
    }
   }
```

ebusd 21.2.v21.2:
```
2021-03-02 07:59:43.959 [main debug] >>> read -f IstwertAbluftdruck
2021-03-02 07:59:43.959 [bus info] send message: 313c40220115
2021-03-02 07:59:44.000 [bus debug] start request 31
2021-03-02 07:59:44.000 [bus debug] arbitration start with 31
2021-03-02 07:59:44.053 [bus debug] arbitration won
2021-03-02 07:59:44.053 [bus debug] arbitration delay 611 micros
2021-03-02 07:59:44.053 [bus debug] switching from ready to send command
2021-03-02 07:59:44.063 [bus debug] send/receive symbol latency 7 ms
2021-03-02 07:59:44.072 [bus debug] send/receive symbol latency 6 ms
2021-03-02 07:59:44.080 [bus debug] send/receive symbol latency 5 ms
2021-03-02 07:59:44.090 [bus debug] send/receive symbol latency 6 ms
2021-03-02 07:59:44.099 [bus debug] send/receive symbol latency 6 ms
2021-03-02 07:59:44.100 [bus debug] switching from send command to send command CRC
2021-03-02 07:59:44.109 [bus debug] send/receive symbol latency 6 ms
2021-03-02 07:59:44.109 [bus debug] switching from send command CRC to receive command ACK
2021-03-02 07:59:44.112 [bus debug] switching from receive command ACK to receive response
2021-03-02 07:59:44.126 [bus debug] switching from receive response to receive response CRC
2021-03-02 07:59:44.130 [bus debug] switching from receive response CRC to send response ACK
2021-03-02 07:59:44.140 [bus debug] send/receive symbol latency 6 ms
2021-03-02 07:59:44.140 [update info] sent MS cmd: 313c40220115 / 02008c
2021-03-02 07:59:44.142 [update notice] sent read kwl IstwertAbluftdruck QQ=31: 0.0
2021-03-02 07:59:44.142 [bus debug] notify request: done
2021-03-02 07:59:44.142 [bus debug] read res: 313c40220115
2021-03-02 07:59:44.144 [main info] read kwl IstwertAbluftdruck: 0.0
2021-03-02 07:59:44.146 [main debug] <<< 0.0
```

with the result:
```
"IstwertAbluftdruck": {
    "name": "IstwertAbluftdruck",
    "passive": false,
    "write": false,
    "lastup": 1614667495,
    "zz": 60,
    "fields": {
     "0": {"name": "", "value": 0.0}
    }
}
```

i can see:
2021-03-02 08:06:01.954 [bus debug] read res: 02008c
vs
2021-03-02 07:59:44.142 [bus debug] read res: 313c40220115

but the read result (I assume it is 02008c) transported in both cases:
2021-03-02 08:06:01.943 [update info] sent MS cmd: 313c40220115 / 02008c
2021-03-02 07:59:44.140 [update info] sent MS cmd: 313c40220115 / 02008c

please tell me if you need more logs. :-)

Thanks!

Best regards
Alois

elliot64, 2021-03-05:
Hi,

same problem here.
Several values seem to be correct but most are zero since update.

Regards,
Roland

john30, 2021-03-07:
please send the message definition. best would be to get the result of `ebusctl f -f IstwertAbluftdruck` from both ebusd versions.
which platform are you on?

AloisKlingler, 2021-03-07:
ebusd 3.4
```
ebusctl f -f IstwertAbluftdruck
r,kwl,IstwertAbluftdruck,40220115,,3c,4022,15,,s,UIR,10,Pa,PressureExhaust
```

ebusd 21.2

```
ebusctl f -f IstwertAbluftdruck
r,kwl,IstwertAbluftdruck,40220115,,3c,4022,15,,s,UIR,10,Pa,PressureExhaust
```

RPI1 which is armv6:

```
uname -a
Linux heliotherm 5.10.11+ #1399 Thu Jan 28 12:02:28 GMT 2021 armv6l GNU/Linux

uname -m
armv6l
```

also attached the 3c.csv

thank you!

Best regards
Alois
[3c.txt](https://github.com/john30/ebusd/files/6097529/3c.txt)

john30, 2021-04-05:
I can't reproduce this. can you try compiling and running the latest sources?

GuillaumeC, 2021-06-04:
Hi,

Same problem for me.
I tried to downgrade from last v21 to v3.4 and now I have more values.

john30, 2021-06-06:
ping @AloisKlingler

AloisKlingler, 2021-06-06:
> ping @AloisKlingler

I will try to find some spare time this or next week.
Br
Alois

JeeCee1, 2021-06-08:
Hi John, I had the same issue.
I compiled the lasted source (using [these](https://github.com/john30/ebusd/wiki/1.-Build-and-install) instructions on a Raspbi), and now it works!
Do you need any debug info? Please explain me how to get it.
If it helps, the values which gave a '0' value had a semicolon ';' appended.

I'm having this #398 issue also on version 21.2 but on 3.4 it does see my VRC700.

JeeCee1, 2021-06-11:
ping @john30 (I'm not sure if you get topic update notifications?)

reulitz, 2021-09-11:
I had the same problem as the OP, could solve the problem by compiling the souce, ended up with version: ebusd 21.2.v21.2-36-gfb5ab14

Hardware: Raspberry Pi A+; armv61

john30, 2021-10-06:
note to myself: seems like the docker built deb package didn't get the right compiler

john30, 2021-11-10:
please re-check with latest release

john30, 2022-03-12:
closed due to inactivity and probably fixed with 4bda1a5 & f1cac3c already

closed 2022-03-12 by john30