john30/ebusd #405: Need to use /get to read Mqtt values even when subscribed to the topic

rgillam100, 2021-03-07:
Hi John,

Firstly, thanks for such a great project !
I have been reading my flow and return temperatures using  ebusd via mosquito mqtt  to node red.
To get the temperatures to update in the mqtt topic ebusd/bai/FlowTemp, I first need to publish to ebusd/bai/FlowTemp/get
Although this woks successfully, I would have expected  ebusd  to publish new values to the broker whenever the value changes.
I see that the topics ebusd/global do get updated when they change.
Is there a setting to tell ebusd to publish all topics when their value changes?

I have read https://github.com/john30/ebusd/wiki/3.3.-MQTT-client ...many times. 
I have tried --accesslevel="*"
Also tried --mqttchanges  

This is my start command
 ebusd --mqttport=1883 -s -f --device=/dev/ttyUSB1

Thanks
Rob

rgillam100, 2021-03-07:
version: ebusd 3.4.v3.4-83-gab75329
update check: version 21.2 available
signal: acquired
symbol rate: 79
max symbol rate: 150
min arbitration micros: 17
max arbitration micros: 505
min symbol latency: 0
max symbol latency: 13
reconnects: 0
masters: 2
messages: 214
conditional: 3
poll: 0
update: 11
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0603;HW=1303", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 64: slave, scanned "MF=Vaillant;ID=V6500;SW=0007;HW=8301", loaded "vaillant/64.v65.csv"

rgillam100, 2021-03-07:
This is my find output, only the topics I have previously sent a /get to have any data stored

bai AccessoriesOne = no data stored
bai AccessoriesTwo = no data stored
bai ACRoomthermostat = no data stored
bai APCComStatus = no data stored
bai APCLegioProtection = no data stored
bai averageIgnitiontime = no data stored
bai BlockTimeHcMax = no data stored
bai BoilerType = no data stored
bai ChangesDSN = no data stored
bai CirPump = no data stored
bai CounterStartattempts1 = no data stored
bai CounterStartattempts2 = no data stored
bai CounterStartAttempts3 = no data stored
bai CounterStartAttempts4 = no data stored
bai currenterror = no data stored
bai DateTime = no data stored
bai dcfState = no data stored
bai DCFTimeDate = no data stored
bai DCRoomthermostat = no data stored
bai DeactivationsIFC = no data stored
bai DeactivationsTemplimiter = no data stored
bai DeltaFlowReturnMax = no data stored
bai DisplayMode = no data stored
bai DSN = no data stored
bai DSNOffset = no data stored
bai DSNStart = no data stored
bai EBusHeatcontrol = no data stored
bai EbusSourceOn = no data stored
bai EbusVoltage = no data stored
bai errorhistory = no data stored
bai expertlevel_ReturnTemp = no data stored
bai ExternalFaultmessage = no data stored
bai externalFlowTempDesired = no data stored
bai externalHwcSwitch = no data stored
bai ExternGasvalve = no data stored
bai ExtFlowTempDesiredMin = no data stored
bai ExtStorageModulCon = no data stored
bai extWP = no data stored
bai FanHours = no data stored
bai FanMaxSpeedOperation = no data stored
bai FanMinSpeedOperation = no data stored
bai FanPWMSum = no data stored
bai FanPWMTest = no data stored
bai FanSpeed = no data stored
bai FanSpeedOffsetMax = no data stored
bai FanSpeedOffsetMin = no data stored
bai FanStarts = no data stored
bai Flame = no data stored
bai FlameSensingASIC = no data stored
bai FloorHeatingContact = no data stored
bai FlowsetHcMax = no data stored
bai FlowsetHwcMax = no data stored
bai FlowSetPotmeter = no data stored
bai FlowTemp = 24.12;ok
bai FlowTempDesired = no data stored
bai FlowTempMax = no data stored
bai Fluegasvalve = no data stored
bai FluegasvalveOpen = no data stored
bai Gasvalve3UC = no data stored
bai Gasvalve = no data stored
bai GasvalveASICFeedback = no data stored
bai GasvalveUC = no data stored
bai GasvalveUCFeedback = no data stored
bai HcHours = 6373
bai HcPumpMode = no data stored
bai HcPumpStarts = no data stored
bai HcStarts = no data stored
bai HcUnderHundredStarts = no data stored
[... 94 lines cut ...]
bai WaterpressureBranchControlOff = no data stored
bai WaterpressureMeasureCounter = no data stored
bai WaterpressureVariantSum = no data stored
bai WP = no data stored
bai WPPostrunTime = no data stored
bai WPSecondStage = no data stored
broadcast datetime = no data stored
broadcast error = no data stored
broadcast hwcStatus = no data stored
broadcast id = no data stored
broadcast id = no data stored
broadcast load = no data stored
broadcast outsidetemp = no data stored
broadcast signoflife = no data stored
broadcast vdatetime = no data stored
general valuerange = no data stored
memory eeprom = no data stored
memory ram = no data stored
scan id = no data stored
scan.08  = Vaillant;BAI00;0603;1303
scan.08 id = 21;15;46;0010015673;1300;606111;N4
scan.64  = Vaillant;V6500;0007;8301
v65 State = standby;standby;no;-

192.168.0.4:

Dinth, 2021-03-07:
Add this script to your crontab https://github.com/john30/ebusd/blob/master/contrib/scripts/readall.sh

rgillam100, 2021-03-07:
Hi Michal,

Thanks for the script, it updates all the topics .
If I use this with ----mqttchanges  I get the result i was looking for.

Cheers
Rob

closed 2021-03-07 by rgillam100

pagaille, 2021-03-07:
Why not use selective polling at node red level instead of full poll buried in cron ?

I use this :

```
var registers = {
'vmd': [
	'T5', // Status01
	'T6',
	'FlowRate',
	'T1',
	'T7'
]
};

var result = [];

for (var circuit in registers) {
	registers[circuit].forEach(function(reg) {
		result.push({
		    topic: "ebusd/"+circuit+"/"+reg+"/get",
		    payload: 1
		});
	});
}

return [result];
```
[image]

rgillam100, 2021-03-07:
Thats what I was doing, and will revert back too.
I needed to see all the values and monitor which ones were being updated  for some testing I was doing.
It would be nice if ebusd updated all topics when they changed, then subscribing to ebusd/# would be all thats needed