john30/ebusd #406: ebuspicloader: setting IP fails

schinken, 2021-03-18:
hi,

today I tried to set the IP for my ebusd3-Ethernet Adapter. Connected the USB and instantly shows up in DMESG. 
Then I tried the following:
```bash
sudo ./ebuspicloader -i 10.23.42.2 -m 24 /dev/ttyUSB0
```

This results in a  `did not receive sync: 0x20`. If I try to use "slow mode"

```bash
sudo ./ebuspicloader -i 10.23.42.2 -m 24 -s /dev/ttyUSB0
```

It says `did not receive sync: 0x20`. What am I doing wrong? :)

Best,
schinken

schinken, 2021-03-18:
I recompiled ebuspicloader with DEBUG_RAW:

```bash
sudo ./ebuspicloader -i 10.23.42.2 -m 24 -v /dev/ttyUSB0                                         1 ↵
> 1/1: 55
> 9/9: 00 00 00 00 00 00 00 00 00
< 1/1: 20
did not receive sync: 0x20
```
And with slow

```bash
sudo ./ebuspicloader -i 10.23.42.2 -m 24 -s /dev/ttyUSB0                                         1 ↵
> 1/1: 55
> 9/9: 00 00 00 00 00 00 00 00 00
< 1/1: fe
did not receive sync: 0xfe
```

john30, 2021-03-18:
did you follow the [instructions](https://adapter.ebusd.de/picfirmware.en.html#firmware-update)? i.e.
* disconnect the USR
* jumper to USB function
* disconnect power
* jumper J11 pins 3-4
* power-on

schinken, 2021-03-18:
Sorry for my dumb question. I totally did not see that I had to use the bootloader-mode to set the IP address. Just thought for update.

closed 2021-03-18 by schinken