john30/ebusd #411: Docker - missing plattforms in stable

jutonium, 2021-03-25:
Thanks for the new ebus adapter 3. Unfortunately the stable docker image does not contain all platforms which are listed on the docker page. I had to switch to devel for linux/arm64/v8.
```
docker run --rm -it -p 8888 john30/ebusd -f --scanconfig -d 192.168.182.13:9999 --latency=80
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
standard_init_linux.go:219: exec user process caused: exec format error

docker run --platform linux/arm64/v8 --rm -it -p 8888 john30/ebusd -f --scanconfig -d 192.168.182.13:9999 --latency=80
Unable to find image 'john30/ebusd:latest' locally
latest: Pulling from john30/ebusd
Digest: sha256:fb8cb181acdec4444b32f1742f5ceaa2cfd9a763b9631c0ff21269176cdd65e9
Status: Image is up to date for john30/ebusd:latest
docker: Error response from daemon: image with reference john30/ebusd was found but does not match the specified platform: wanted linux/arm64/v8, actual: linux/amd64.
See 'docker run --help'.

sudo docker run --rm -it -p 8888 john30/ebusd:devel -f --scanconfig -d 192.168.182.13:9999 --latency=80
2021-03-25 22:10:38.912 [main notice] ebusd 21.2.v21.2-12-g86b700c started with auto scan on device 192.168.182.13:9999
2021-03-25 22:10:39.229 [bus notice] bus started with own address 31/36
^C2021-03-25 22:10:50.517 [main notice] SIGINT received
2021-03-25 22:10:54.368 [main notice] ebusd stopped

```

john30, 2021-04-11:
strange, seems the docker hooks don't work sometimes

netweaver1970, 2021-04-23:
same as my new issue (in wrong area, mea culpa) : [https://github.com/john30/ebusd-esp/issues/42](https://github.com/john30/ebusd-esp/issues/42)

I'm going to the native Debian route, for the time being.

bdariusb, 2021-10-16:
there are platform specific images https://hub.docker.com/r/john30/ebusd/tags
At least for RPI4 I used `john30/ebusd:v21.2-arm32v7`

john30, 2021-11-10:
solved now for the latest and v21.3 tags

closed 2021-11-10 by john30