john30/ebusd #413: I can't get any reading/data at all with a USB Ebusd on any version later than 3.4

nimmerpeople, 2021-04-10:
Hi,

I have an Ebus adapter from Florin here: https://www.tindie.com/products/foxrom/ebus-usb-adapter-ebuzzz-adapter/ to connect to my Vaillant boiler to read values from it.

I struggled for aged (having never used ebus or any of this hardware) but have, with Florin's help determined that ANY version above 3.4 just won't read the device. at all.

I have since downgraded and all is 100% fine. I just post this here incase anyone else has the same problem. I'm running Debian on a Raspberry Pi and happy to provide logs if this helps.

The version I have working is here:

https://github.com/john30/ebusd/releases/download/v3.4/ebusd-3.4_armhf-stretch_mqtt1.deb

john30, 2021-04-11:
try compiling from sources and post some logging as the description is not enough to find the issue

nimmerpeople, 2021-04-11:
Ok, will do. Thanks. I realised you’d need more info, just wasn’t sure what..

nimmerpeople, 2021-04-11:
Ok, will do. Thanks. I realised you’d need more info, just wasn’t sure what..
Sorry to be so lame but I'm new to this. I've followed the guide from the wifi and got this far:

'apt install git autoconf automake g++ make'
- which worked fine

Then:
'git clone https://github.com/john30/ebusd.git'
'cd ebusd'

Again ok

Then I want to compile? so I type:

'make'
and I get errors:

make: *** No targets specified and no makefile found.  Stop.

I'll try and read up and see what I should do next..

john30, 2021-04-11:
just follow [the documentation](https://github.com/john30/ebusd/wiki/1.-Build-and-install)

nimmerpeople, 2021-04-11:
> just follow [the documentation](https://github.com/john30/ebusd/wiki/1.-Build-and-install)

err I did - it's from that documentation that I'm quoting above..

john30, 2021-04-11:
I don't see autogen.sh being called before running make

nimmerpeople, 2021-04-11:
> I don't see autogen.sh being called before running make

Ah thanks - that's just my ignorance then -I didn't think I needed that bit.. i'm learning though ;-) 
 - thanks!

nimmerpeople, 2021-04-11:
Ok, so I am closer to working out the problem (but will likely need help to progress).
When I compile the latest version 21.x - it works fine in the foreground all fine. It's with MQTT that I can't get it to work. I've added libmosquitto-dev support - by typing
 _sudo apt-get install libmosquitto-dev_
which I think is right but nothing starts in the /var/log/ebusd.log once I reboot.

I'll see if I can get any further. The contents of 
/etc/default/ebusd

are the same as I had working on 3.4, so I don't think it's that.

john30, 2021-05-06:
you need to rebuild everything after you have added the libmosquitto-dev package