john30/ebusd #414: eBus config file loading?

Racailloux, 2021-04-11:
Hello,

We have a ebus adapter v2.2 since monthes, working fine working under USB on a Raspberry Pi3B.
Since 2/3 days, we lost the update of our variables that we used to retrieve by eBus.

We made several tests, including the reinstall of a fresh Rasbian buster with only the ebusd.
So we installed a buster raspbian on another R.Pi3B + the package from https://github.com/john30/ebusd-debian and connected only the ebus adapter on it.

We can connect to the ebus and retrieve data, but only sometime. The major part of the time, there is "ERR: element not found" issue that bocks the rest of the process.
This seems to be linked to the reading/accessing of the config files. We resent the "scan full" request, but that doesn't seem to reload the config files.
The only way to make it run it to stop/start the ebus daemon several time until the retrieve of the config files is OK. (we have a stable Internet connection).

Here is a "classic" return that we have :

> 2021-04-10 22:41:38.664 [main notice] ebusd 21.2.v21.2 started with auto scan on device /dev/serial/by-id/usb-Silico$
2021-04-10 22:41:43.681 [main error] error reading config files from http://ebusd.eu/config/: ERR: element not found$
2021-04-10 22:41:43.698 [bus notice] bus started with own address 31/36
2021-04-10 22:41:43.733 [bus notice] signal acquired
2021-04-10 22:43:58.130 [main error] update check connect error
2021-04-10 22:44:47.888 [bus notice] new master 03, master count 2
2021-04-10 22:44:48.509 [bus notice] scan 08: ;Vaillant;BAI00;0609;5502
2021-04-10 22:44:48.509 [update notice] store 08 ident: done
2021-04-10 22:44:48.510 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0609;5502
2021-04-10 22:44:48.510 [bus notice] scan 08: ;Vaillant;BAI00;0609;5502
2021-04-10 22:44:48.635 [main error] error reading templates in vaillant: ERR: element not found, last error: vailla$
2021-04-10 22:44:48.692 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0609, HW5502
2021-04-10 22:44:48.692 [main error] error resolving conditions: ERR: element not found, last error: condition scan $
2021-04-10 22:44:48.818 [main error] error executing instructions: ERR: element not found, last error: error resolvi$
2021-04-10 22:44:48.818 [main notice] found messages: 1 (0 conditional on 25 conditions, 0 poll, 0 update)
2021-04-10 22:46:50.877 [main notice] update check: OK

On the other side, the scan result is OK:
>  sudo ebusctl scan result
08;Vaillant;BAI00;0609;5502

But we can't retrive data (because the element is not well listed:
> ebusctl read -f -c bai FlowTemp
ERR: element not found

We also tested without using it as a service and this seems to be less complicated for the deamon to get the config files :
> sudo ebusd --scanconfig --port=8888 -d /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -f --loglevel=debug

Is there any way to get the config files locally (internet access is not always a need to run a system)?
Do you have some suggestions of tests or actions to solve this?

john30, 2021-04-11:
this sounds weird and you have funny "$" signs in your logging, which should not be there.
you'd have to do a tcpdump or wireshark dump of the traffic to and from the webservice.
using local files is easy with cloning the config repo and pointing ebusd to the ebusd-2.1.x/de/vaillant/de (or /en) folder using "--configpath"

Racailloux, 2021-04-11:
Hello John30!
Thanks for your fast answer!

The "$" sign should come from the copy/paste from Putty, don't take care of it :)

I cloned the git of https://github.com/john30/ebusd-configuration and used this setting:
> EBUSD_OPTS="--scanconfig --port=8888 -d /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/en/vaillant --loglevel=debug"

There is still some issue at loading files and still can't load correctly the config files (as I understand the logs).
The boiler is detected on the bus, but I still can't retrieve the data:
> 2021-04-11 10:19:23.289 [main notice] ebusd 21.2.v21.2 started with auto scan on device /dev/serial/by-id/usb-Silico$
2021-04-11 10:19:23.291 [main info] loading configuration files from /home/pi/ebusd-configuration/ebusd-2.1.x/en/vai$
2021-04-11 10:19:23.310 [main info] reading templates /
2021-04-11 10:19:23.314 [main error] error reading templates in /: ERR: element not found, last error: _templates.cs$
2021-04-11 10:19:23.314 [main info] reading file 05.vd4.csv
2021-04-11 10:19:23.328 [main error] error reading config files from /home/pi/ebusd-configuration/ebusd-2.1.x/en/vai$
2021-04-11 10:19:23.344 [bus notice] bus started with own address 31/36
2021-04-11 10:19:23.345 [main info] registering data handlers
2021-04-11 10:19:23.345 [main info] registered data handlers
2021-04-11 10:19:23.348 [bus notice] signal acquired
2021-04-11 10:19:33.345 [main debug] performing regular tasks
2021-04-11 10:19:52.697 [main debug] performing regular tasks
2021-04-11 10:20:02.697 [main debug] performing regular tasks
2021-04-11 10:20:12.698 [main debug] performing regular tasks
2021-04-11 10:20:22.698 [main debug] performing regular tasks
2021-04-11 10:20:32.699 [main debug] performing regular tasks
2021-04-11 10:20:42.700 [main debug] performing regular tasks
2021-04-11 10:20:52.700 [main debug] performing regular tasks
2021-04-11 10:21:02.701 [main debug] performing regular tasks
2021-04-11 10:21:12.701 [main debug] performing regular tasks
2021-04-11 10:21:22.702 [main debug] performing regular tasks
2021-04-11 10:21:32.702 [main debug] performing regular tasks
2021-04-11 10:21:32.764 [main notice] update check: OK
2021-04-11 10:21:42.764 [main debug] performing regular tasks
2021-04-11 10:21:52.764 [main debug] performing regular tasks
2021-04-11 10:22:02.765 [main debug] performing regular tasks
2021-04-11 10:22:12.765 [main debug] performing regular tasks
2021-04-11 10:22:22.766 [main debug] performing regular tasks
2021-04-11 10:22:32.766 [main debug] performing regular tasks
2021-04-11 10:22:42.766 [main debug] performing regular tasks
2021-04-11 10:22:52.767 [main debug] performing regular tasks
2021-04-11 10:23:02.767 [main debug] performing regular tasks
2021-04-11 10:23:12.768 [main debug] performing regular tasks
2021-04-11 10:23:22.768 [main debug] performing regular tasks
2021-04-11 10:23:32.768 [main debug] performing regular tasks
2021-04-11 10:23:42.769 [main debug] performing regular tasks
2021-04-11 10:23:52.769 [main debug] performing regular tasks
2021-04-11 10:23:53.862 [bus notice] new master 03, master count 2
2021-04-11 10:23:53.891 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-04-11 10:23:59.242 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-04-11 10:24:02.770 [main debug] performing regular tasks
2021-04-11 10:24:02.770 [bus info] scan 08 cmd: 3108070400
2021-04-11 10:24:02.811 [bus debug] start request 31
2021-04-11 10:24:02.811 [bus debug] arbitration start with 31
2021-04-11 10:24:02.858 [bus debug] arbitration won
2021-04-11 10:24:02.858 [bus debug] arbitration delay 1646 micros
2021-04-11 10:24:02.858 [bus info] arbitration delay 1646 - 1646 micros
2021-04-11 10:24:02.858 [bus debug] switching from ready to send command
2021-04-11 10:24:02.867 [bus debug] send/receive symbol latency 5 ms
2021-04-11 10:24:02.867 [bus info] send/receive symbol latency 5 - 5 ms
2021-04-11 10:24:02.876 [bus debug] send/receive symbol latency 5 ms
2021-04-11 10:24:02.885 [bus debug] send/receive symbol latency 5 ms
2021-04-11 10:24:02.894 [bus debug] send/receive symbol latency 5 ms
2021-04-11 10:24:02.894 [bus debug] switching from send command to send command CRC
2021-04-11 10:24:02.901 [bus debug] send/receive symbol latency 5 ms
2021-04-11 10:24:02.901 [bus debug] switching from send command CRC to receive command ACK
2021-04-11 10:24:02.906 [bus debug] switching from receive command ACK to receive response
2021-04-11 10:24:02.958 [bus debug] switching from receive response to receive response CRC
2021-04-11 10:24:02.962 [bus debug] switching from receive response CRC to send response ACK
2021-04-11 10:24:02.971 [bus debug] send/receive symbol latency 5 ms
2021-04-11 10:24:02.971 [update info] sent MS cmd: 3108070400 / 0ab5424149303006095502
2021-04-11 10:24:02.971 [bus notice] scan 08: ;Vaillant;BAI00;0609;5502
2021-04-11 10:24:02.971 [bus debug] switching from send response ACK to send SYN
2021-04-11 10:24:02.971 [main error] unable to load scan config 08: list files in vaillant ERR: element not found
2021-04-11 10:24:02.971 [main error] scan config 08: ERR: element not found
2021-04-11 10:24:02.979 [bus debug] send/receive symbol latency 4 ms
2021-04-11 10:24:02.980 [bus info] send/receive symbol latency 4 - 5 ms
2021-04-11 10:24:04.598 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-04-11 10:24:04.972 [main debug] performing regular tasks
2021-04-11 10:24:09.956 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-04-11 10:24:14.972 [main debug] performing regular tasks

john30, 2021-04-11:
I just tested it and online config works fine for me.

john30, 2021-10-06:
does this issue still occur?