john30/ebusd #415: Docker exec format error

Trilis29, 2021-04-12:
standard_init_linux.go:219: exec user process caused: exec format error

Getting this error. I can not run on docker. Has anyone solved it?

andig, 2021-04-12:
Youre likely running the wrong image for your platform.

Trilis29, 2021-04-12:
tried DEVEL works ok. Seems stable got problems. RPI 3b+ HA OS

netweaver1970, 2021-04-28:
> tried DEVEL works ok. Seems stable RPI 3b+ HA OS

How are you running ebusd on HassOS? As a manual "docker run" (from within the terminal & SSH add-on) or as a custom own-build HA  add-on pointing to the john30/ebusd docker hub image or did you install ebusd as a Debian package? 

Right now I'm running it as a manual "docker run" also on a separate Raspi 3b+but would like to consolidate all on the HassOs box.

Trilis29, 2021-04-29:
Just use ebusd addon in supervisor HassOS. Developer works ok for me.

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30