john30/ebusd #417: ERR: read timeout during skip, switching to no signal

lennyzaman, 2021-04-18:
Hello all,

I recently bought the ebus adapter v3 in usb mode. 
this is connected to my rpi 3 model B via USB.
The adapter is wired to the ebus 2 wire line in the VRC370 wireless module.
I installed ebusd via repo through apt-get.
when running ebusd (in foreground for testing) I get following output.
`root@raspberrypi1:/home/pi# ebusd -d /dev/ttyUSB1 --mqtthost=192.168.0.188 --mqttport=1883 --mqttclientid=ebusd1 --mqttuser=******* --mqttpass=******* --scanconfig=full -f --configpath=http://ebusd.eu/config/ --loglevel=debug --enablehex --latency=10000 --receivetimeout=50000 --sendretries=4 --accesslevel=* -p 8889 --lograwdata=bytes`
2021-04-19 00:24:13.513 [main notice] ebusd 21.2.v21.2 started with full scan on device /dev/ttyUSB1
2021-04-19 00:24:13.513 [main info] loading configuration files from http://ebusd.eu/config/
2021-04-19 00:24:13.576 [main info] reading templates /
2021-04-19 00:24:13.641 [main info] read templates in /
2021-04-19 00:24:13.641 [main info] reading file memory.csv
2021-04-19 00:24:13.708 [main info] successfully read file memory.csv
2021-04-19 00:24:13.708 [main info] reading file broadcast.csv
2021-04-19 00:24:13.777 [main info] successfully read file broadcast.csv
2021-04-19 00:24:13.777 [main info] read config files
2021-04-19 00:24:13.779 [bus notice] bus started with own address 31/36
2021-04-19 00:24:13.780 [main info] registering data handlers
2021-04-19 00:24:13.780 [mqtt info] mosquitto version 1.5.7 (compiled with 1.5.7)
2021-04-19 00:24:13.799 [mqtt debug] connection requested
2021-04-19 00:24:13.799 [main info] registered data handlers
2021-04-19 00:24:13.805 [mqtt notice] connection established
2021-04-19 00:24:13.805 [mqtt debug] publish ebusd/global/version ebusd 21.2.v21.2
2021-04-19 00:24:13.805 [mqtt debug] publish ebusd/global/running true
2021-04-19 00:24:16.054 [bus notice] <fa
2021-04-19 00:24:16.054 [bus notice] signal acquired
2021-04-19 00:24:18.036 [bus debug] ERR: read timeout during skip, switching to no signal
2021-04-19 00:24:18.036 [bus error] signal lost
2021-04-19 00:24:23.800 [main debug] performing regular tasks
2021-04-19 00:24:23.800 [mqtt debug] publish ebusd/global/scan finished
2021-04-19 00:24:29.812 [mqtt debug] publish ebusd/global/uptime 16
2021-04-19 00:24:33.800 [main debug] performing regular tasks
2021-04-19 00:24:43.801 [main debug] performing regular tasks
2021-04-19 00:24:45.835 [mqtt debug] publish ebusd/global/uptime 32
2021-04-19 00:24:53.801 [main debug] performing regular tasks
2021-04-19 00:24:58.276 [bus notice] <ff
2021-04-19 00:24:58.276 [bus notice] signal acquired
2021-04-19 00:25:00.009 [bus debug] ERR: read timeout during skip, switching to no signal
2021-04-19 00:25:00.009 [bus error] signal lost
2021-04-19 00:25:01.857 [mqtt debug] publish ebusd/global/uptime 48
2021-04-19 00:25:03.802 [main debug] performing regular tasks
2021-04-19 00:25:13.802 [main debug] performing regular tasks
2021-04-19 00:25:17.879 [mqtt debug] publish ebusd/global/uptime 64
2021-04-19 00:25:20.402 [bus notice] <fa
2021-04-19 00:25:20.402 [bus notice] signal acquired
2021-04-19 00:25:22.013 [bus debug] ERR: read timeout during skip, switching to no signal
2021-04-19 00:25:22.014 [bus error] signal lost
2021-04-19 00:25:23.802 [main debug] performing regular tasks
2021-04-19 00:25:33.803 [main debug] performing regular tasks
2021-04-19 00:25:33.902 [mqtt debug] publish ebusd/global/uptime 80
2021-04-19 00:25:43.803 [main debug] performing regular tasks
2021-04-19 00:25:49.923 [mqtt debug] publish ebusd/global/uptime 96
2021-04-19 00:25:53.804 [main debug] performing regular tasks
2021-04-19 00:26:03.299 [bus notice] <fa
2021-04-19 00:26:03.299 [bus notice] signal acquired
2021-04-19 00:26:03.804 [main debug] performing regular tasks
2021-04-19 00:26:03.804 [main notice] starting initial full scan
2021-04-19 00:26:03.805 [bus info] scan 02 cmd: 3102070400
2021-04-19 00:26:03.805 [mqtt debug] publish ebusd/global/scan running
2021-04-19 00:26:05.033 [bus info] scan 04 cmd: 3104070400
2021-04-19 00:26:05.033 [bus info] scan 05 cmd: 3105070400
2021-04-19 00:26:05.033 [bus info] scan 06 cmd: 3106070400
2021-04-19 00:26:05.033 [bus info] scan 08 cmd: 3108070400
2021-04-19 00:26:05.033 [bus info] scan 09 cmd: 3109070400
2021-04-19 00:26:05.033 [bus info] scan 0a cmd: 310a070400
2021-04-19 00:26:05.033 [bus info] scan 0b cmd: 310b070400
2021-04-19 00:26:05.033 [bus info] scan 0c cmd: 310c070400
2021-04-19 00:26:05.034 [bus info] scan 0d cmd: 310d070400
2021-04-19 00:26:05.034 [bus info] scan 0e cmd: 310e070400
[... 203 lines cut ...]
2021-04-19 00:26:05.047 [bus info] scan ec cmd: 31ec070400
2021-04-19 00:26:05.047 [bus info] scan ed cmd: 31ed070400
2021-04-19 00:26:05.048 [bus info] scan ee cmd: 31ee070400
2021-04-19 00:26:05.048 [bus info] scan ef cmd: 31ef070400
2021-04-19 00:26:05.048 [bus info] scan f2 cmd: 31f2070400
2021-04-19 00:26:05.048 [bus info] scan f4 cmd: 31f4070400
2021-04-19 00:26:05.048 [bus info] scan f5 cmd: 31f5070400
2021-04-19 00:26:05.048 [bus info] scan f6 cmd: 31f6070400
2021-04-19 00:26:05.048 [bus info] scan f8 cmd: 31f8070400
2021-04-19 00:26:05.048 [bus info] scan f9 cmd: 31f9070400
2021-04-19 00:26:05.049 [bus info] scan fa cmd: 31fa070400
2021-04-19 00:26:05.049 [bus info] scan fb cmd: 31fb070400
2021-04-19 00:26:05.049 [bus info] scan fc cmd: 31fc070400
2021-04-19 00:26:05.049 [bus info] scan fd cmd: 31fd070400
2021-04-19 00:26:05.049 [bus debug] ERR: read timeout during skip, switching to no signal
2021-04-19 00:26:05.049 [bus error] signal lost
2021-04-19 00:26:05.812 [mqtt debug] publish ebusd/global/uptime 112
2021-04-19 00:26:13.805 [main debug] performing regular tasks

I notice It receives 1 byte, acquired signal then disconnects...
Any idea what is going wrong here?
I tried multiple combinations of latency without any help...

Thank you
Lenny

lennyzaman, 2021-04-19:
As an additional info:
Vaillant HR-gas-wandheizgerät mit brennwerttechnik VCW BE 346/5-5 R4
Vaillant VRC 370f

EBUS connected through 2 wire cable (in the heating unit on the EBUS connector(which is also branched into the VRC370f remote unit)

USB cable to connect to Raspberry pi 3 model B+ running Raspbian buster.

I tried a few more settings and with following I read the bus but I only see '<fe' output.(at a very high rate, sometimes also '<ff')
I also flashed the latest PIC firmware to the device (though it reported this was already present)

`ebusd -d /dev/ttyUSB0 --mqtthost=192.168.0.188 --mqttport=1883 --mqttclientid=ebusd1 --mqttuser=openhab --mqttpass=habopen --scanconfig=full -f --configpath=http://ebusd.eu/config/ --loglevel=debug --enablehex --accesslevel=* -p 8889 --lograwdata=bytes`
(The device is now ttyUSB0 iso USB1 as in the before example)
See following log: 
> 2021-04-19 18:12:56.748 [main notice] ebusd 21.2.v21.2 started with full scan on device /dev/ttyUSB0
> 2021-04-19 18:12:56.749 [main info] loading configuration files from http://ebusd.eu/config/
> 2021-04-19 18:12:56.820 [main info] reading templates /
> 2021-04-19 18:12:56.893 [main info] read templates in /
> 2021-04-19 18:12:56.893 [main info] reading file memory.csv
> 2021-04-19 18:12:56.959 [main info] successfully read file memory.csv
> 2021-04-19 18:12:56.959 [main info] reading file broadcast.csv
> 2021-04-19 18:12:57.033 [main info] successfully read file broadcast.csv
> 2021-04-19 18:12:57.033 [main info] read config files
> 2021-04-19 18:12:57.037 [bus notice] bus started with own address 31/36
> 2021-04-19 18:12:57.038 [main info] registering data handlers
> 2021-04-19 18:12:57.038 [mqtt info] mosquitto version 1.5.7 (compiled with 1.5.7)
> 2021-04-19 18:12:57.058 [mqtt debug] connection requested
> 2021-04-19 18:12:57.058 [main info] registered data handlers
> 2021-04-19 18:12:57.062 [mqtt notice] connection established
> 2021-04-19 18:12:57.062 [mqtt debug] publish ebusd/global/version ebusd 21.2.v21.2
> 2021-04-19 18:12:57.062 [mqtt debug] publish ebusd/global/running true
> 2021-04-19 18:12:57.068 [bus notice] <fe
> 2021-04-19 18:12:57.069 [bus notice] signal acquired
> 2021-04-19 18:12:57.111 [bus notice] <fe
> 2021-04-19 18:12:57.155 [bus notice] <fe
> 2021-04-19 18:12:57.198 [bus notice] <fe
> 2021-04-19 18:12:57.240 [bus notice] <fe
> 2021-04-19 18:12:57.283 [bus notice] <fe
> 2021-04-19 18:12:57.326 [bus notice] <fe
> 2021-04-19 18:12:57.368 [bus notice] <fe
> 2021-04-19 18:12:57.412 [bus notice] <fe
> 2021-04-19 18:12:57.456 [bus notice] <fe
> 2021-04-19 18:12:57.499 [bus notice] <fe
> 2021-04-19 18:12:57.541 [bus notice] <fe
> 2021-04-19 18:12:57.585 [bus notice] <fe
> 2021-04-19 18:12:57.628 [bus notice] <fe
> 2021-04-19 18:12:57.672 [bus notice] <fe
> 2021-04-19 18:12:57.715 [bus notice] <fe

lennyzaman, 2021-04-19:
output from ebusctl -p 8889 info

> root@raspberrypi1:/home/pi# ebusctl -p 8889 info
> version: ebusd 21.2.v21.2
> update check: OK
> access: *
> signal: no signal
> reconnects: 221
> masters: 1
> messages: 11
> conditional: 0
> poll: 0
> update: 4
> address 31: master #8, ebusd
> address 36: slave #8, ebusd

output from ebusctl -p 8889 scan:

> root@raspberrypi1:/home/pi# ebusctl -p 8889 scan
> empty

john30, 2021-06-06:
if you jumpered the adapter to enhanced mode, you need to pass "-d enh:/dev/ttyUSB1" to ebusd

lennyzaman, 2021-06-06:
Hi John30,
I will try that when I am home next week! Thank you for the input!

closed 2021-10-06 by john30