john30/ebusd #418: Connection to ebus adapter lost until adapter or ebusd is restarted

mbisak, 2021-04-19:
Hi,
I have finally managed to configure WiFi ebus adapter with my Vaillant ecoTEC and home assistant using MQTT. In home assistant I register like 10 Vaillant variables to be polled automatically. Some are registered as ?1 polls, some as ?9 polls. It runs fine for few hours only and then it hangs. To be precise, ebusd stops retrieving polled variables from ebus adapter/ebus. Ebus adapter is still alive, connected to the network, sending 0xaa(SYN) and 0xc6(CRC) bytes, TCP stack of the "ebusd machine" sends ACK to these packets but there is no more information from ebusd side.
Diggind deeper into the issue it turns out that there is some intermittent network issue or ebus adapter issue which causes TCP retransmits. After these retransmits ebus adapter is only sending 0xaa and 0xc6, ebusd ACKs these messages and that is all, what happens further.

Snapshot of the ebusd log:
2021-04-19 12:06:36.030 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1333, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:06:36.030 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:06:38.032 [mqtt debug] publish ebusd/bai/WP off
2021-04-19 12:06:38.032 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1334, 'ebusd/bai/WP', ... (3 bytes))
2021-04-19 12:06:38.033 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/bai/WP', ... (3 bytes))
2021-04-19 12:06:42.710 [main debug] performing regular tasks
2021-04-19 12:06:43.070 [bus info] poll cmd: 3108b509030d5500
2021-04-19 12:06:43.070 [bus debug] start request 31
2021-04-19 12:06:43.070 [bus debug] arbitration start with 31
**2021-04-19 12:06:44.044 [bus error] poll bai HwcWaterflow failed: ERR: no signal
2021-04-19 12:06:44.044 [bus debug] ERR: read timeout during skip, switching to no signal
2021-04-19 12:06:44.044 [bus error] signal lost
2021-04-19 12:06:44.447 [bus debug] ERR: SYN received during no signal, switching to ready
2021-04-19 12:06:44.447 [bus notice] signal acquired**
2021-04-19 12:06:44.487 [bus error] device status: unexpected available enhanced byte 2
2021-04-19 12:06:52.046 [mqtt debug] publish ebusd/global/uptime 8945
2021-04-19 12:06:52.046 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1335, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:06:52.047 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:06:52.711 [main debug] performing regular tasks
2021-04-19 12:07:02.712 [main debug] performing regular tasks
2021-04-19 12:07:08.063 [mqtt debug] publish ebusd/global/uptime 8961
2021-04-19 12:07:08.063 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1336, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:07:08.064 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))
...
...
...
...
2021-04-19 12:07:24.079 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1337, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:07:24.079 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:07:32.714 [main debug] performing regular tasks
2021-04-19 12:07:35.034 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-04-19 12:07:40.093 [mqtt debug] publish ebusd/global/uptime 8993
2021-04-19 12:07:40.093 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1338, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:07:40.094 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:07:40.418 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-04-19 12:07:42.715 [main debug] performing regular tasks
2021-04-19 12:07:45.848 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-04-19 12:07:51.236 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-04-19 12:07:52.716 [main debug] performing regular tasks
2021-04-19 12:07:56.107 [mqtt debug] publish ebusd/global/uptime 9009
2021-04-19 12:07:56.107 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1339, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:07:56.107 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:08:02.716 [main debug] performing regular tasks
2021-04-19 12:08:12.123 [mqtt debug] publish ebusd/global/uptime 9025
2021-04-19 12:08:12.123 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1340, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:08:12.124 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:08:12.717 [main debug] performing regular tasks
2021-04-19 12:08:14.550 [bus error] device status: unexpected available enhanced byte 2
2021-04-19 12:08:22.718 [main debug] performing regular tasks
2021-04-19 12:08:28.133 [mqtt debug] publish ebusd/global/uptime 9041
2021-04-19 12:08:28.133 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1341, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:08:28.134 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:08:32.719 [main debug] performing regular tasks
2021-04-19 12:08:42.719 [main debug] performing regular tasks
2021-04-19 12:08:44.148 [mqtt debug] publish ebusd/global/uptime 9057
2021-04-19 12:08:44.148 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1342, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:08:44.149 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:08:52.720 [main debug] performing regular tasks
2021-04-19 12:09:00.162 [mqtt debug] publish ebusd/global/uptime 9073
2021-04-19 12:09:00.162 [mqtt debug] log Client ebusd_21.2_903 sending PUBLISH (d0, q0, r0, m1343, 'ebusd/global/uptime', ... (4 bytes))
2021-04-19 12:09:00.163 [mqtt debug] log Client ebusd_21.2_903 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (4 bytes))

[image]

mbisak, 2021-04-19:
If needed I can upload pcap file as well.

john30, 2021-10-11:
is it possible that the Wemos has issues with WiFi stability?

mbisak, 2021-10-11:
I don't think so. The Wemos has external antenna and AP is 3 meters away from Wemos...

alphachris, 2021-11-06:
> so is this an adapter 3 or something else? if adapter 3, did you set the jumpers correctly? your mentioned CRC in the first post isn't a CRC but part of the [enhanced protocol](https://github.com/john30/ebusd/blob/master/docs/enhanced_proto.md)

in my case it's an adapter 3, yes. jumpers are set appropriately according to the guide (https://adapter.ebusd.eu/img/smd-wifi.jpg). can't speak for @mbisak

john30, 2021-11-06:
which versions do you use? ebusd and abusd-esp
please provide some ebusd logging

alphachris, 2021-11-06:
ebusd = latest docker release (= ebusd 21.2.v21.2-48-g24a6525)
ebusd-esp = build 20201122

the error happens after some time, there is no indication/event like WiFi disconnect before the occurence. somehow the connection between ebusd and the Adapter seems to get stuck and remains in an unresponsive state - reading or writing to the bus is not possible then and the custom defined priority polls are stopped being polled from that point on. note that if the problem occurs, issuing "ebusctl info" within the docker container becomes unresponsive aswell while the log still continues. not every restart of the ebusd container fixes the problem. sometimes 2-3 attempts are needed to make it work again. I've attached the log from 2 hours of today, since the error itself may not be sufficient. at 15:38 the container has been restarted, since then sending and reading from the bus has been working again without an issue (by the time of writing).

[ebusd.log](https://github.com/john30/ebusd/files/7491307/ebusd.log)

mbisak, 2021-11-07:
> 
> 
> which versions do you use? ebusd and abusd-esp please provide some ebusd logging

I use adapter version 3, ebusd-esp HW, firmware for Wemos D1 mini Pro. configured as Adapter 3 Rx+Tx, ebusd 21.2.v21.2, jumpers set as on the pictures.

BR, Martin 

localhost: i
version: ebusd 21.2.v21.2
update check: OK, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.bai.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available
signal: acquired
symbol rate: 23
max symbol rate: 101
min arbitration micros: 1
max arbitration micros: 92
min symbol latency: 9
max symbol latency: 39
reconnects: 0
masters: 2
messages: 213
conditional: 3
poll: 23
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0105;HW=7803", loaded "vaillant/bai.0010021877.inc" ([PROD='0010021877']), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
[image]
[image]

john30, 2021-11-16:
please try with a build from the latest sources

mbisak, 2021-11-17:
Hi, seems fixed with new version. I have also upgraded ebusd virtual machine to Debian 11. It works fine since yesterday evening. 
I will update the case in few days.

Martin

alphachris, 2021-11-17:
> 
> 
> please try with a build from the latest sources

are you referring to ebusd or ebusd-esp?

john30, 2021-11-20:
> are you referring to ebusd or ebusd-esp?

the code in this repo

mbisak, 2021-11-23:
Failed again:
2021-11-23 08:51:18.200 [update notice] sent poll-read bai ModulationTempDesired QQ=31: 28.0
2021-11-23 08:51:18.201 [bus debug] notify request: done
2021-11-23 08:51:18.201 [bus debug] switching from send response ACK to send SYN
2021-11-23 08:51:18.213 [bus debug] send/receive symbol latency 12 ms
2021-11-23 08:51:19.704 [main debug] performing regular tasks
2021-11-23 08:51:19.933 [mqtt debug] publish ebusd/bai/ModulationTempDesired 28.0
2021-11-23 08:51:19.933 [mqtt debug] log Client ebusd_21.3_2779 sending PUBLISH (d0, q0, r0, m5140, 'ebusd/bai/M
odulationTempDesired', ... (4 bytes))
2021-11-23 08:51:19.934 [mqtt debug] log Client ebusd_21.3_2779 received PUBLISH (d0, q0, r0, m0, 'ebusd/bai/Mod
ulationTempDesired', ... (4 bytes))
2021-11-23 08:51:25.938 [mqtt debug] publish ebusd/global/uptime 34324
2021-11-23 08:51:25.938 [mqtt debug] log Client ebusd_21.3_2779 sending PUBLISH (d0, q0, r0, m5141, 'ebusd/globa
l/uptime', ... (5 bytes))
2021-11-23 08:51:25.939 [mqtt debug] log Client ebusd_21.3_2779 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/
uptime', ... (5 bytes))
2021-11-23 08:51:29.054 [bus info] poll cmd: 3108b509030d9800
2021-11-23 08:51:29.054 [bus debug] start request 31
2021-11-23 08:51:29.054 [bus debug] arbitration start with 31
2021-11-23 08:51:29.704 [main debug] performing regular tasks
2021-11-23 08:51:30.026 [bus error] device status: unexpected available enhanced byte 2
2021-11-23 08:51:30.042 [bus debug] arbitration won
2021-11-23 08:51:30.042 [bus debug] arbitration delay 2 micros
2021-11-23 08:51:30.042 [bus debug] switching from ready to send command
2021-11-23 08:51:30.053 [bus debug] send/receive symbol latency 10 ms
2021-11-23 08:51:30.063 [bus debug] send/receive symbol latency 10 ms
2021-11-23 08:51:30.074 [bus debug] send/receive symbol latency 10 ms
2021-11-23 08:51:30.084 [bus debug] send/receive symbol latency 9 ms
2021-11-23 08:51:30.134 [bus debug] notify request: ERR: read timeout
2021-11-23 08:51:30.134 [bus error] poll bai ReturnTemp failed: ERR: read timeout
2021-11-23 08:51:30.134 [bus debug] ERR: read timeout during send command, switching to skip
2021-11-23 08:51:32.048 [bus debug] ERR: read timeout during skip, switching to no signal
2021-11-23 08:51:32.048 [bus error] signal lost
2021-11-23 08:51:38.090 [bus notice] signal acquired
2021-11-23 08:51:39.022 [bus notice] max. symbols per second: 153
2021-11-23 08:51:39.704 [main debug] performing regular tasks
2021-11-23 08:51:40.039 [bus info] poll cmd: 3108b509030d0500
2021-11-23 08:51:40.039 [bus debug] start request 31
2021-11-23 08:51:40.039 [bus debug] arbitration start with 31
2021-11-23 08:51:40.132 [bus debug] arbitration won
2021-11-23 08:51:40.132 [bus debug] arbitration delay 3 micros
2021-11-23 08:51:40.132 [bus debug] switching from ready to send command
2021-11-23 08:51:40.132 [bus debug] notify request: ERR: SYN received
2021-11-23 08:51:40.132 [bus error] poll bai Flame failed: ERR: SYN received
2021-11-23 08:51:40.132 [bus debug] ERR: SYN received during send command, switching to ready
2021-11-23 08:51:40.186 [bus debug] ERR: SYN received during receive command, switching to ready
2021-11-23 08:51:41.954 [mqtt debug] publish ebusd/global/uptime 34340
2021-11-23 08:51:41.954 [mqtt debug] log Client ebusd_21.3_2779 sending PUBLISH (d0, q0, r0, m5142, 'ebusd/global/uptime', ... (5 bytes))
2021-11-23 08:51:41.954 [mqtt debug] log Client ebusd_21.3_2779 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2021-11-23 08:51:42.841 [bus error] device status: unexpected available enhanced byte 2
2021-11-23 08:51:49.705 [main debug] performing regular tasks
2021-11-23 08:51:51.052 [bus info] poll cmd: 3108b509030d0200
2021-11-23 08:51:51.053 [bus debug] start request 31
2021-11-23 08:51:51.053 [bus debug] arbitration start with 31
2021-11-23 08:51:51.105 [bus debug] arbitration won
2021-11-23 08:51:51.105 [bus debug] arbitration delay 3 micros
2021-11-23 08:51:51.105 [bus debug] switching from ready to send command
2021-11-23 08:51:51.122 [bus debug] send/receive symbol latency 16 ms
2021-11-23 08:51:51.132 [bus debug] send/receive symbol latency 10 ms
2021-11-23 08:51:51.142 [bus debug] send/receive symbol latency 9 ms
2021-11-23 08:51:51.151 [bus debug] send/receive symbol latency 9 ms
2021-11-23 08:51:51.162 [bus debug] send/receive symbol latency 10 ms
2021-11-23 08:51:51.176 [bus debug] send/receive symbol latency 13 ms
2021-11-23 08:51:51.185 [bus debug] send/receive symbol latency 9 ms
2021-11-23 08:51:51.186 [bus debug] switching from send command to send command CRC
2021-11-23 08:51:51.196 [bus debug] send/receive symbol latency 10 ms
2021-11-23 08:51:51.196 [bus debug] switching from send command CRC to receive command ACK
2021-11-23 08:51:51.200 [bus debug] switching from receive command ACK to receive response
2021-11-23 08:51:51.224 [bus debug] switching from receive response to receive response CRC
2021-11-23 08:51:51.227 [bus debug] switching from receive response CRC to send response ACK
[... 16 lines cut ...]
2021-11-23 08:52:02.614 [bus debug] arbitration won
2021-11-23 08:52:02.614 [bus debug] arbitration delay 7 micros
2021-11-23 08:52:02.614 [bus debug] switching from ready to send command
2021-11-23 08:52:02.614 [bus debug] notify request: ERR: SYN received
2021-11-23 08:52:02.614 [bus error] poll bai GasvalveUC failed: ERR: SYN received
2021-11-23 08:52:02.614 [bus debug] ERR: SYN received during send command, switching to ready
[... cut ...]
2021-11-23 08:52:13.409 [bus debug] arbitration delay 2 micros
2021-11-23 08:52:13.409 [bus debug] switching from ready to send command
2021-11-23 08:52:13.416 [bus debug] notify request: ERR: SYN received
2021-11-23 08:52:13.416 [bus error] poll bai WP failed: ERR: SYN received
2021-11-23 08:52:13.416 [bus debug] ERR: SYN received during send command, switching to ready
2021-11-23 08:52:13.474 [bus debug] ERR: SYN received during receive command, switching to ready
2021-11-23 08:52:13.986 [mqtt debug] publish ebusd/global/uptime 34372
...

john30, 2021-11-25:
does resetting the wemos with the reset switch solve it?

mbisak, 2021-11-26:
Seems that it can solve the issue:
2021-11-26 16:18:15.140 [mqtt debug] log Client ebusd_21.3_58 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/up
time', ... (6 bytes))
2021-11-26 16:18:20.769 [main debug] performing regular tasks
2021-11-26 16:18:30.770 [main debug] performing regular tasks
2021-11-26 16:18:31.156 [mqtt debug] publish ebusd/global/uptime 152586
2021-11-26 16:18:31.156 [mqtt debug] log Client ebusd_21.3_58 sending PUBLISH (d0, q0, r0, m9850, 'ebusd/global/
uptime', ... (6 bytes))
2021-11-26 16:18:31.156 [mqtt debug] log Client ebusd_21.3_58 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/up
time', ... (6 bytes))
2021-11-26 16:18:40.770 [main debug] performing regular tasks
2021-11-26 16:18:43.101 [bus debug] ERR: read timeout during skip, switching to no signal
2021-11-26 16:18:43.101 [bus error] signal lost
2021-11-26 16:18:43.251 [bus debug] ERR: SYN received during no signal, switching to ready
2021-11-26 16:18:43.251 [bus notice] signal acquired
2021-11-26 16:18:43.256 [bus error] device status: unexpected available enhanced byte 2
2021-11-26 16:18:47.171 [mqtt debug] publish ebusd/global/uptime 152602
2021-11-26 16:18:47.171 [mqtt debug] log Client ebusd_21.3_58 sending PUBLISH (d0, q0, r0, m9851, 'ebusd/global/
uptime', ... (6 bytes))
2021-11-26 16:18:47.171 [mqtt debug] log Client ebusd_21.3_58 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/up
time', ... (6 bytes))
2021-11-26 16:18:50.771 [main debug] performing regular tasks
2021-11-26 16:19:00.771 [main debug] performing regular tasks
2021-11-26 16:19:03.186 [mqtt debug] publish ebusd/global/uptime 152618
2021-11-26 16:19:03.186 [mqtt debug] log Client ebusd_21.3_58 sending PUBLISH (d0, q0, r0, m9852, 'ebusd/global/
uptime', ... (6 bytes))
2021-11-26 16:19:03.187 [mqtt debug] log Client ebusd_21.3_58 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/up
time', ... (6 bytes))
2021-11-26 16:19:10.772 [main debug] performing regular tasks
2021-11-26 16:19:13.078 [bus debug] ERR: read timeout during skip, switching to no signal
2021-11-26 16:19:13.078 [bus error] signal lost
2021-11-26 16:19:13.221 [bus debug] ERR: SYN received during no signal, switching to ready
2021-11-26 16:19:13.221 [bus notice] signal acquired
2021-11-26 16:19:13.227 [bus error] device status: unexpected available enhanced byte 2
2021-11-26 16:19:19.203 [mqtt debug] publish ebusd/global/uptime 152634
2021-11-26 16:19:19.203 [mqtt debug] log Client ebusd_21.3_58 sending PUBLISH (d0, q0, r0, m9853, 'ebusd/global/
uptime', ... (6 bytes))
2021-11-26 16:19:19.203 [mqtt debug] log Client ebusd_21.3_58 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (6 bytes))
2021-11-26 16:19:20.772 [main debug] performing regular tasks
2021-11-26 16:19:30.772 [main debug] performing regular tasks
2021-11-26 16:19:35.215 [mqtt debug] publish ebusd/global/uptime 152650
2021-11-26 16:19:35.215 [mqtt debug] log Client ebusd_21.3_58 sending PUBLISH (d0, q0, r0, m9854, 'ebusd/global/uptime', ... (6 bytes))
2021-11-26 16:19:35.215 [mqtt debug] log Client ebusd_21.3_58 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (6 bytes))
2021-11-26 16:19:36.068 [bus debug] ERR: read timeout during skip, switching to no signal
2021-11-26 16:19:36.068 [bus error] signal lost
2021-11-26 16:19:40.773 [main debug] performing regular tasks
2021-11-26 16:19:41.288 [bus error] arbitration start error
****2021-11-26 16:19:41.288 [bus notice] device invalid
2021-11-26 16:19:46.322 [bus notice] device status: resetting
2021-11-26 16:19:46.322 [bus notice] re-opened 192.168.x.y:9999
2021-11-26 16:19:46.325 [bus debug] ERR: SYN received during no signal, switching to ready
2021-11-26 16:19:46.325 [bus notice] signal acquired**
2021-11-26 16:19:46.325 [bus info] poll cmd: 3108b509030d2800
2021-11-26 16:19:46.325 [bus debug] start request 31
2021-11-26 16:19:46.325 [bus debug] arbitration start with 31
2021-11-26 16:19:46.373 [bus debug] arbitration won
2021-11-26 16:19:46.373 [bus debug] arbitration delay 3 micros
2021-11-26 16:19:46.373 [bus info] arbitration delay 3 - 3 micros
2021-11-26 16:19:46.373 [bus debug] switching from ready to send command
2021-11-26 16:19:46.384 [bus debug] send/receive symbol latency 10 ms
2021-11-26 16:19:46.384 [bus info] send/receive symbol latency 10 - 10 ms
2021-11-26 16:19:46.395 [bus debug] send/receive symbol latency 10 ms
2021-11-26 16:19:46.405 [bus debug] send/receive symbol latency 10 ms
2021-11-26 16:19:46.415 [bus debug] send/receive symbol latency 9 ms
2021-11-26 16:19:46.415 [bus info] send/receive symbol latency 9 - 10 ms
2021-11-26 16:19:46.425 [bus debug] send/receive symbol latency 9 ms
2021-11-26 16:19:46.442 [bus debug] send/receive symbol latency 17 ms
2021-11-26 16:19:46.442 [bus info] send/receive symbol latency 9 - 17 ms
2021-11-26 16:19:46.453 [bus debug] send/receive symbol latency 10 ms
2021-11-26 16:19:46.453 [bus debug] switching from send command to send command CRC
[... 126 lines cut ...]
2021-11-26 16:20:30.103 [bus debug] send/receive symbol latency 9 ms
2021-11-26 16:20:30.113 [bus debug] send/receive symbol latency 9 ms
2021-11-26 16:20:30.123 [bus debug] send/receive symbol latency 10 ms
2021-11-26 16:20:30.140 [bus debug] send/receive symbol latency 17 ms
2021-11-26 16:20:30.140 [bus debug] switching from send command to send command CRC
2021-11-26 16:20:30.151 [bus debug] send/receive symbol latency 10 ms
2021-11-26 16:20:30.151 [bus debug] switching from send command CRC to receive command ACK
[... cut ...]
2021-11-26 16:20:36.279 [mqtt debug] log Client ebusd_21.3_58 received PUBLISH (d0, q0, r0, m0, 'ebusd/bai/PrEnergyCountHwc1', ... (7 bytes))
2021-11-26 16:20:39.281 [mqtt debug] publish ebusd/global/uptime 152714
2021-11-26 16:20:39.281 [mqtt debug] log Client ebusd_21.3_58 sending PUBLISH (d0, q0, r0, m9863, 'ebusd/global/uptime', ... (6 bytes))
2021-11-26 16:20:39.281 [mqtt debug] log Client ebusd_21.3_58 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (6 bytes))
2021-11-26 16:20:40.776 [main debug] performing regular tasks
2021-11-26 16:20:41.037 [bus info] poll cmd: 3108b509030df600

mbisak, 2021-11-27:
after some time, it fails again

john30, 2021-11-30:
ok, need to look at the Wemos firmware then. this will take some time

mbisak, 2021-12-01:
24hrs with new ESP and PIC SW. I'm still receiving EBUS statistics, but ebusd.log contains a lot of error messages. Fortunately seems that EBUS controller or ebusd are able to recover after upgrade. Can I somehow prevent these errors?

cat /var/log/ebusd.log | grep -i err
2021-12-01 19:51:10.188 [bus error] poll bai StorageTemp failed: ERR: read timeout
2021-12-01 19:51:10.188 [bus debug] ERR: read timeout during send command, switching to skip
2021-12-01 19:51:10.290 [bus debug] ERR: SYN received during receive command, switching to ready
2021-12-01 19:51:25.652 [bus error] device status: unexpected available enhanced byte 2
2021-12-01 19:52:25.675 [bus error] device status: unexpected available enhanced byte 2
2021-12-01 19:52:55.981 [bus error] device status: unexpected available enhanced byte 2
2021-12-01 19:53:11.212 [bus debug] notify request: ERR: read timeout
2021-12-01 19:53:11.212 [bus error] poll bai FanSpeed failed: ERR: read timeout
2021-12-01 19:53:11.212 [bus debug] ERR: read timeout during send command, switching to skip
2021-12-01 19:53:11.235 [bus debug] ERR: SYN received during receive command, switching to ready
2021-12-01 19:53:55.905 [bus error] device status: unexpected available enhanced byte 2
2021-12-01 19:54:12.496 [bus error] device status: unexpected available enhanced byte 2
2021-12-01 19:54:26.087 [bus debug] ERR: read timeout during skip, switching to no signal
2021-12-01 19:54:26.087 [bus error] signal lost
2021-12-01 19:54:26.138 [bus debug] ERR: SYN received during no signal, switching to ready
2021-12-01 19:54:26.142 [bus error] device status: unexpected available enhanced byte 2
2021-12-01 19:55:14.737 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-01 19:55:20.125 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-01 19:55:26.028 [bus debug] ERR: SYN received during receive command ACK, switching to ready
2021-12-01 19:55:30.959 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-01 19:55:56.083 [bus debug] ERR: read timeout during skip, switching to no signal
2021-12-01 19:55:56.083 [bus error] signal lost
2021-12-01 19:55:56.282 [bus debug] ERR: SYN received during no signal, switching to ready
2021-12-01 19:55:56.285 [bus error] device status: unexpected available enhanced byte 2
2021-12-01 19:56:26.071 [bus debug] ERR: read timeout during skip, switching to no signal
2021-12-01 19:56:26.071 [bus error] signal lost
2021-12-01 19:56:26.134 [bus debug] ERR: SYN received during no signal, switching to ready
2021-12-01 19:56:29.175 [bus debug] notify request: ERR: read timeout
2021-12-01 19:56:29.175 [bus error] poll bai Flame failed: ERR: read timeout
2021-12-01 19:56:29.175 [bus debug] ERR: read timeout during send command, switching to skip
2021-12-01 19:56:56.385 [bus error] device status: unexpected available enhanced following byte 1
2021-12-01 19:57:26.589 [bus error] device status: unexpected available enhanced byte 2
2021-12-01 19:57:46.126 [bus error] device status: eBUS comm error: framing
2021-12-01 19:57:46.126 [bus debug] notify request: ERR: read timeout
2021-12-01 19:57:46.127 [bus error] poll bai FlowTemp failed: ERR: read timeout
2021-12-01 19:57:46.127 [bus debug] ERR: read timeout during send command, switching to skip
2021-12-01 19:57:57.242 [bus debug] notify request: ERR: read timeout
2021-12-01 19:57:57.242 [bus error] poll bai FanSpeed failed: ERR: read timeout
2021-12-01 19:57:57.242 [bus debug] ERR: read timeout during send command, switching to skip

john30, 2021-12-05:
please update the esp firmware to the latest release from today

mbisak, 2021-12-07:
upgraded immediately and from that time everything works fine.

mbisak, 2021-12-08:
Unfortunately, it failed today:

2021-12-08 07:05:59.144 [bus debug] send/receive symbol latency 9 ms
2021-12-08 07:05:59.144 [bus debug] switching from send command CRC to receive command ACK
2021-12-08 07:05:59.150 [bus debug] switching from receive command ACK to receive response
2021-12-08 07:05:59.176 [bus debug] switching from receive response to receive response CRC
2021-12-08 07:05:59.181 [bus debug] switching from receive response CRC to send response ACK
2021-12-08 07:05:59.194 [bus debug] send/receive symbol latency 12 ms
2021-12-08 07:05:59.194 [update info] sent MS cmd: 3108b509030d1800 / 03660300
2021-12-08 07:05:59.194 [update notice] sent poll-read bai FlowTemp QQ=31: 54.38;ok
2021-12-08 07:05:59.194 [bus debug] notify request: done
2021-12-08 07:05:59.194 [bus debug] switching from send response ACK to send SYN
2021-12-08 07:05:59.208 [bus debug] send/receive symbol latency 13 ms
2021-12-08 07:06:00.056 [main debug] performing regular tasks
2021-12-08 07:06:00.767 [mqtt debug] publish ebusd/bai/FlowTemp 54.38;ok
2021-12-08 07:06:00.767 [mqtt debug] log Client ebusd_21.3_1154 sending PUBLISH (d0, q0, r0, m37296, 'ebusd/bai/FlowTemp', ... (8 bytes))
2021-12-08 07:06:00.767 [mqtt debug] log Client ebusd_21.3_1154 received PUBLISH (d0, q0, r0, m0, 'ebusd/bai/FlowTemp', ... (8 bytes))
2021-12-08 07:06:01.768 [mqtt debug] publish ebusd/global/uptime 252546
2021-12-08 07:06:01.768 [mqtt debug] log Client ebusd_21.3_1154 sending PUBLISH (d0, q0, r0, m37297, 'ebusd/global/uptime', ... (6 bytes))
2021-12-08 07:06:01.769 [mqtt debug] log Client ebusd_21.3_1154 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (6 bytes))
2021-12-08 07:06:10.038 [bus info] poll cmd: 3108b509030d8300
2021-12-08 07:06:10.038 [bus debug] start request 31
2021-12-08 07:06:10.038 [bus debug] arbitration start with 31
2021-12-08 07:06:10.056 [main debug] performing regular tasks
2021-12-08 07:06:11.041 [bus error] poll bai FanSpeed failed: ERR: no signal
2021-12-08 07:06:11.041 [bus debug] ERR: read timeout during skip, switching to no signal
2021-12-08 07:06:11.041 [bus error] signal lost
2021-12-08 07:06:11.279 [bus debug] ERR: SYN received during no signal, switching to ready
2021-12-08 07:06:11.279 [bus notice] signal acquired
2021-12-08 07:06:11.283 [bus error] device status: unexpected available enhanced byte 2
2021-12-08 07:06:17.785 [mqtt debug] publish ebusd/global/uptime 252562
2021-12-08 07:06:17.785 [mqtt debug] log Client ebusd_21.3_1154 sending PUBLISH (d0, q0, r0, m37298, 'ebusd/global/uptime', ... (6 bytes))
2021-12-08 07:06:17.785 [mqtt debug] log Client ebusd_21.3_1154 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (6 bytes))
2021-12-08 07:06:20.057 [main debug] performing regular tasks
2021-12-08 07:06:21.180 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-08 07:06:26.567 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-08 07:06:30.057 [main debug] performing regular tasks
2021-12-08 07:06:31.954 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-12-08 07:06:33.800 [mqtt debug] publish ebusd/global/uptime 252578
2021-12-08 07:06:33.800 [mqtt debug] log Client ebusd_21.3_1154 sending PUBLISH (d0, q0, r0, m37299, 'ebusd/global/uptime', ... (6 bytes))
2021-12-08 07:06:33.800 [mqtt debug] log Client ebusd_21.3_1154 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (6 bytes))
:
--------------------------------------------
Build: 20211205
ebusd device string: enh:192.168.2.5:9999
WiFi signal: 76% (-62dBm)
OTA enabled: waiting

ebus firmware :
Chip ID: 00e45c62
Hostname: ebus
Up time: 230094
Free heap: 29024
ebusd connected: yes (inactive)
eBUS signal: acquired
eBUS initial signal: yes 

M.

mbisak, 2021-12-08:
Restart of ebusd does not help.

Ebus status shows after restart:
Chip ID: 00e45c62
Hostname: ebus
Up time: 234042
Free heap: 28728
ebusd connected: no
eBUS signal: no signal
eBUS initial signal: yes 

soft reset of the esp module nor ebusd reset does not help. ESP hw reset didn't help as well. Ebus module needed reset (power cycle)

john30, 2021-12-12:
how does your last post correlate with the one before? in the one before the esp status says signal: acquired and in the second it doesn't

mbisak, 2021-12-12:
Whenever communication fails, I try these steps to restore it:
1) ebusd daemon restart
2) esp restart
3) module power cycle
With the first version of SW (from the beginning of 2021 the communication error could be solved just by restarting ebusd daemon. With the newest SW (PIC, ebusd, esp) when communication fails only module power cycle helps. On the other hand communication fails less often than before. Unfortunately power cycle is not easy to perform remotely:-(
To your question:
in this case I have tried to restart ebusd daemon because of communication error and the two outputs are before and after ebusd restart.

Martin

john30, 2021-12-15:
what power supply are you using?

mbisak, 2022-01-01:
[image]

JaVier-Lloret, 2022-01-04:
I have the same problem. I am using the docker edusd and ebusd-devel, with similar result.
My wemos interface has the build 20211226 and with build 20211030 looked better.
With build 20211030 usualy with restart the docker, was suficient
Can I aport some thing?

john30, 2022-01-26:
do you see the same behaviour when using usb instead of wifi?

JaVier-Lloret, 2022-01-27:
HiI am using wifi, with an excelent signal. Now I can't connect by USB my computer is far.Do you know how far it works the EBUS bus. I have 12 meters betwen boiler to computer.Now I am using ebusd v21, when I start with this version all was good but 2 dais after had the same behaviour.Last monday i recibed a new "wemos pro 16" and will test with him.I wait for your response.Thanks JaVier LloretEnviado desde mi Galaxy
-------- Mensaje original --------De: John ***@***.***> Fecha: 26/1/22  20:09  (GMT+01:00) Para: john30/ebusd ***@***.***> Cc: JaVier-Lloret ***@***.***>, Comment ***@***.***> Asunto: Re: [john30/ebusd] Connection to ebus adapter lost until adapter or ebusd is restarted (#418) 
do you see the same behaviour when using usb instead of wifi?

JaVier-Lloret, 2022-01-27:
I have moved my Ebus adapter near the computer losing the connection to the boiler.
For the analysis we are doing I think it is not necessary.

I have connected the adapter to a usb of the computer that runs docker and after identifying the tty port by looking through it, I have executed the following command as indicated in the instructions on the web.
*********
docker run --rm -it --device=/dev/ttyUSB1:/dev/ttyUSB0 -p 8888 john30/ebusd -f --scanconfig --latency=20 --mqttport=1883 --mqtthost=192.168.100.XXX --logfile=/var/log/ebusd.log --loglevel=debug
*********
and afterwards, I received the same result. No "/ebusd/global/" messages are sent to the mqtt server.

Copy the boot trace here:
**************************
root@MyServer:/dev# docker run --rm -it --device=/dev/ttyUSB1:/dev/ttyUSB0 -p 8888 john30/ebusd -f --scanconfig --latency=20 --mqttport=1883 --mqtthost=192.168.100.5 --logfile=/var/log/ebusd.log --loglevel=debug
2022-01-27 11:45:19.648 [main notice] ebusd 21.3.v21.3 started with auto scan on device /dev/ttyUSB0
2022-01-27 11:45:19.649 [main info] loading configuration files from http://ebusd.eu/config/
2022-01-27 11:45:24.774 [main info] reading templates /
2022-01-27 11:45:29.900 [main info] read templates in /
2022-01-27 11:45:29.900 [main info] reading file memory.csv
2022-01-27 11:45:35.161 [main info] successfully read file memory.csv
2022-01-27 11:45:35.161 [main info] reading file broadcast.csv
2022-01-27 11:45:40.264 [main info] successfully read file broadcast.csv
2022-01-27 11:45:40.264 [main info] read config files
2022-01-27 11:45:40.265 [bus notice] bus started with own address 31/36
2022-01-27 11:45:40.265 [main info] registering data handlers
2022-01-27 11:45:40.265 [mqtt info] mosquitto version 2.0.11 (compiled with 2.0.11)
2022-01-27 11:45:40.266 [mqtt debug] connection requested
2022-01-27 11:45:40.266 [main info] registered data handlers
2022-01-27 11:45:40.266 [mqtt notice] connection established
2022-01-27 11:45:40.266 [mqtt debug] publish ebusd/global/version ebusd 21.3.v21.3
2022-01-27 11:45:40.266 [mqtt debug] publish ebusd/global/running true
2022-01-27 11:45:50.266 [main debug] performing regular tasks
2022-01-27 11:45:50.266 [mqtt debug] publish ebusd/global/scan finished
2022-01-27 11:45:56.272 [mqtt debug] publish ebusd/global/uptime 16
2022-01-27 11:46:00.266 [main debug] performing regular tasks
2022-01-27 11:46:10.267 [main debug] performing regular tasks
2022-01-27 11:46:12.287 [mqtt debug] publish ebusd/global/uptime 32
2022-01-27 11:46:20.267 [main debug] performing regular tasks
2022-01-27 11:46:28.302 [mqtt debug] publish ebusd/global/uptime 48
2022-01-27 11:46:30.267 [main debug] performing regular tasks
**************************

thanks for your time

JaVier-Lloret, 2022-01-27:
I had a test with tcpdump and it showed this:
Every 16 seconds eBusd sends global/uptime but the server doesn't understand the message.
I attached part of the 
[EBusd-comunication.txt](https://github.com/john30/ebusd/files/7952039/EBusd-comunication.txt)
file.

john30, 2022-01-29:
what do you mean by "server doesn't understand"? this is MQTT and if your broker doesn't accept the message, it is probably a configuration issue on the broker or in the ebusd MQTT settings.
is this related to an adapter 3 at all @JaVier-Lloret ? if not, then please don't misuse this ticket here

JaVier-Lloret, 2022-01-29:
Hi John.
Maybe I explained myself wrong, I start at the beginning.
Of course I have an Ebus 3 adapter and through an eBusd running in a docker it communicates, or tries to, with an mqtt server. Is there another thread that only affects the eBusd service?
This MQTT server communicates with other equipment, none of them with problems. It is a mosquitto server that provides Home Assistant and I check the traffic with an MQTT Explorer and for the rest of the equipment there are no problems. In case you are wondering, I use neither username nor password for its operation.
Once explained this I tell you:
The communications between the Ebus3 adapter and the boiler seem to work correctly, I don't know all the commands to be emphatic in the assessment.
The communications between the Ebus3 adapter and the eBusd driver also seem to be correct as it correctly scans the eBus bus and responds correctly to each ebusctl question, and even with the readall.sh script I get all the values ​​corresponding to my boiler. I have tried to do some set and it has not worked for me but for me this is a lesser evil and that we can look at later.
The problem appears on the other side of the Ebusd driver, where the traffic is sent to MQTT.
Why do I say that he does not understand the messages?
Initially, looking at the logs, ebusd correctly locates the MQTT server and starts sending "ebusd/global/uptime" frames and other global parameters. Usually these messages are not received (understood) by the MQTT server. Could there be a problem in the formation of the MQTT packets? Last week I analyzed the traffic between the ebusd and the MQTT server and I noticed, you can see in the attached file ("EBusd-cominication.txt") that in all the mqtt packets a checksum error indication appears.
The fact is that sometimes the communication is correct and this state lasts one or two days in the best of cases. To later block and stop communicating and not recover communication even with an eBusd restart.
In the ebusd MQTT configuration I have tried multiple variations. In the documentation there is no mention of values ​​for parameters like "mqttversion" and some others.
Next week I will compare the MQTT traffic between other devices and the MQTT server and tell you.
I have some more evidence that can give me more information.
Thanks

JaVier-Lloret, 2022-02-11:
Hi all
I have delayed in publishing my conclusions to be more sure of what has happened.

Thanks John for your help.

I have verified that when the docker starts and it has to dump its results on an MQTT server, it is mandatory to put an mqttuser and an mqttpass, otherwise sometimes it works and sometimes it doesn't and that misleads us. For other equipment that I have it was not necessary.

So now you know the solution to my problem.

Thank you all.

mbisak, 2022-04-03:
Ebusd 22.2, latest esp and pic software and still lost connection after some time. From the error indicated below no info is read from EBUS.

2022-04-02 04:01:35.015 [bus info] poll cmd: 3108b509030d5800
2022-04-02 04:01:35.015 [bus debug] start request 31
2022-04-02 04:01:35.015 [bus debug] arbitration start with 31
2022-04-02 04:01:36.026 [bus error] poll bai HwcDemand failed: ERR: no signal
2022-04-02 04:01:36.027 [bus debug] ERR: read timeout during skip, switching to no signal
**2022-04-02 04:01:36.027 [bus error] signal lost**
2022-04-02 04:01:37.405 [bus debug] ERR: SYN received during no signal, switching to ready
2022-04-02 04:01:37.405 [bus notice] signal acquired
2022-04-02 04:01:37.414 [bus debug] ERR: SYN received during receive command, switching to ready
2022-04-02 04:01:42.313 [main debug] performing regular tasks
2022-04-02 04:01:46.239 [mqtt debug] publish ebusd/global/uptime 51414
2022-04-02 04:01:46.239 [mqtt debug] log Client ebusd_22.2_677 sending PUBLISH (d0, q0, r0, m7454, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:01:46.239 [mqtt debug] log Client ebusd_22.2_677 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:01:52.313 [main debug] performing regular tasks
2022-04-02 04:02:02.256 [mqtt debug] publish ebusd/global/uptime 51430
2022-04-02 04:02:02.256 [mqtt debug] log Client ebusd_22.2_677 sending PUBLISH (d0, q0, r0, m7455, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:02:02.256 [mqtt debug] log Client ebusd_22.2_677 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:02:02.314 [main debug] performing regular tasks
2022-04-02 04:02:06.053 [bus debug] ERR: read timeout during skip, switching to no signal
2022-04-02 04:02:06.053 [bus error] signal lost
2022-04-02 04:02:06.254 [bus debug] ERR: SYN received during no signal, switching to ready
2022-04-02 04:02:06.254 [bus notice] signal acquired
2022-04-02 04:02:06.258 [bus error] device status: unexpected available enhanced byte 2
2022-04-02 04:02:12.314 [main debug] performing regular tasks
2022-04-02 04:02:18.271 [mqtt debug] publish ebusd/global/uptime 51446
2022-04-02 04:02:18.271 [mqtt debug] log Client ebusd_22.2_677 sending PUBLISH (d0, q0, r0, m7456, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:02:18.271 [mqtt debug] log Client ebusd_22.2_677 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:02:22.315 [main debug] performing regular tasks
2022-04-02 04:02:32.315 [main debug] performing regular tasks
2022-04-02 04:02:34.286 [mqtt debug] publish ebusd/global/uptime 51462
2022-04-02 04:02:34.286 [mqtt debug] log Client ebusd_22.2_677 sending PUBLISH (d0, q0, r0, m7457, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:02:34.286 [mqtt debug] log Client ebusd_22.2_677 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:02:42.316 [main debug] performing regular tasks
2022-04-02 04:02:50.300 [mqtt debug] publish ebusd/global/uptime 51478
2022-04-02 04:02:50.300 [mqtt debug] log Client ebusd_22.2_677 sending PUBLISH (d0, q0, r0, m7458, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:02:50.300 [mqtt debug] log Client ebusd_22.2_677 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:02:52.316 [main debug] performing regular tasks
2022-04-02 04:03:02.317 [main debug] performing regular tasks
2022-04-02 04:03:06.314 [mqtt debug] publish ebusd/global/uptime 51494
2022-04-02 04:03:06.314 [mqtt debug] log Client ebusd_22.2_677 sending PUBLISH (d0, q0, r0, m7459, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:03:06.314 [mqtt debug] log Client ebusd_22.2_677 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:03:06.505 [bus error] device status: unexpected available enhanced byte 2
2022-04-02 04:03:12.317 [main debug] performing regular tasks
2022-04-02 04:03:22.318 [main debug] performing regular tasks
2022-04-02 04:03:22.330 [mqtt debug] publish ebusd/global/uptime 51510
2022-04-02 04:03:22.330 [mqtt debug] log Client ebusd_22.2_677 sending PUBLISH (d0, q0, r0, m7460, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:03:22.330 [mqtt debug] log Client ebusd_22.2_677 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:03:32.318 [main debug] performing regular tasks
2022-04-02 04:03:38.344 [mqtt debug] publish ebusd/global/uptime 51526
2022-04-02 04:03:38.344 [mqtt debug] log Client ebusd_22.2_677 sending PUBLISH (d0, q0, r0, m7461, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:03:38.344 [mqtt debug] log Client ebusd_22.2_677 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:03:42.319 [main debug] performing regular tasks
2022-04-02 04:03:52.319 [main debug] performing regular tasks
2022-04-02 04:03:54.361 [mqtt debug] publish ebusd/global/uptime 51542
2022-04-02 04:03:54.361 [mqtt debug] log Client ebusd_22.2_677 sending PUBLISH (d0, q0, r0, m7462, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:03:54.361 [mqtt debug] log Client ebusd_22.2_677 received PUBLISH (d0, q0, r0, m0, 'ebusd/global/uptime', ... (5 bytes))
2022-04-02 04:04:02.320 [main debug] performing regular tasks
2022-04-02 04:04:06.701 [bus error] device status: unexpected available enhanced byte 2
2022-04-02 04:04:10.379 [mqtt debug] publish ebusd/global/uptime 51558
[... cut ...]
2022-04-03 14:40:13.789 [bus error] arbitration start error
2022-04-03 14:40:13.789 [bus debug] ERR: arbitration lost during ready, retry
2022-04-03 14:40:13.789 [bus debug] start request 31
2022-04-03 14:40:13.789 [bus debug] arbitration start with 31
2022-04-03 14:40:13.805 [bus debug] arbitration won
2022-04-03 14:40:13.805 [bus debug] arbitration delay 2 micros
2022-04-03 14:40:13.805 [bus info] arbitration delay 2 - 2 micros
2022-04-03 14:40:13.805 [bus debug] switching from ready to send command****

mbisak, 2022-06-28:
This issue seems to be resolved with the latest release of the ebusd. I will monitor it for a longer time.

Martin

john30, 2022-09-17:
thanks, closing

closed 2022-09-17 by john30