john30/ebusd #419: connection lost after few minutes (device invalid)

wouldsmina, 2021-04-20:
Hi,
When I start ebusd, the signal is acquired, but after few minutes I get some errors.

Boiler : Saunier Duval F25E with Exacontrol E7

```
2021-04-20 14:15:19.430 [main notice] ebusd 21.2.v21.2 started with auto scan on device tcp:192.168.0.3:9999
2021-04-20 14:15:29.729 [bus notice] bus started with own address 31/36
2021-04-20 14:15:29.755 [bus notice] signal acquired
2021-04-20 14:17:39.834 [main notice] update check: OK
2021-04-20 14:19:41.442 [bus notice] device invalid
2021-04-20 14:19:46.470 [bus notice] re-opened tcp:192.168.0.3:9999
2021-04-20 14:20:59.351 [bus notice] device invalid
2021-04-20 14:21:04.379 [bus notice] re-opened tcp:192.168.0.3:9999
2021-04-20 14:21:05.250 [bus notice] device invalid
2021-04-20 14:21:10.278 [bus notice] re-opened tcp:192.168.0.3:9999
2021-04-20 14:21:11.645 [bus notice] device invalid
2021-04-20 14:21:16.673 [bus notice] re-opened tcp:192.168.0.3:9999
2021-04-20 14:21:17.534 [bus notice] device invalid
...
2021-04-20 14:22:17.899 [main notice] ebusd stopped
2021-04-20 14:22:22.749 [main info] successfully read file memory.csv
2021-04-20 14:22:22.749 [main info] reading file broadcast.csv
2021-04-20 14:22:36.845 [main info] successfully read file broadcast.csv
2021-04-20 14:22:36.845 [main info] read config files
2021-04-20 14:22:36.850 [bus error] unable to open tcp:192.168.0.3:9999: ERR: generic I/O error
2021-04-20 14:22:36.851 [bus notice] bus started with own address 31/36
2021-04-20 14:22:36.851 [bus notice] device invalid
2021-04-20 14:22:36.851 [main info] registering data handlers
2021-04-20 14:22:36.851 [main info] registered data handlers
2021-04-20 14:22:42.883 [bus error] unable to open tcp:192.168.0.3:9999: ERR: generic I/O error
2021-04-20 14:22:42.883 [bus notice] device invalid
2021-04-20 14:22:46.852 [main debug] performing regular tasks
2021-04-20 14:22:47.886 [bus error] unable to open tcp:192.168.0.3:9999: ERR: generic I/O error
2021-04-20 14:22:47.886 [bus notice] device invalid
```

I need to reset the adapter and restart ebusd to get new connection.

When I add  --enablehex --lograwdata and the signal acquired, I get this logs : 
```
2021-04-20 14:27:54.071 [bus notice] <c6
2021-04-20 14:27:54.111 [bus notice] <c6
2021-04-20 14:27:54.151 [bus notice] <c6
2021-04-20 14:27:54.191 [bus notice] <c6
2021-04-20 14:27:54.231 [bus notice] <c6
2021-04-20 14:27:54.270 [bus notice] <c6
2021-04-20 14:27:54.372 [bus notice] <1008c6b5100900001462c7bfc7bf01c7bf00c6af000101c69a00c6
2021-04-20 14:27:54.406 [bus notice] <c6
2021-04-20 14:27:54.445 [bus notice] <c6
2021-04-20 14:27:54.486 [bus notice] <c6
```
I can reproduce the problem when executing `ebusctl scan full` : 
```
2021-04-20 14:31:09.911 [bus notice] <c6
2021-04-20 14:31:09.951 [bus notice] <c6
2021-04-20 14:31:10.042 [bus notice] <1008c6b5110101c68900093e00c7a00900000000c7bf1100c6
2021-04-20 14:31:10.082 [bus notice] <c6
2021-04-20 14:31:10.798 [bus notice] >310207040030<c6
2021-04-20 14:31:10.802 [bus notice] <c6
2021-04-20 14:31:10.803 [bus notice] >31<c6
2021-04-20 14:31:10.808 [bus notice] device invalid
2021-04-20 14:31:15.836 [bus notice] re-opened tcp:192.168.0.3:9999
2021-04-20 14:31:15.861 [bus notice] <c6
2021-04-20 14:31:15.901 [bus notice] <c6
2021-04-20 14:31:16.699 [bus notice] >31040704006f<c6
2021-04-20 14:31:16.705 [bus notice] <c6
2021-04-20 14:31:16.707 [bus notice] >31<c6
2021-04-20 14:31:16.714 [bus notice] device invalid
```

Der-Schubi, 2021-05-06:
Are you sure there is no IP conflict? I had the exact same problem, when there was a second device in my network with the IP of the ebus adapter. When the ebus adapter answered faster than the other device everything was fine.

wouldsmina, 2021-05-20:
Yes, I'm sure, I'm using static dhcp adresses and there is no ping when the adapter is disconnected.

john30, 2021-06-06:
it seems you set the adapter to enhanced mode, but not ebusd. please change ebusd config to enh:192.168.0.3:9999

wouldsmina, 2021-06-12:
I also tested enh:192.168.0.3:9999. it was same. 
I just tested again and this time the problem seems to be solved ... recently I changed my router, I can't believe it, but maybe there is a link. 

logs : 
```
2021-06-13 00:06:29.436 [update notice] received read bai Status01 QQ=10: 26.0;0.0;9.875;0.0;0.0;off
2021-06-13 00:06:31.401 [update notice] received read bai DateTime QQ=10: nosignal;-:-:-;-.-.-;-
2021-06-13 00:06:31.635 [update notice] received read bai Status02 QQ=10: auto;76;73.0;76;65.0
2021-06-13 00:06:32.544 [bus notice] device invalid
2021-06-13 00:06:37.574 [bus notice] device status: resetting
2021-06-13 00:06:37.574 [bus notice] re-opened 192.168.1.3:9999
2021-06-13 00:06:37.796 [bus notice] device status: reset
2021-06-13 00:06:39.476 [update notice] received read bai Status01 QQ=10: 26.0;0.0;9.875;0.0;0.0;off
2021-06-13 00:06:44.498 [update notice] received update-write bai SetMode QQ=10: auto;10.0;49.0;-;1;0;0;0;0;0
```

closed 2021-06-12 by wouldsmina

Madrox24, 2025-12-07:
@wouldsmina Hi, did you manage to fix that issue with the thermostat? I'm looking for **_e700** configuration file

```
2021-04-20 22:10:50.764 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "_e700", SW0218, HW0302
2021-04-20 22:10:50.764 [main error] scan config 15: ERR: element not found
```

wouldsmina, 2025-12-07:
Hi @Madrox24,
It’s been a very long time since I last worked with ebusd. I managed to get it working the way I wanted (retrieving some information to make statistics) and then I stopped there.
 
Here’s what I found:
I created a file /etc/ebusd/en/vaillant/15._e700.csv from another.
Here is its content:

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment
#,E7f00,ExaCONTROL E7 radio,E7f00 206,,,,,,,,,,
*r,,,,,,"B509","0D",,,,,,
*w,,,,,,"B509","0E",,,,,,
*wi,#install,,,,,"B509","0E",,,,,,
*[SW],scan,,,SW,,,,,,,,,
# ##### general #####,,,,,,,,,,,,,
[SW>=201]r,,DisplayedRoomTemp,DisplayedRoomTemp,,,,"3E00",,,temp,,,displayed room temperature
[SW=102-131]r,,RoomTemp,RaumtemperaturIst,,,,"0000",,,tempsensor,,,raw value current room temperature
r,,OutsideTemp,OutsideTempSensor,,,,"0300",,,tempsensor,,,current outdoor temperature and sensorstate
r;w,,OperatingMode,OperatingMode,,,,"0500",,,mcmode,,,operating mode
r;w,,ActualWeekday,ActualWeekday,,,,"0600",,,bday,,,current day of week
r;w,,Time,Time,,,,"0700",,,time,,,current time
r;w,,ReducedNightTemp,ReducedNightTemp,,,,"0800",,,temp,,,drop temperature setpoint
r;wi,,YearCalendarActive,YearCalendarActive,,,,"0B00",,,onoff,,,enable calendar (=> automatic switching to and fro daylight savings time)
r;w,,Date,Date,,,,"0D00",,,date,,,current date (calendar must be enabled previously!)
r,,IsInHoliday,Heating curve,,,,"1100",,,UCH,,,holiday function - remaining days
r,,IsInQuickVeto,IsInQuickVeto,,,,"1600",,,onoff,,,temperature override (quick veto)
r,,QuickVetoTemp,QuickVetoTemp,,,,"1700",,,temp,,,room temperature setpoint during activated quick veto function
r;w,,HwcTempDesired,DesiredDHWTemp,,,,"1E00",,,temp,,,DHW setpoint
r;w,,ComfTempDesired,DesiredComfTemp,,,,"2000",,,temp;temp;temp;temp,,,room temperature setpoint
# ##### extra #####,,,,,,,,,,,,,
r;wi,,RoomTempOffset,RaumtemperaturIstOffset,,,,"0200",,,temp,,,room temperature correction
r;wi,,OutsideTempOffset,OutDoorTempCorr,,,,"0400",,,temp,,,outside temperature correction
r;wi,,HeatingCurve,Heating curve,,,,"2300",,,curve,,,heating curve (only relevant if an outside temperature sensor is connected to the BMU)
r;wi,,ZweipunktAnalogSlct,ZweipunktAnalogSlct,,,,"2900",,,UCH,0=two-point control;1=analogue control,,two-point / analogue control
r;wi,,WeekDayProgSwitch,WeekDayProgSwitch,,,,"2B00",,,UCH,1=daily;7=weekly,,switch from weekly program over to a daily one
# ##### internal #####,,,,,,,,,,,,,
r,,StackeBUSTaskMax,MaxStackeBUSTask,,,,"1800",,,UCH,,,maximum number of bytes needed for the eBUS task stack
r,,StackMainTaskMax,MaxStackMainTask,,,,"1900",,,UCH,,,maximum number of bytes needed for the main task stack
r,,StackLifeCheckTaskMax,MaxStackLifeCheckTask,,,,"1A00",,,UCH,,,maximum number of bytes needed for the life check task stack
r,,eBUSFifoDiffCntMax,eBUSFifoDiffCntMax,,,,"2800",,,UCH,,,max. eBUS-fifo level
r,,RTCAdjustment,RTCAdjustment,,,,"2C00",,,SIN,,,adjustment value for RTC-timer (add or subtract 1sec in RTC-synchronisation time every x minute(s); add: positive value; sub: negative value)
r,,EEpromMaxInkonsCnt,EEpromMaxInkonsCnt,,,,"2D00",,,UIN,,,EEPROM inconsistency counter
r,,POCResetCnt,POCResetCnt,,,,"2E00",,,UIN,,,power on resets
r,,WDResetCnt,WDResetCnt,,,,"2F00",,,UIN,,,watchdog resets
r,,LVResetCnt,LVResetCnt,,,,"3000",,,UIN,,,low voltage resets
r,,CMResetCnt,CMResetCnt,,,,"3100",,,UIN,,,clock monitor resets
r,,COMErrorCnt,COMErrorCnt,,,,"3200",,,UIN,,,eBUS-communication errors
r,,VariantDKRefreshCnt,VariantDKRefreshCnt,,,,"3300",,,UCH,,,counts how often variant ID has been corrected
r,,RfCommErrorPending,RfCommErrorPending,,,,"3500",,,onoff,,,radio communication fails
r,,BMUErrorStatus,BMU_ErrorStatus_DK,,,,"3600",,,UCH,,,error status of the BMU
r,,BatteryError,BatteryError,,,,"3700",,,onoff,,,low battery
r,,eBUSCommErrorPending,eBUSCommErrorPending,,,,"3900",,,onoff,,,eBUS communication fails
!include,timerhc.inc,,,,,,,,,,,,
!include,errors.inc,,,,,,,,,,,,

```

With this, I was able to retrieve the boiler’s status and temperature.

wouldsmina, 2025-12-07:
Oh yes, and finally, I configured Telegraf to store the data in an InfluxDB database.

In telegraf.conf:
```
[[inputs.http]]
  urls = ["http://localhost:8080/data?cmd=json"]
  method = "GET"
  data_format = "json"
  tag_keys = ["name"]
  json_string_fields = ["data"]

```

The temperature is located in the field bai_messages_Status01_fields_0_value and the status in bai_messages_SetMode_fields_flowtempdesired_value.

I hope this can help you.