john30/ebusd #420: Install issue on rpi using eBUS Adapter 3 using ebusd-21.2_armhf-buster.deb

hihipto, 2021-04-21:
Hi,
I've tried to install ebusd on my old RPIv1 extended with the eBUS Adapter 3 HAT

The unit is not yet connected to the heating systems eBUS, but I can't get ebusd started

```
root@raspberrypi:/etc/apt/sources.list.d# dpkg -i ebusd-21.2_armhf-buster.deb
(Reading database ... 44639 files and directories currently installed.)
Preparing to unpack ebusd-21.2_armhf-buster.deb ...
Unpacking ebusd (21.2) over (21.2) ...
Setting up ebusd (21.2) ...
Instructions:
1. Edit /etc/default/ebusd if necessary
   (especially if your device is not /dev/ttyUSB0)
2. Start the daemon with 'service ebusd start'
3. Check the log file /var/log/ebusd.log
4. Make the daemon autostart with 'update-rc.d ebusd enable'
Processing triggers for systemd (241-7~deb10u6+rpi1) ...

root@raspberrypi:/etc/apt/sources.list.d# service ebusd start
Job for ebusd.service failed because the control process exited with error code.
See "systemctl status ebusd.service" and "journalctl -xe" for details.

root@raspberrypi:/etc/apt/sources.list.d#
```

I've restarted and reinstalled ebusd, but no success so far.
Any tip what I can check next to install it successfully? 

Thanks

hihipto, 2021-04-21:
some more details on "systemctl status ebusd.service" and "journalctl -xe"

`root@raspberrypi:/home/pi# systemctl status ebusd.service
● ebusd.service - LSB: controls ebusd, the daemon for communication with eBUS heating systems.
   Loaded: loaded (/etc/init.d/ebusd; generated)
   Active: failed (Result: exit-code) since Wed 2021-04-21 10:42:32 CEST; 8min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 841 ExecStart=/etc/init.d/ebusd start (code=exited, status=139)

Apr 21 10:42:31 raspberrypi systemd[1]: Starting LSB: controls ebusd, the daemon for communication with eBUS heating systems....
Apr 21 10:42:32 raspberrypi ebusd[841]: Starting ebusd: ebusd/etc/init.d/ebusd: line 43:   846 Segmentation fault      start-stop-daemon --start --quiet --oknodo --pidfile $pidfile --exec $DAEMON -- --pidfile $pid
Apr 21 10:42:32 raspberrypi ebusd[841]:  failed!
Apr 21 10:42:32 raspberrypi systemd[1]: ebusd.service: Control process exited, code=exited, status=139/n/a
Apr 21 10:42:32 raspberrypi systemd[1]: ebusd.service: Failed with result 'exit-code'.
Apr 21 10:42:32 raspberrypi systemd[1]: Failed to start LSB: controls ebusd, the daemon for communication with eBUS heating systems..
lines 1-12/12 (END)
root@raspberrypi:/home/pi# journalctl -xe
--
-- The job identifier is 370.
Apr 21 10:41:50 raspberrypi systemd[1]: Reloading.
Apr 21 10:41:54 raspberrypi systemd[1]: getty@tty1.service: Current command vanished from the unit file, execution of the command list won't be resumed.
Apr 21 10:42:31 raspberrypi systemd[1]: Reached target Network is Online.
-- Subject: A start job for unit network-online.target has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit network-online.target has finished successfully.
--
-- The job identifier is 431.
Apr 21 10:42:31 raspberrypi systemd[1]: Starting LSB: controls ebusd, the daemon for communication with eBUS heating systems....
-- Subject: A start job for unit ebusd.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit ebusd.service has begun execution.
--
-- The job identifier is 374.
Apr 21 10:42:32 raspberrypi ebusd[841]: Starting ebusd: ebusd/etc/init.d/ebusd: line 43:   846 Segmentation fault      start-stop-daemon --start --quiet --oknodo --pidfile $pidfile --exec $DAEMON -- --pidfile $pid
Apr 21 10:42:32 raspberrypi ebusd[841]:  failed!
Apr 21 10:42:32 raspberrypi systemd[1]: ebusd.service: Control process exited, code=exited, status=139/n/a
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit ebusd.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 139.
Apr 21 10:42:32 raspberrypi systemd[1]: ebusd.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit ebusd.service has entered the 'failed' state with result 'exit-code'.
Apr 21 10:42:32 raspberrypi systemd[1]: Failed to start LSB: controls ebusd, the daemon for communication with eBUS heating systems..
-- Subject: A start job for unit ebusd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit ebusd.service has finished with a failure.
--
-- The job identifier is 374 and the job result is failed.
lines 1491-1534/1534 (END)
--
-- The job identifier is 370.
Apr 21 10:41:50 raspberrypi systemd[1]: Reloading.
Apr 21 10:41:54 raspberrypi systemd[1]: getty@tty1.service: Current command vanished from the unit file, execution of the command list won't be resumed.
Apr 21 10:42:31 raspberrypi systemd[1]: Reached target Network is Online.
-- Subject: A start job for unit network-online.target has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
[... 61 lines cut ...]
Apr 21 10:42:32 raspberrypi ebusd[841]:  failed!
Apr 21 10:42:32 raspberrypi systemd[1]: ebusd.service: Control process exited, code=exited, status=139/n/a
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- An ExecStart= process belonging to unit ebusd.service has exited.
--
-- The process' exit code is 'exited' and its exit status is 139.
Apr 21 10:42:32 raspberrypi systemd[1]: ebusd.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- The unit ebusd.service has entered the 'failed' state with result 'exit-code'.
Apr 21 10:42:32 raspberrypi systemd[1]: Failed to start LSB: controls ebusd, the daemon for communication with eBUS heating systems..
-- Subject: A start job for unit ebusd.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
--
-- A start job for unit ebusd.service has finished with a failure.
--
-- The job identifier is 374 and the job result is failed.

lines 1491-1534/1534 (END)`

hihipto, 2021-04-21:
solved it by build-and-install on the system using guidelines:
https://github.com/john30/ebusd/wiki/1.-Build-and-install

john30, 2021-06-06:
duplicate of #394, see https://github.com/john30/ebusd-debian/issues/5

closed 2021-06-06 by john30