john30/ebusd #421: ebusd.log file within Docker

salvq, 2021-04-21:
Hello,

I have created docker and running just fine. What I can not find is actual ebusd.log file within a docker (there is actually no file called ebusd.log within Docker)

**Logging works fine within docker (results are being shown on terminal)**
```
2021-04-21 20:55:48.263 [main notice] ebusd 21.1.v21.1-3-g62221bb started with auto scan                                                                                                                                                                         
2021-04-21 20:55:48.264 [main info] loading configuration files from /home/pi/ebusd-configuration/ebusd-2.1.x/en                                                                                                                                                                                                                                ​2021-04-21 20:55:48.264 [bus notice] bus started with own address 31/36                                                                                                                                                                                          
2021-04-21 20:55:48.264 [main info] registering data handlers                                                                                                                                                                                                    
2021-04-21 20:55:48.264 [mqtt info] mosquitto version 1.4.10 (compiled with 1.4.10)                                                                                                                                                                              
2021-04-21 20:55:48.268 [mqtt debug] connection requested                                                                                                                                                                                                        
2021-04-21 20:55:48.268 [main info] registered data handlers                                                                                                                                                                                                     
2021-04-21 20:55:48.269 [mqtt notice] connection established                                                                                                                                                                                                     
2021-04-21 20:55:48.269 [mqtt debug] publish ebusd/global/version "ebusd 21.1.v21.1-3-g62221bb"                                                                                                                                                                  
2021-04-21 20:55:48.269 [mqtt debug] publish ebusd/global/running true                                                                                                                                                                                           
2021-04-21 20:55:58.269 [main debug] performing regular tasks                                                                                                                                                                                                    
2021-04-21 20:55:58.269 [mqtt debug] publish ebusd/global/scan "finished"                                                                                                                                                                                        
2021-04-21 20:56:04.403 [mqtt debug] publish ebusd/global/uptime 16                                                                                                                                                                                              
2021-04-21 20:56:08.269 [main debug] performing regular tasks                                                                                                                                                                                                    
2021-04-21 20:56:18.269 [main debug] performing regular tasks       
```                    

**Docker-compose.yml file**

```
version: '3'

services:
  test:
    image: john30/ebusd:latest
    container_name: ebusd
    restart: unless-stopped
    network_mode: host
    ports:
      - "8888:8888"
    command: -d udp:192.168.1.50:10000 --latency=80 -f -l /var/log/ebusd.log --logareas=all --loglevel=debug --httpport=8080 --accesslevel=* --scanconfig --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/en --mqtthost=192.168.1.106 --mqttport=1883 --mqttuser=XXXX --mqttpass=YYYY --mqttjson --mqtttopic=ebusd/%circuit/%name
```

Why there is no ebusd.log file ? This logging option was working just fine on my RPI (without docker)

Thanks

netweaver1970, 2021-04-28:
Some ideas...

"docker logs ebusd" should give you the log file contents.

I believe Docker images log by default into stderr, so they are being syphoned off to a docker managed file.
The ebusd command is a normal UserLevel application, so you have the full control on the output of it.

If you want the 'real file', see this link : [https://sematext.com/blog/docker-logs-location/#:~:text=Remember%2C%20one%20log%20file%20per,on%20a%20Linux%20Docker%20host.](https://sematext.com/blog/docker-logs-location/#:~:text=Remember%2C%20one%20log%20file%20per,on%20a%20Linux%20Docker%20host.)

also, it might be that the log file is being generated inside the container and unless you are bringing that file path out of your container using a volume in your docker-compose, you can't see it outside your container.

Can you see the file when you go inside the container?

salvq, 2021-04-28:
@netweaver1970 No, there is no ebusd.log file in docker

Running docker in background or in foreground did not solve this, `sudo docker-compose up -d` or `sudo docker-compose up`

[image]

Running `docker logs ebusd` shows log content and as you pointed 👍 the "real log file" in json format is located as described by link. So now, I do have a log which I can work with.

But it still bothering me why the log is not created :)

salvq, 2021-05-03:
If somebody is interested having Docker-compose.yml with logging to syslog server, see for inspiration...

```
version: '3'

services:
  test:
    image: john30/ebusd:latest
    container_name: ebusd
    restart: unless-stopped
    network_mode: host
    ports:
      - "8888:8888"
    command: -d udp:192.168.1.50:10000 --latency=80 -f -l /var/log/ebusd.log --logareas=all --loglevel=debug --httpport=8080 --accesslevel=* --scanconfig --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/en --mqtthost=192.168.1.106 --mqttport=1883 --mqttuser=XXXX --mqttpass=YYYY --mqttjson --mqtttopic=ebusd/%circuit/%name
   logging:
      driver: "syslog"
      options:
        syslog-address: "tcp://192.168.X.106:514"
        tag: "{{.Name}}/{{.ImageName}}"
```

closed 2021-05-03 by salvq