john30/ebusd #423: udp connection only

RoSulek1, 2021-04-26:
ebusd adapter3 with wemos wifi connection.
Only on UDP connection on port 9999 i have ebusd connected. On TCP or Enhanced connection there is no connecion. Is this normal?

[image]

netweaver1970, 2021-04-29:
What is your configuration screen settings?
[image]

What is your device string when you start ebusd? 
Below are mine: 2 variants: once direct, once via docker. The docker port config looks a bit funny but hat's because there's a current issue with the latest docker on systems with IPV6 disabled. The mqtttopic parameter is not needed, as it's the same as the default.

`docker run --rm -it -p “0.0.0.0:8888:8888/tcp” john30/ebusd:devel -f --scanconfig -d enh:192.168.0.128:9999 --mqttport=1883 --mqtthost=192.168.0.11 --mqttuser=****** -d enh:192.168.0.95:9999 --mqttpass=***** --mqttjson --mqtttopic=ebusd/%circuit/%name --latency=20000 –loglevel=debug`

`ebusd -f --scanconfig --mqttport=1883 --mqtthost=192.168.0.11 --mqttuser=****** -d enh:192.168.0.95:9999 --mqttpass=****** --mqttjson --mqtttopic=ebusd/%circuit/%name --latency=20000 -–loglevel=debug`

RoSulek1, 2021-04-29:
ebusd device string: udp:192.168.0.141:9999,
There is no any connection to ebusd from wemos on tcp or enhanced
connection :(
I dont know im' doing something wrong :/

docker run --rm -it -p 8888 john30/ebusd:devel -f --scanconfig -d udp:
192.168.0.141:9999 --loglevel=debug --ble, ble ble ;)
docker connect with mqtt, but there is no output of ebusctl info

[image: image.png]

john30, 2021-05-06:
enhanced proto is tcp only and you need to use "enh:<ip>:9999" on ebusd

closed 2022-02-20 by john30