john30/ebusd #429: readall.sh nc: invalid option -- 'q'

Der-Schubi, 2021-05-06:
Hello,

i'm using Arch Linux and the readall.sh Script isn't working, as netcat has no switch 'q'.
This switch is only used on OpenBSD netcat (Debian, Ubuntu, ...), but Arch uses GNU netcat.

Both versions have a switch "w" for the timeout, even though the long version is different.
As for the EOF part of 'q', GNU netcat knows a 'c' switch. I'm not sure if 'w' would be enough here?

GNU netcat 0.7.1:
-w, --wait=SECS            timeout for connects and final net reads
-c, --close                close connection on EOF from stdin

OpenBSD netcat (Debian patchlevel 1.206-1ubuntu1):
-w timeout	Timeout for connects and final net reads
-q secs		quit after EOF on stdin and delay of secs

For reference: #67 #68 

Best Regards,
Christian

GuillaumeC, 2021-05-17:
Hello,

To force read all values I use this command:
ebusctl f -l "*" -a|grep bai|awk '{print $2}' | xargs -L1 -t ebusctl r

Regards

kratz00, 2021-09-09:
@Der-Schubi Arch Linux gives you a choice to install your preferred netcat flavor (The default is *gnu-netcat* though).

- https://archlinux.org/packages/extra/x86_64/gnu-netcat/
- https://archlinux.org/packages/community/x86_64/openbsd-netcat/

```
[kratz00@helios scripts]$ sudo pacman -S netcat
:: There are 2 providers available for netcat:
:: Repository extra
   1) gnu-netcat
:: Repository community
   2) openbsd-netcat

Enter a number (default=1): 2
```