john30/ebusd #430: connections to bai and vr_70 work but issues with VRC700

spi43984, 2021-05-12:
I've got a Multimatic 700, a BAI and VR_70. My ebus USB adapter is connected to the ebus interface in the VR_70. Ebusd 21.2 runs on a rpi 2. Config files are the latest 2.1.x from ebusd-configuration.

Intermittently ebusd has issues finding the VRC700 on address 15. BAI and VR_70 are found almost immediately.

I run ebusd like
`sudo ebusd --device=/dev/heizung  -f --configlang=de --mqtthost=fhem --mqtttopic=vaillant --mqttport=1883 -c /etc/ebusd --scanconfig --mqttjson --accesslevel='*'`

After a successful scan I get my devices, config files get loaded:
```
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.0010003886.inc" ([PROD='0010009351']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
```

Some times VRC700 is missing, I can run a `ebusctl scan 15` but have to repeat it several times until I don't receive an `ERR: read timeout` anymore and the device is found:
```
ebusctl scan 15
15;Vaillant;70000;0419;4603;21;16;43;0020218357;0082;024373;N5
```

Unfortunately the config files don't get loaded, the device list then looks like:
```
version: ebusd 21.2.v21.2
update check: OK
access: *
signal: acquired
symbol rate: 22
max symbol rate: 119
min arbitration micros: 358
max arbitration micros: 482
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 3
messages: 218
conditional: 0
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.0010003886.inc" ([PROD='0010009351']), "vaillant/08.bai.csv"
address 10: master `#2`
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
```

Sometimes after a `ebusctl reload` the config file for the VRC700 gets loaded too. Sometimes I need to restart ebusd several times for all devices to be found and config files loaded.

The other issue I am experiencing is I can run for instance a `ebusctl r -c 700 Time` and get the result (sometimes I need to repeat the command one or twice as if get an `ERR: read timeout`). But if I run a `mosquitto_pub -h fhem -t vaillant/700/Time/get -m ''` I see immediately
```
2021-05-12 18:33:27.404 [bus error] send to 15: ERR: read timeout, retry
2021-05-12 18:33:28.087 [bus error] send to 15: ERR: read timeout, retry
2021-05-12 18:33:28.810 [bus error] send to 15: ERR: read timeout
2021-05-12 18:33:28.810 [bus error] send message part 0: ERR: read timeout
2021-05-12 18:33:28.810 [mqtt error] read 700 Time: ERR: read timeout
```

MQTT requests for BAI and VR_70 work.

I wonder if both issues are connected due to the connectivity/scan issues with the VRC700.

Has anybody an idea how to track this issue down?

spi43984, 2021-05-13:
I added `--latency=60000` to ebusd run parameters and things are getting better. Now I don't see any `send to 15: ERR: read timeout, retry` anymore but every now and then `send to 15: ERR: arbitration lost, retry` but I can work with that and see if I can get rid of those too.

spi43984, 2021-05-13:
Couldn't get rid of `send to 15: ERR: arbitration lost, retry` totally but that's the lesser evil as it happens only under heavy reading load now and then.

Curious if somebody else has some advice.

john30, 2021-06-06:
arbitration loss is nothing unusual in such a multi master bus, so that's fine.
the latency might need to be added depending on which USB device and eBUS interface you're using

spi43984, 2021-06-07:
> arbitration loss is nothing unusual in such a multi master bus, so that's fine.
> the latency might need to be added depending on which USB device and eBUS interface you're using

I thought so too. I can live with some arbitration lost now and then. With `--latency=55000` I have a stable connection to the VRC700 for weeks so all is good now.

closed 2021-06-07 by spi43984