john30/ebusd #432: ebusd does not work on Wirenboard 6.7

ashumkin, 2021-05-13:
I have a self/handmade eBus-UART-adapter v2.2 (https://github.com/jvitkauskas/ebus-adapter)
I've connected it via CP2102 USB-UART adapter to a my laptop and to a Vaillant Calormatic 470 "remote control" (RC) device.
It works with `ebusd`. It detects a boiler and a Calormatic RC. Also, it changes values (desired temperature, holiday dates, etc)...

But when I connect the SAME CP2102 adapter to the [Wirenboard controller](https://wirenboard.com/en/catalog/kontrollery/) (WB) (_just switch USB dongle to the other computer_) and run `ebusd` on the latter, it cannot "catch the data" (but does the signal). _Also, I remember, first I've tested the ebus-adapter board, I've tested it with the WB controller "on a table", and it also, worked, but when I've moved the "stage" to the working place, the issue occured_.
How can I debug the reasons?

```
$ ebusd -f  --scanconfig --log all:debug --lograwdata=bytes
2021-05-13 12:53:58.264 [main notice] ebusd 21.2.v21.2 started with auto scan on device /dev/ttyUSB0
2021-05-13 12:53:58.264 [main info] loading configuration files from http://ebusd.eu/config/
2021-05-13 12:53:58.673 [main info] reading templates /
2021-05-13 12:53:59.086 [main info] read templates in /
2021-05-13 12:53:59.091 [main info] reading file memory.csv
2021-05-13 12:53:59.290 [main info] successfully read file memory.csv
2021-05-13 12:53:59.292 [main info] reading file broadcast.csv
2021-05-13 12:53:59.702 [main info] successfully read file broadcast.csv
2021-05-13 12:53:59.706 [main info] read config files
2021-05-13 12:53:59.718 [main info] registering data handlers
2021-05-13 12:53:59.720 [main info] registered data handlers
2021-05-13 12:53:59.725 [bus notice] bus started with own address 31/36
2021-05-13 12:53:59.760 [bus notice] <aa
2021-05-13 12:53:59.760 [bus debug] ERR: SYN received during no signal, switching to ready
2021-05-13 12:53:59.760 [bus notice] signal acquired
2021-05-13 12:53:59.804 [bus notice] <aa
2021-05-13 12:53:59.848 [bus notice] <af
2021-05-13 12:53:59.877 [bus debug] ERR: read timeout during receive command, switching to skip
2021-05-13 12:53:59.892 [bus notice] <aa
2021-05-13 12:53:59.944 [bus notice] <aa
2021-05-13 12:53:59.985 [bus notice] <ba
2021-05-13 12:54:00.012 [bus debug] ERR: read timeout during receive command, switching to skip
2021-05-13 12:54:00.032 [bus notice] <aa
2021-05-13 12:54:00.076 [bus notice] <aa
2021-05-13 12:54:00.121 [bus notice] <aa
2021-05-13 12:54:00.166 [bus notice] <da
2021-05-13 12:54:00.193 [bus debug] ERR: read timeout during receive command, switching to skip
2021-05-13 12:54:00.210 [bus notice] <aa
2021-05-13 12:54:00.254 [bus notice] <aa
2021-05-13 12:54:00.299 [bus notice] <aa
2021-05-13 12:54:00.344 [bus notice] <aa
2021-05-13 12:54:00.390 [bus notice] <f5
2021-05-13 12:54:00.416 [bus debug] ERR: read timeout during receive command, switching to skip
2021-05-13 12:54:00.433 [bus notice] <aa
2021-05-13 12:54:00.478 [bus notice] <aa
2021-05-13 12:54:00.521 [bus notice] <aa
2021-05-13 12:54:00.566 [bus notice] <ea
2021-05-13 12:54:00.593 [bus debug] ERR: read timeout during receive command, switching to skip
...
2021-05-13 12:54:02.178 [bus debug] ERR: read timeout during receive command, switching to skip
2021-05-13 12:54:02.191 [bus notice] <aa
2021-05-13 12:54:02.235 [bus notice] <aa
2021-05-13 12:54:02.280 [bus notice] <aa
2021-05-13 12:54:02.284 [bus notice] <d0
2021-05-13 12:54:02.289 [bus notice] <08
2021-05-13 12:54:02.294 [bus notice] <b5
2021-05-13 12:54:02.298 [bus notice] <10
2021-05-13 12:54:02.303 [bus notice] <09
2021-05-13 12:54:02.308 [bus notice] <1f
2021-05-13 12:54:02.312 [bus notice] <00
2021-05-13 12:54:02.316 [bus notice] <00
2021-05-13 12:54:02.321 [bus notice] <ff
2021-05-13 12:54:02.326 [bus notice] <ff
2021-05-13 12:54:02.330 [bus notice] <ff
2021-05-13 12:54:02.334 [bus notice] <05
2021-05-13 12:54:02.339 [bus notice] <00
2021-05-13 12:54:02.344 [bus notice] <e0
2021-05-13 12:54:02.348 [bus notice] <7f
2021-05-13 12:54:02.349 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2021-05-13 12:54:02.352 [bus notice] <00
2021-05-13 12:54:02.352 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2021-05-13 12:54:02.372 [bus notice] <01
2021-05-13 12:54:02.376 [bus notice] <01
2021-05-13 12:54:02.380 [bus notice] <9a
2021-05-13 12:54:02.384 [bus notice] <f0
2021-05-13 12:54:02.389 [bus notice] <aa
2021-05-13 12:54:02.433 [bus notice] <aa
2021-05-13 12:54:02.478 [bus notice] <aa
```

`ebusd` that works on a laptop is also v21.2
Versions are the same, but the "problematic" is from `ebusd-21.2_armhf-stretch_mqtt1.deb` (on a Debian GNU/Linux 9.13 (stretch)) and the "working" is `ebusd-21.2_amd64-stretch_mqtt1.deb` on a `Ubuntu 20.04.1 LTS (focal)`

john30, 2021-06-06:
the log shows that the physical reception is not good enough as there should only be <aa when there is no real traffic, but you can see af, ba, da etc.

ashumkin, 2021-06-28:
It turned out ebus-adapter has no isolation between eBus and UART, so, the connection have became unstable.

closed 2021-06-28 by ashumkin