john30/ebusd #435: device status: eBUS comm error: framing

majowi5, 2021-05-25:
Every 5 to 10 min. I see errors in the logfile.
_[bus error] device status: eBUS comm error: framing_
_[bus error] arbitration start error_
[busError.log](https://github.com/john30/ebusd/files/6540136/busError.log)

As I understand this log in line 2 ebusAdater3 sends its Address (QQ=31) to start arbitration.
Unfortuntely ebusAdapter3 receives **"aa"** instead of a valid address; this causes the bus error entries in line 4 and line 5.

As I do not have any documentation or source code about ebusAdapter3 firmware, I can only guess what is happening on the line or the ebusAdapter3.
Could this be caused by a race condition, and the <aa was received  earlier, but was still in the buffer?

Most of the time ebusd and ebusAdapter3 work fine.
Hardware:  ebusAdapter3 in USB Mode; enhenced mode selected

Is there any parameter to change to get rid of  these errors?

majowi5, 2021-06-17:
Diving deeper my first assumption seems to be wrong.

ERR_FRAMING is a message generated by the interface and sent to the ebusd.

2021-06-15 08:26:01.739 [bus info] send message: **1708b509030dbb00**
2021-06-15 08:26:01.794 [bus error] device status: eBUS comm error: framing
2021-06-15 08:26:01.794 [bus error] arbitration start error
2021-06-15 08:26:01.931 [update info] received MS cmd: 1008b5110101 / 0966ff1017006c0000ff
2021-06-15 08:26:01.931 [update notice] received read bai Status01 QQ=10: 51.0;-;23.062;0.0;54.0;off
2021-06-15 08:26:02.146 [update info] sent MS cmd: **1708b509030dbb00** / 01f0

In my case a framing error occurs **only** in case of arbitration.
Most probable the firmware/hardware does not accept the arbitration byte as a valid symbol.

As the firmware for the adapter3 has not been disclosed, I cannot go further.

**Workaround**
In order to avoid disturbances on the ebus I changed the priority to the lowest value in the system.
This avoids that adapter3 wins the arbitration, but ebusd does not recognize this, because it receives a framing error.

**Questions**
Which conditions cause a framing error?

arbitration start failed
_…Indicates that the last arbitration request failed (arbitration was lost or sending failed)_
Arbitration lost is a normal condition as the ebusd has by default a medium priority, but what means _sending failed_?

**Remark to documentation**
Error codes (from interface to ebusd)
ERR_FRAMING 0x00: framing error
ERR_OVERRUN 0x00: buffer overrun error
Both have the same hex code? Seems to be a mistyping.

**Further processing**
If my assumptions are right, should this issue be transferred to ebusd-esp?

majowi5, 2021-07-03:
Reduced the ebus bus cable to 2.5 m to reduce influence by cable.
Number of errors did not decrease.

majowi5, 2021-07-05:
Analyzing the errors of ebusd.log per day (see below)

- There are almost only _arbitration start errors_ caused by framing errors during arbitration.
- Framing errors occur only during arbitration!
- There is no recognized arbitration (won or lost) at all.
- Ebus communication works fine, as almost no errors occur other than arbitration errors.
- As the _framing errors_ occur in **ebus-adapter3** I do not see any chance to modify a parameter of ebusd

**ebusd.log**_ errors per day
21-07-03_23:58
Total errors                            422
device status: eBUS comm error: framing 205
arbitration start errors                205
send to xx: ERR: read timeout, retry    10
Other errors                            2
Arbitration won                         0
Arbitration lost                        0

21-07-04_23:58
Total errors                            444
device status: eBUS comm error: framing 217
arbitration start errors                217
send to xx: ERR: read timeout, retry    9
Other errors                            1
Arbitration won                         0
Arbitration lost                        0

john30, 2021-07-07:
whcih ebusd version are you using? if you installed a .deb package, try compiling from sources first

majowi5, 2021-07-08:
I used the ebusd-21.2_amd64-buster.deb package.
Compiling from source will be done soon.

majowi5, 2021-07-09:
Compiling from source did not change anything. 
As the message _[bus error] device status: eBUS comm error: framing_ is **triggerd by the ebus-adapter 3** I assume the problem has to be solved there.
I do not know how to contribute to problem solving, as I don't have any info about ebus-adapter 3 details

john30, 2021-10-06:
what's the problem anyway? the communication works fine and yes, especially during arbitration framing errors might occur. this is logged now where it was hidden (as not detected at all) with previous adapters.
if you want, you can hide specific area errors using the log command, if that's what you want

john30, 2021-10-13:
please check if the communication is working in principle when deactivating enhanced mode, i.e.
* set the jumper on the adapter on J12 from 1-2 to 2-3
* remove the "enh:" prefix from the ebusd device string

john30, 2021-10-14:
Hm, that's weird. Does arbitration in either mode work ever (at least once)? Because if not, I'd rather assume a PCB issue.
I'll check if it is possible to implement a kind of self-test.
Some raw byte logging would be helpful, for both modes.
And can you please provide the ebuspicloader output?

john30, 2021-10-14:
do you have a power supply, e.g. 12V or 15V and a resistor?

john30, 2021-10-15:
thanks for the logs, I'll look at them tonight.

> I cannot supply the output from ebuspicloader now as the necessary pins are not soldered and I don't have them available.

how is that? the ebuspicloader uses the USB connection as well. The only thing you need to do is to short-cut pins 3 and 4 on J11 before you plugin the USB connector. You can do that with a short wire, a screwdriver, or tweezers.

15V DC would be ideal, 12V could be enough, but I need to check that. accuracy is not really a problem but it should be stabilized. A resistor with a value around 300 to 500 ohms would be good, 1k might work as well.

majowi5, 2021-10-15:
I do not know if communication ever worked as expected. Standard mode I never tested. I started testing Ebus-Adapter 3 end of April. This issue I posted first on May 25th.

john30, 2021-10-17:
the output looks good so far.
as said before, communcation works sometimes, but not reliable.
ok, then send me a mail with your address to ebusd@ebusd.eu

john30, 2021-11-16:
did you do the test?

john30, 2021-11-20:
I had the idea last night, that the framing issue might be due to the adapter responding to quickly to the received SYN symbol. So I added an option to configure the delay in the PIC via ebuspicloader (see commit https://github.com/john30/ebusd/commit/1f3a445e188ba4e7c27243d0e55f1a62b6945da8) with the new [PIC firmware 20211120](https://adapter.ebusd.eu/picfirmware.en.html#versions).

I was playing around with some values in my own setup today and got a bit less framing errors with 750 ns (0 is not better and 3000 is not better either). Maybe you want to give it a try.

You don't happen to have an oscilloscope, do you? Then you could just measure the delay your components do themselves.

Is the ecoVit you have a rather new hardware or how old is it? Another reason might be that faster chips are being used that do the arbitration a little bit different than others.

majowi5, 2021-11-24:
The test was unreliable; probably because of the low quality of my power supply.

I appreciate the possibility to modify a parameter in ebusAdapter3 very much. I will start a test tonight. In order to avoid random results I will keep _arbdel_ constant for about 24 hours. Therefore expect results in about 5 to 7 days.
My vaillant devices (ecoVit and VRC470) are about 9 years old. 

Newer hardware should not cause a problem as the minimum delay after a syn-character (AA) is defined as tzmin = 4300µs. (according to ebus spec chapter 10.8 (timing after syn-character)).
tzmax = tzmin +tvmax
tvmax depends on the speed tolerance (14,38µs to 104,40 µs)

Is ebusAdapter3 able to adhere to these limits?
I suppose that my Vaillant devices do adhere to the specification but are close to one or the other limit.

majowi5, 2021-11-24:
After restarting the ebusAdapter3 with arbdel=750 I got these values  for the arbitration delay in the logfile:
 ```
2021-11-24 22:39:17.142 [bus info] arbitration delay 726 - 2132 micros
2021-11-24 22:39:18.062 [bus info] arbitration delay 726 - 2450 micros
2021-11-24 22:41:22.210 [bus info] arbitration delay 726 - 2547 micros
....
2021-11-24 22:46:02.061 [bus info] arbitration delay 584 - 2569 micros
```
 can you explain the meaning of these values?

majowi5, 2021-11-25:
Setting arbdel =750 shows the following results:

- arbitration errors disappear totally.
- more _ebus comm error: framing_  than before.

I suppose a value of 750 for _arbdel_ is too high; next trial will be _arbdel=250_

Ref _tzmin_:
Is the documentation correct and the delay has to be 4300**µs** or is this a typing mistake and the value should be 4300**ns**?

john30, 2021-11-25:
hm, a bad power supply doesn't seem to be a good idea to me. any chance to get that replaced?
the specs are not taken seriously by any device afaik, but I'll measure the delay of the adapter this weekend with my oscilloscope.
the output of ebusd is what ebusd measured as arbitration delay. this is merely to be taken as a hint, as there is no possibility to measure this exactly. the output shows the overall minimum and maximum measured so far.
oh, you're right. the delay is in micros, not nanos. I'll correct this in ebuspicloader

john30, 2021-11-28:
After spending the whole weekend with measuring the delay after the SYN symbol, I finally [built a new firmware](https://adapter.ebusd.eu/picfirmware.en.html#versions) that is capable of handling the timing exactly as noted in chapter 10.8 and 6.3 of the eBUS spec physics.

By default, the new firmware uses a value close to the middle of tZmin and tZmax. It can now be adjusted within a range of 4167 (setting the delay with the [latest ebuspicloader](https://github.com/john30/ebusd/blob/master/src/tools/ebuspicloader.cpp) to 0 us) to 4790 us (delay 620 us).
The middle of the spec is at around 4377 us and the default was measured to be at around 4364 us, which should be close enough.

An interesting fact is that one of my heating devices uses a delay of the spec minimum of 4300 us, which is rather surprising but of course within the allowed range.
So the values to be set when being within the spec range are these:
* min 4300 us with delay=130
* max 4454 us with delay=290 (because the bitrate of the PIC is 0,01% above 2400 Bd and the edge steepness is below 1 us according to the osci)

Please try it and I'd recommend to start with the default delay and then test the minimum of the spec allowed (delay=130).

majowi5, 2021-11-30:
John, you did a great job!!!
During a period of 24 hours I had only 14 errors.
9 [bus error] send to 08: ERR: read timeout, retry
2 [bus error] device status: eBUS comm error: framing
3 [bus error] send to 08: ERR: wrong symbol received, retry

Before changing the firmware I had (within 24 h) about 
180 - 230 _device status: eBUS comm error: framing_  and
5-20 __send to xx: ERR: read timeout, retry_ errors_
During the next days I will try some minor modification of _arbdel_ but I think the problem is solved.
I will keep you updated
Many thanks.

john30, 2021-11-30:
that's great to hear! so the many hours spent were worth it :)

closed 2022-02-20 by john30