john30/ebusd #437: eigene CSV erstellen und einbinden

Yunkie, 2021-06-09:
Hallo zusammen,

zu aller erst vielen Dank an alle Beteiligten dieses Projekts! Wirklich toll das uns anderen so die Möglichkeit geschaffen wird, unsere Systeme zu überwachen / zu steuern!

zu meiner Frage/Problem:
Da die bereitgestellten Konfigurationsdateien leider wenige werte liefern, habe ich mich weiter mit dem Thema der Erstellung einer eigenen CSV-Datei beschäftigt. Dazu habe ich mir die Anleitung hier im GitHub angesehen, verstehe aber leider einiges nicht, wenn ich das mit bestehenden CSV-Dateien vergleiche.

Woher kommt zum Beispiel die Spalte "ID"? Woher weiß ich, welchen Datentyp ich angeben muss?

Ich bin auch etwas anders vorgegangen, als in der Anleitung, um an die jeweiligen Werte zu kommen und würde dazu gerne wissen, ob es bei meiner Herangehensweise Probleme gibt. Ich habe folgendes gemacht:
Ich habe an meiner Steuerung die Fachhandwerkerebene aufgerufen und habe das Aktor-/Sensor-Test Menü aufgerufen.
Hier konnte ich alle werte finden die mich interessieren (Vorlauf-/Rücklauftemperatur, Gebäudepumpendurchfluss etc.).
Diese habe ich im Wechsel jeweils 4 Mal aufgerufen um sie im Log einfacher zu finden. Anschließend habe ich mir das ebusd.log angesehen und ausgewertet, mit folgendem kürzergefassten Ergebnis:

<details>
<summary> Log Auszug </summary>

unknown MS cmd: 7108b51405051403ffff / 0414000000					`0414 = T.0.20 = 4-Wege-Ventil (Einheit)`
unknown MS cmd: 7108b51405051403ffff / 0414000000					`0414 = T.0.20 = 4-Wege-Ventil (Einheit)`
unknown MS cmd: 7108b51405051303ffff / 0413000000					`0413 = T.0.19 = Kondensatwannenheizung`
unknown MS cmd: 7108b51405051403ffff / 0414000000					`0414 = T.0.20 = 4-Wege-Ventil (Einheit)`
unknown MS cmd: 1076b5110101 / 0959ffe81fff630000ff
unknown MS cmd: 1076b512030f0201 / 071d0300cd0210ff
unknown MS cmd: 7108b51405051703ffff / 0417000000					`0417 = T.0.23 = Heizwendel Kompressor`
unknown MS cmd: 1076b5110101 / 0959ffe81fff630000ff
unknown MS cmd: 1076b512030f0201 / 071d0300cd0210ff
unknown MS cmd: 7108b51405051703ffff / 0417000000					`0417 = T.0.23 = Heizwendel Kompressor`
unknown MS cmd: 7108b51a03044f32 / 0e4f07c0cfc07b3c00b01b20000000
unknown MS cmd: 7108b51a03045033 / 0e500000c083020000000000000000
unknown MS cmd: 7108b51405051703ffff / 0417000000					`0417 = T.0.23 = Heizwendel Kompressor `
unknown MS cmd: 7108b51405051703ffff / 0417000000					`0417 = T.0.23 = Heizwendel Kompressor `
unknown MS cmd: 7108b51405052903ffff / 042900d801					`0429 = T.0.41 = Rücklauftemperatur (°C)`
unknown MS cmd: 1076b5110101 / 0959ffe81fff630000ff
unknown MS cmd: 1076b512030f0201 / 071d0300cf0210ff
unknown MS cmd: 7108b51405052903ffff / 042900d801					`0429 = T.0.41 = Rücklauftemperatur (°C)`
unknown MS cmd: 7108b51405052903ffff / 042900d801					`0429 = T.0.41 = Rücklauftemperatur (°C)`
unknown MS cmd: 7108b51405052903ffff / 042900d801					`0429 = T.0.41 = Rücklauftemperatur (°C)`
unknown MS cmd: 7108b51405052803ffff / 042800e001					`0428 = T.0.40 = Vorlauftemperatur (°C)`
unknown MS cmd: 7108b51405052803ffff / 042800e001					`0428 = T.0.40 = Vorlauftemperatur (°C)`
unknown MS cmd: 1076b5110101 / 0959fff51fff630000ff
unknown MS cmd: 1076b512030f0201 / 071d0300ce0210ff
unknown MS cmd: 7108b51405052903ffff / 042900d701					`0429 = T.0.41 = Rücklauftemperatur (°C)`
unknown MS cmd: 7108b51a03045932 / 0e5907c0cfc07b3c00b01b20000000
unknown MS cmd: 7108b51a03045a33 / 0e5a0000c083020000000000000000
unknown MS cmd: 7108b51a03045b34 / 0e5bcf2000c0022101001800061800
unknown MS cmd: 7108b51405052a03ffff / 042a001000					`042a = T.0.42 = Gebäudekreis: Wasserdruck (Bar)`
unknown MS cmd: 7108b51405052b03ffff / 042b000000					`042b = T.0.43 = Gebäudekreis: Durchfluss (l/h)`
unknown MS cmd: 1076b512030f0201 / 071d0300cf0210ff
unknown MS cmd: 7108b51405052b03ffff / 042b000000					`042b = T.0.43 = Gebäudekreis: Durchfluss (l/h)`
unknown BC cmd: 10feb505025c00
unknown MS cmd: 1076b5040100 / 0a0351391708060221f51f
unknown MS cmd: 1008b507020900 / 023707
unknown MS cmd: 1008b5110100 / 09ff0210000000000000
unknown MS cmd: 1076b51303040d00 / 020d00
unknown MS cmd: 7108b51405052a03ffff / 042a001000					`042a = T.0.42 = Gebäudekreis: Wasserdruck (Bar)`
unknown MS cmd: 1008b5120204ff / 00
unknown MS cmd: 1008b513020528 / 0101
unknown BC cmd: 10feb510020601
unknown BC cmd: 10feb508020900
unknown MS cmd: 7108b51405052b03ffff / 042b000000					`042b = T.0.43 = Gebäudekreis: Durchfluss (l/h)`
unknown MS cmd: 1076b5110101 / 0959fff51fff630000ff
unknown MS cmd: 1076b512030f0201 / 071d0300cf0210ff
unknown MS cmd: 7108b51405053703ffff / 043700fe01					`0437 = T.0.55 = Kompressorauslasstemperatur`
unknown MS cmd: 1076b5110101 / 0959fff51fff630000ff
unknown MS cmd: 1076b512030f0201 / 071d0300ce0210ff
unknown MS cmd: 7108b51405053003ffff / 043000fc00					`0430 = T.0.48 = Lufteinlasstemperatur (°C)`
</details>

Um diese Werte abfragen zu können, habe ich folgende CSV-Datei erstellt. Ist das so richtig? Könnte ebusd damit den entsprechenden Wert abfragen? Und wie binde ich die CSV ein, sodass ebusd diese heranzieht um Werte abzufragen?
<details>
<summary> CSV-Datei </summary>

type (r[1-9],w,u)",circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment
r,,,Ruecklauftemperatur,,08,B514,29,,,tempv,,,

</details>

Was mich irritiert ist, dass wenn ich mir vorhandene CSV-Dateien (Beispielsweise 15.700.csv) ansehe, dass dort teilweise die PBSB nicht angegeben sind. Warum ist das so? 

Der Vollständigkeit halber hier noch ein paar Systemdaten von mir:
Vaillant aroTherm Plus VWL 75/6
Vaillant UniTower Plus VIH QW 190/6 mit HMU-IN 570 und multiMATIC VRC 700/6

<details>
<summary> ebusctl info </summary>

root@iobroker-pi:/home/iobroker# ebusctl info
version: ebusd 21.2.v21.2
update check: OK
signal: acquired
symbol rate: 64
max symbol rate: 354
min arbitration micros: 575
max arbitration micros: 9916
min symbol latency: 0
max symbol latency: 18
reconnects: 0
masters: 4
messages: 558
conditional: 0
poll: 1
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0605;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0605;HW=5103"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0614;HW=6903", loaded "vaillant/ec.sol.sc.csv"
</details>

<details>
<summary> ebusctl find -d -v </summary>

root@iobroker-pi:/home/iobroker# ebusctl find -d -v
700 DisplayedOutsideTemp = tempv=21.1992
700 Hc1ActualFlowTempDesired = tempv=0.0
700 Hc1FlowTemp = tempv=42.0
700 Hc1MaxFlowTempDesired = tempv=40.0
700 Hc1PumpStatus = =0
700 Hc1Status = =0
700 HcStorageTempTop = tempv=-
700 HwcMaxFlowTempDesired = tempv=80.0
700 HwcStorageTempBottom = tempv=-
700 HwcStorageTempTop = tempv=-
700 HydraulicScheme = =8
700 MaxRoomHumidity = =40
700 MultiRelaySetting = mamode=circulation
700 PrEnergySumHwcThisMonth = energy4=19
700 SystemFlowTemp = tempv=-
700 WaterPressure = pressv=1.6
broadcast outsidetemp = temp2=21.250
broadcast vdatetime = time=10:55:10;date=09.06.2021
hmu ConsumptionTotal =  (ERR: invalid position for 3108b516021802 / 00)
hmu EnergyHc = energy=107
hmu SetMode = hcmode=auto;flowtempdesired=0.0;hwctempdesired=-;hwcflowtempdesired=-;disablehc=1;disablehwctapping=1;disablehwcload=1;remoteControlHcPump=0;releaseBackup=0;releaseCooling=0
hmu State = energy=0;energy=111;onoff=192;state=0
hmu Status01 = temp1=42.0;temp1=33.5;temp2=-;temp1=-;temp1=-;pumpstate=off
hmu StatusCirPump = =on
sc YieldThisYear =  (ERR: invalid position for 31ecb509030d3f00 / 00)
scan.08  = MF=Vaillant;ID=HMU00;SW=0605;HW=5103
scan.15  = MF=Vaillant;ID=70000;SW=0614;HW=6903
scan.15 id = prefix=21;year=20;week=47;product=0020266797;supplier=0082;counter=081439;suffix=N1
scan.76  = MF=Vaillant;ID=VWZIO;SW=0605;HW=5103
scan.76 id = prefix=21;year=20;week=27;product=0010022066;supplier=3100;counter=005471;suffix=N6
scan.ec  = MF=Vaillant;ID=SOL00;SW=0614;HW=6903
scan.ec id = prefix=21;year=20;week=47;product=0020266797;supplier=0082;counter=081439;suffix=N1
</details>

Ich habe leider keine Möglichkeit gefunden das ganze als "Question" zu markieren.
Ich würde mich sehr über etwas Aufklärung freuen! :-)
Vielen Dank!

Gruß,
Hendrik

majowi5, 2021-06-17:
_Was mich irritiert ist, dass wenn ich mir vorhandene CSV-Dateien (Beispielsweise 15.700.csv) ansehe, dass dort teilweise die PBSB nicht angegeben sind. Warum ist das so?_

siehe wiki "4.2. Defaults"; dort ist das ganz gut erklärt

john30, 2021-10-03:
Deine CSV stimmt noch nicht ganz. Für die Nachricht zur Rücklauftemperatur
7108b51405052903ffff / 042900d801
sendet also Master 71 and Slave 08 mit PBSB b514, Länge 05 und Daten 052903ffff. Der Slave 08 antwortet mit Länge 04 und Daten 2900d801. Hier findet sich der gewünschte Wert vermutlich an den letzten beiden Bytes. Hier ist es immer hilfreich, die zur Meldung passenden Werte vom UI zu notieren, dann tut man sich mit der Validierung leichter.
Die ID in CSV ist das, was die Nachricht vom Master von anderen unterscheidet und im Anschluss an die Länge in der Anfrage als zusammenhänges Byte-Array steht. In diesem Fall dürfte das 0529 sein oder sogar 052903. Insofern müsste die ID im CSV auf 0529 lauten statt nur 29.
In der Antwort müsstest Du via CSV dann erstmal die 29 oder sogar die 2900 ignorieren. Das geht mit einem extra Feld (field1) mit Typ IGN (für Ignore). In Summe sähe das dann etwa so aus:
```csv
"type (r[1-9],w,u)",circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates,divider/values,unit,comment
r,,ruecklauf,Ruecklauftemperatur,,08,B514,0529,,,IGN,,,,,,tempv,,,
```

edudele, 2022-08-22:
> Hallo und vielen Dank für die Antwort john30! Ich habe es inzwischen soweit lauffähig mit folgender CSV-Datei hinbekommen: [08.hmu00.csv](https://github.com/john30/ebusd/files/7376346/08.hmu00.csv)
> 
> Nur Falls noch jemand versucht Werte für die aroTherm Plus VWL 75/6 abzurufen.
> 
> Vielleicht noch eine Off-Topic Frage: Prinzipiell funktioniert die CSV recht gut. Leider bekomme ich bei einigen Werten "0022fc" als Antwort zurück was dezimal 64546 darstellt. Sieht für mich undefiniert aus.
> 
> Nur eine Vermutung: Ich gehe davon aus, das diese Werte direkt aus der HMU stammen und daher bei einer Abfrage über das Geräte Display direkt verarbeitet werden und nicht definiert auf den EBUS ausgegeben wird.
> 
> Andere Werte funktionieren aber:
> 
> `[update error] unable to parse read hmu00 HE_Speichertemperatur from 3108b51405052c03ffff / 042c0022fc: ERR: invalid position [main error] read hmu00 HE_Speichertemperatur: decode ERR: invalid position [update notice] sent read hmu00 HE_Sperrkontakt_S20 QQ=31: close [update error] unable to parse read hmu00 HE_Aussentemperatur from 3108b51405054503ffff / 04450022fc: ERR: invalid position [main error] read hmu00 HE_Aussentemperatur: decode ERR: invalid position [update error] unable to parse read hmu00 HE_Systemtemperatur from 3108b51405054603ffff / 04460022fc: ERR: invalid position [main error] read hmu00 HE_Systemtemperatur: decode ERR: invalid position [update notice] sent read hmu00 HE_Vorlauftemperatur QQ=31: 24.0 [update notice] sent read hmu00 HE_Ruecklauftemperatur QQ=31: 23.6 [update notice] sent read hmu00 HE_Wasserdruck QQ=31: 1.4 [update notice] sent read hmu00 HE_Durchfluss QQ=31: 85.7`
> 
> Hat dazu vielleicht jemand eine Idee? Oder eine Alternative wie ich wenigstens an die Speichertemperatur komme?
> 
> Vielen Dank!
> 
> Gruß, Hendrik

Guten Morgen Yunkie,

ich habe mir vor kurzem einen vaillant arotherm plus vwl 85/6 zugelegt.

Und ich treffe auf das gleiche Problem.

Sie haben die Dateien. Ich wäre bereit, sie zu tun, aber ich habe 0 Erfahrung darin, wie CSVs funktionieren.

edudele, 2022-08-22:
buenos días @Yunkie ,

Hace poco compré un vaillant arotherm plus vwl 85/6.

Y me encuentro con el mismo problema.

Tienes los archivos. Estaría dispuesto a hacerlos, pero no tengo experiencia en cómo funcionan los CSV.

jkrauss82, 2022-08-30:
Hallo @Yunkie 

zunächst einmal vielen Dank für Deine Beschreibungen hier und fürs Posten der CSV-Datei - diese funktioniert für meine VWL 75/6 ebenfalls, wie ich jetzt nach etwas hin und her bestätigen kann.

Das hin und her bezieht sich darauf, dass bei meiner Anlage diese Werte nur dann auf dem ebus ausgegeben werden, wenn ich in der Fachhandwerkerebene ins Testmenu gehe. Davor hatte ich mich gewundert, dass ich im Log nichts dazu finden konnte. Gibt es eine Möglichkeit, die Werte auch via dem ebus-Adapter anzufragen? Muss mich wahrscheinlich noch mehr mit der Doku dazu auseinandersetzen.

> Oder eine Alternative wie ich wenigstens an die Speichertemperatur komme?

Meine Anlage hat neben der VWL 75/6 (HMU00;0607;5103) den Regler VRC 720 und das Funkmodul 5 (VR71) verbaut. Als Speicher kommt ein Allstor Exclusive zum Einsatz. Über den VR71, der vom ebusd automatisch erkannt wird, kann ich über die  Message Typen SensorData1 und SensorData2 die Temperaturen des Pufferspeichers auslesen. Falls dieser bei Dir ebenfalls verbaut ist, könntest Du das probieren (die Werte sind dann bei mir in den Feldern S5, S6, S7, S8).

Ich würde empfehlen, einen Pull Request für die von Dir verlinkte CSV ins Repo ebusd-configuration (https://github.com/john30/ebusd-configuration) einzustellen. Diese funktioniert zumindest für die VWL 75/6 deutlich besser als die aktuell im Repo befindliche 08.hmu.csv.

Gruß

Jonas

mf76130, 2022-08-31:
hi,
gehört da in einem template noch was angepasst?
Weil bei checkconfig bekomme ich diesen Fehler:
2022-08-31 13:44:06.566 [main error] error reading config files from /etc/ebusd/: ERR: element not found, last error: vaillant/08.hmu00.csv:20: ERR: element not found, field type OPENCLOSE in field 1
Und es lädt die CSV dann auch nicht auf der Addresse

mfG
Michael

jkrauss82, 2022-08-31:
Hi @mf76130 

> gehört da in einem template noch was angepasst?

Ja, siehe meinen Kommentar weiter oben https://github.com/john30/ebusd/issues/437#issuecomment-1232093321:

> Dazu müssten allerdings auch noch die template-Werte aus der CSV ergänzt werden, also bspw. `onoff` oder `openclose`.

D.h. Du musst in der Spalte K (datatypes/templates) sämtliche nicht-standard-typen-Werte ersetzen. Ich habe z.B. `onoff` mit `UIN` ersetzt, da ich davon ausgehe, dass auf der Datenebene nur 1 (on) und 0 (off) verwendet werden bei dieser Art Info. Alle möglichen Typen findest Du hier: https://github.com/john30/ebusd/wiki/4.3.-Builtin-data-types

mf76130, 2022-08-31:
hab jetzt mal das ans Ende der _template.csv gestellt
openclose,UIN,0=off;1=modulating;2=thermostat,,
bekomme beim check keinen Fehler aber er lädt die csv trotzdem nicht

jkrauss82, 2022-08-31:
Hier werden gerade zwei Issues vermischt, die aber auch zusammengehören thematisch. Das nur als Hinweis an andere Leser.

Wenn ich das hier https://github.com/john30/ebusd-configuration/issues/263#issuecomment-1231307878 richtig sehe, meldet sich Deine WP als `HMU01`. In dem Fall musst Du die hier verlinkte CSV-Datei noch umbennen, z.B. in `08.hmu01.csv`, damit sie erkannt wird.

mf76130, 2022-08-31:
anders, bei mir will es die datei ohne 00 - 08.hmu.csv
man sollte halt beim austauschen der Dateien auch genau schauen :-)

jkrauss82, 2022-08-31:
Ja, so geht's natürlich auch :+1: 

Hier ist beschrieben, wie der Auswahlmechanismus für die Dateien funktioniert: https://github.com/john30/ebusd/wiki/4.7.-Automatic-configuration

mf76130, 2022-09-14:
Ich hab jetzt scanconfig aus der config entfernt und nur noch die csv's die ich brauche im lokalen ordner jetzt gehts auch bei mir.
Und ebusd erkennt die Werte laufend, kein neustart der Wärmepumpe mehr notwendig.

john30, 2022-09-17:
@mf76130 das liegt dann aber eher an deinem interface als an ebusd

mf76130, 2022-09-17:
ja, das kann gut sein, aber man kriegt ja momentan nichts besseres :-(

novadvbs, 2022-09-26:
Hallo!

Ich habe eine Ochsner Wärmepumpe und das CSV File von https://github.com/john30/ebusd/issues/437 verwendet.
jedoch verstehe ich den Zusammenhang im CSV File nicht. Möchte die Temperaturen vereinfacht mit dem Read Befehl (ebusctl read -f temperature twr1) auslesen. könnte mir jemand einen Tipp geben wie ich das in der CSV einstellen kann.

LG
Robin
[image]

MassiveAttack0403, 2022-09-27:
Habe eine VWL 87/4, kann auch grundsätzlich vieles auslesen. Schafft es jemand Details wie "aktuelle Leistung", "Arbeitszahl" usw. bzw. alles was in der Fachhandwerkebene zu finden ist, auszulesen?
Was bringt das modifizierte File 08.hmu00.csv aus Beitrag 1? Gehört das noch mit dem Origina gemergt?

jkrauss82, 2022-09-28:
> Habe eine VWL 87/4, kann auch grundsätzlich vieles auslesen. Schafft es jemand Details wie "aktuelle Leistung", "Arbeitszahl" usw. bzw. alles was in der Fachhandwerkebene zu finden ist, auszulesen?

Dafür ist genau die Datei von Yunkie und diese sollte mit, wie Du schreibst, mit der aktuell im Repo mitgelieferten Datei gemergt werden. Allerdings funktionieren dort einzelne Einträge im Moment - bei mir mit einer VWL 75/6 - noch nicht korrekt. Ich arbeite daran, allerdings kann ich noch keine Schätzung abgeben, wann ich das PR dafür erstellen kann.

Ich versuche, meinen Zwischenstand heute Abend hier zu posten.

MassiveAttack0403, 2022-09-28:
Das wäre fein, wenn du das hier bereitstellen könntest. Werde es - nach meinem Urlaub - auf meiner Anlage testen und die Ergebnisse posten

jkrauss82, 2022-09-28:
Hier die zwei Dateien, wobei die 08.hmu00.csv die mitgelieferte 08.hmu.csv mit einbettet. Achtung: Je nachdem, wie die Gerätekennung der VWL 87/4 sich in ebusd meldet, kann es sein, dass Du das "00" im Dateinamen anpassen musst, damit ebusd die richtige Zuordnung herstellt. Ggf. kann es auch sein, dass Wert-Definitionen für die VWL 75/6 und die VWL 87/4 abweichen und Nachrichten von ebusd nicht erkannt werden - Versuch macht klug ;)

[08.hmu.csv](https://github.com/john30/ebusd/files/9668340/08.hmu.csv)
[08.hmu00.csv](https://github.com/john30/ebusd/files/9668341/08.hmu00.csv)

Ich habe an ein paar Stellen Yunkies Datei so angepasst, dass sie lauffähig ist (in Yunkies Version sind ein paar Templates benutzt, die ich nicht zur Verfügung habe). Weiter habe ich einen Wert in der 08.hmu.csv angepasst, der für die VWL 75/6 zumindest falsch war (glaube das war CurrentCompressorUtil).

Allerdings funktionieren etliche Werte auch nicht bisher. Da bin ich wie gesagt dran, aber das wird noch etwas dauern.

MassiveAttack0403, 2022-09-29:
Kannst du die files uploaden, so dass sie automatisch geladen werden?
Wie gesagt, ich komme aktuell nicht zum Testen, erst nach dem Urlaub. 
Danke vorerst und stay tuned ;)

john30, 2022-10-03:
> Hier die zwei Dateien, wobei die 08.hmu00.csv die mitgelieferte 08.hmu.csv mit einbettet. Achtung: Je nachdem, wie die Gerätekennung der VWL 87/4 sich in ebusd meldet, kann es sein, dass Du das "00" [...]

08.hmu.csv genügt. es darf dann allerdings nicht noch eine 08.hmu0.csv oder 08.hmu00.csv im Ordner liegen. ebusd wirft der Reihe nach "0" aus dem Suffix raus, um den besten Treffer zu finden

jkrauss82, 2022-10-06:
Hallo @john30 

Ich hatte die 08.hmu00.csv so umgebaut, dass sie die 08.hmu.csv automatisch einbindet via der !include Direktive. In diesem Fall können also beide Dateien im Ordner liegen und wenn die hmu00 erkannt wird, so wird die hmu ebenfalls mit eingebunden.

Wäre das der korrekte Weg oder ist es besser, wenn die 08.hmu00.csv einfach auch alle Einträge der 08.hmu.csv erhält?

john30, 2022-10-16:
wenn sich die zwei nicht unterscheiden, kannst einfach einen Symlink machen.
Wenn sie sich unterscheiden, wäre es besser bspw. die Unterscheidung via HW oder SW zu machen wie in anderen CSVs, also spezifische Nachrichten für eine Version isolieren und via HW/SW Unterscheidung bedingt einbinden. Siehe bspw. https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/08.ehp.csv#L97

morphZ, 2022-10-24:
@jkrauss82 und @Yunkie, habe mal versucht eure gesammelten Erkenntnisse in Form der 08.hmu00.csv bei meinem System einzusetzen (habe eine andere Hydraulikstation, nämlich die VWZ MEH 97/6). Leider wurde damit nur ein einzelner Wert dekodiert (Durchflussmenge). Versuche mich deshalb auch am Entschlüsseln der entsprechenden ebus Nachrichten.

Leider mache ich zurzeit trotz vieler Stunden Doku und Quelltexte lesen kaum Fortschritte... Falls ihr also ein paar Tipps hättet, wäre ich super dankbar... Habe meine bisherigen Ergebnis in #689 gesammelt.

jkrauss82, 2022-10-24:
Hallo @john30, denke wir bräuchten Tests von anderen Modellen bspw. Geotherm und Arotherm Split, um zu sehen, ob sich eventuell die Nachrichten-Definitionen unterscheiden und wie da die Modell- und Softwarekennungen sind. Dann könnte man die Dateinamen entsprechend anpassen. Wäre toll, wenn es User gibt die das testen könnten, so dass man das möglichst allgemeingültig im Repo hinterlegen kann.

@morphZ Die VWZ MEH 97/6 habe ich auch, da klappen bei mir eigentlich die meisten Werte. Dein Setup sieht auch sehr ähnlich aus, eventuell hast Du einfach ebusd noch nicht dazu gebracht, die Konfigurationsdatei auch zu nutzen? Musst mal nach dem Start im Log schauen, dort gibt er Dir aus, welche Geräte er erkennt und sagt auch, welche CSV er nimmt um die Nachrichten-Definitionen zu laden.

Ich habe vor ein paar Tagen rausgefunden, wie ich die Nachrichtendefinitionen für die Pumpeneinstellungen in der Hydraulikstation sind. Ich versuche, das auch baldmöglichst hier wieder zu posten.

morphZ, 2022-10-25:
Hi @jkrauss82, das war schonmal ein guter Tipp, danke! Die Configfiles waren tatsächlich nicht korrekt geladen (leider funktioniert bei mir das scannen nicht: die Geräte antworten. Ich bin noch nicht sicher, ob es daran liegt, das mein ebus-usb nicht schreiben kann. Das hier ist das zugehörige issue #677.)

Deshalb habe ich das ebus-configuration gecloned und eure beiden Files dort reingepackt (hintereinander in einer datei). Seit ich den Bug behoben habe, können jetzt die Werte aus dem Fachhandwerker Menü (Sensortest) gelesen werden. Das ist schonmal ein Fortschritt 👍 

Leider kann ich die Werte auch nicht mit `ebusctl read HE_...` aktiv auslesen. (Vllt. ist es dann doch ein write Problem).

Hast du den esera-usb?
Hast du schonmal versucht die `b51a` Werte zu entziffern, die zwischen der VWZ und der aroTHERM ausgetauscht werden?