john30/ebusd #438: Enable cooling on VRC700

minitrue-nl, 2021-06-16:
Quick question:

I'm trying to get cooling to work on my VRC700 via ebusd, but I cannot find the right commands. So I'm trying to put my heat pump system in cooling mode, either trough the main settings, or as manually cooling (setting for a number of days.)
Can somebody point me in the right direction?

mwildbolz, 2021-06-18:
Same problem here - haven't found the right codes at the moment.
Probably there is another possible value for the operatingMode fields - other than the ones provided?

InToSSH, 2021-07-27:
I would like to see this option too. For some reason my VRC700 doesn't switch my heatpump to cooling automatically (although it should based on what Vaillant technician told me), I have to enable it manually, but this change is not reflected on any of those commands.

minitrue-nl, 2021-08-23:
No news here, but I will ask my Vaillant technician again about the option of switching to cooling automatically. Although I don't expect to use it in the short term with the Belgian summer we had. 😢 I'll post here if I know more.

javierin, 2021-09-24:
I've been using ebusd for almost a year with a Saunier Duval rebranded arotherm split 8 + VRC720. I am about to set everything up again, because I have a defective microsd in my EBUS RPi and also made an absolute mess setting things up a year ago.

@nafgarda Do you own a Vaillant or a Saunier Duval unit?

nafgarda, 2021-09-24:
I've got Vaillant.
Regarding this question it would be nice to know how all of you are controlling the whole system. Just to share ideas.

My zones are not controlled by Vaillant, so that is the plan. I have 9 analog thermostat and 11 zones controlled with 11 solenoids. My plan is to replace the analog thermostats with Shelly wifi relays equipped with a thermocouple. At first I would use the Shelly cloud interface to set up the desired temperatures and it scheduling. My solenoids have a dry contact circuit that tells if they are open or close. That circuit from all solenoids is already connected to the heat pump and when all the solenoids are close, the heating stops automatically. 

I'm planning to use as well ebusd with Node Red installed in the same RPI:

1. I will set up an interface to activate or disactivate the whole system using Exec node and commands like "ebus ebusctl  write -c 700 HwcOpmode day"
2. I will probably set up interfaces to change the schedules in the VRC700.
3. I will set up a rule in node red that will switch off the heat pump when the demanded power measured at home with a Shelly EM goes over the power limit. Here in Spain we pay a fixed tariff to the electric utility according to the maximum power you can use. The electricity meters power down the system when you pass the limit (with some thresholds). So it is cheaper to have a lower limit.
4. As I will also install a 4 kWp photovoltaic system on the roof in near future, I will set up rules in Node Red to use the solar surplus in two different ways: rising the the temperature of the whole house opening the solenoids with the shelly relays and/or heating water in the cylinder with ebusd.

Total aprox budget: 
130 euros in Shelly relays and Energy Meter
30 euros from and old RPI2
20 euros from EBUS USB interface

That's all I think.
Hope it is useful for others
Comments are welcome.

javierin, 2021-09-24:
I am only interested in pushing hot water control commands through ebus, as I don't have any other thermostat different from the remote control unit. I turn the fancoils on and off using plugs (the fancoil units remember their last state).

john30, 2021-10-03:
please discuss in the right area (e.g. https://github.com/john30/ebusd-configuration/issues/), not here as this repo is for the source code of ebusd and not for configurations

closed 2021-10-03 by john30