john30/ebusd #439: No Data after a day running via http

rainscha1, 2021-06-25:
I have follwing situation:
new ebus adapter 3 connected to a weishaupt wrsol 1.1.
raspberry pi: running ebusd service (connected via lan to the ebus adapter)
Openhab3 also running on the raspberry pi.

openhab read cyclic via http request
http://xxxxxxxx:yyyyy/data/wrsol?maxage=0
from the ebusd service.

And yes that works, but only maximum of one day. Than i will get no data anymore.
Solution so solve this problem: Unplug ebus adapter3 and plug in again.

BTW: I didn't got mqtt running on that raspberry pi. I use the mqtt image, configure it, but nothing happens. No logs, nothing.

john30, 2021-10-03:
please provide some ebusd logs, otherwise it's hard to reproduce

john30, 2022-09-17:
closed due to inactivity

closed 2022-09-17 by john30