john30/ebusd #440: Heizung ebus Ochsner WP - Spendenaktion

m4tze-smart, 2021-06-29:
Hallo community,

bin nun seit Freitag am verzweifeln und versuche eine Ochnser WP am ebus auszulesen. 
Leider ohne nennenswerten Erfolg.

Wer hat so etwas bereits zum laufen gebracht und kann gegen eine kleine Spende unterstützen ?
(Paypal, Discord, Teamview etc.)

Ochsner WP GWML 14
Esera TCP Koppler (via UDP angebunden)
ebusd 21.2.v21.2

`ebusd -f --configpath=/etc/ebusd -d udp:192.168.180.104:5000 -p 8888 --pollinterval=1 --acquireretries=6 --acquiretimeout=100 --receivetimeout=75 --sendretries=6 --scanconfig=full --loglevel=debug --latency=10 --receivetimeout=10000 -l /var/log/ebusd.log --accesslevel=* --httpport=8080 --htmlpath=/var/www/ --logareas=bus`

version: ebusd 21.2.v21.2
update check: OK
access: *
signal: no signal
reconnects: 1
masters: 3
messages: 13
conditional: 0
poll: 0
update: 4
address 01: master #6
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 70: master #4
address 75: slave #4

m4tze-smart, 2021-07-01:
Update Tag 5

Habe nun die ebus Leitung so kurz wie möglich gemacht und an den Latency rumgespielt.

-c /etc/ebusd/22102.csv 

EBUSD_OPTS="--configpath=/etc/ebusd/22102.csv -d udp:192.168.180.104:5000 -p 8888 --pollinterval=20 --acquireretries=9 --acquiretimeout=99 --sendretries=6 --scanconfig=full --logareas=all --loglevel=debug --latency=30000 --receivetimeout=20000 -l /var/log/ebusd.log --scanconfig=full --accesslevel=* --httpport=8080 --htmlpath=/var/www/"

version: ebusd 21.2.v21.2
update check: OK
signal: acquired
symbol rate: 36
max symbol rate: 66
reconnects: 0
masters: 2
messages: 0
conditional: 0
poll: 0
update: 0
address 03: master #11
address 31: master #8, ebusd
address 36: slave #8, ebusd

2021-07-01 10:50:51.928 [bus debug] ERR: SYN received during receive command, switching to ready
2021-07-01 10:50:52.022 [bus debug] ERR: SYN received during receive command, switching to ready
2021-07-01 10:50:52.163 [bus debug] ERR: SYN received during receive command, switching to ready
2021-07-01 10:50:52.303 [bus debug] ERR: SYN received during receive command, switching to ready
2021-07-01 10:50:52.397 [bus debug] ERR: SYN received during receive command, switching to ready
2021-07-01 10:50:52.538 [bus debug] ERR: SYN received during receive command, switching to ready
2021-07-01 10:50:52.678 [bus debug] ERR: SYN received during receive command, switching to ready
2021-07-01 10:50:52.819 [bus debug] ERR: SYN received during receive command, switching to ready
2021-07-01 10:50:52.926 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2021-07-01 10:50:52.930 [bus debug] ERR: ACK error during receive command ACK, switching to skip

2021-07-01 11:06:27.504 [bus notice] <fa
2021-07-01 11:06:27.551 [bus notice] <aa
2021-07-01 11:06:27.598 [bus notice] <aa
2021-07-01 11:06:27.645 [bus notice] <fe
2021-07-01 11:06:27.692 [bus notice] <aa
2021-07-01 11:06:27.739 [bus notice] <aa
2021-07-01 11:06:27.785 [bus notice] <fe
2021-07-01 11:06:27.832 [bus notice] <aa
2021-07-01 11:06:27.879 [bus notice] <aa
2021-07-01 11:06:27.926 [bus notice] <ff
2021-07-01 11:06:27.973 [bus notice] <aa
2021-07-01 11:06:27.977 [bus notice] <c1
2021-07-01 11:06:27.983 [bus notice] <d1
2021-07-01 11:06:27.996 [bus notice] <10
2021-07-01 11:06:28.003 [bus notice] <80
2021-07-01 11:06:28.016 [bus notice] <fa
2021-07-01 11:06:28.020 [bus notice] <82
2021-07-01 11:06:28.024 [bus notice] <fa
2021-07-01 11:06:28.031 [bus notice] <02
2021-07-01 11:06:28.035 [bus notice] <6b
2021-07-01 11:06:28.040 [bus notice] <26
2021-07-01 11:06:28.044 [bus notice] <e4
2021-07-01 11:06:28.051 [bus notice] <10
2021-07-01 11:06:28.056 [bus notice] <c8
2021-07-01 11:06:28.062 [bus notice] <b1
2021-07-01 11:06:28.071 [bus notice] <1d

majowi5, 2021-07-01:
Ich würde zuerst ebusd im readonly modus starten u. dann den esera Koppler so am Trimmer einstellen, dass gültige Datenpakete empfangen werden.

Berlenz, 2021-07-25:
Vielleicht kann ich was bzgl. Ochsner Konfigurationsdateien beitragen:
Meine Wärmepumpe: Ochser GMLW 9 HK plus, Hardware Version 1.10, Software Version 3.73, Installationjahr 2011.
Ich verwende die ebus Adapterplatine V1.6 (inzwischen veraltete aus dem Jahr 2016) mit USB nach RS232 (Seriel) Konverter am Raspberry PI4. Poti der ebusd Adapterplatine habe ich damals mit Oszilloskop eingestellt.

Ich vewende ebusd Daemon v21.2:
wget https://github.com/john30/ebusd/releases/download/v21.2/ebusd-21.2_armhf-stretch.deb
sudo dpkg -i --force-overwrite ebusd-21.2_armhf-stretch.deb
(Achtung: Datei /etc/default/ebusd wird überschrieben)
sudo systemctl enable ebusd

Habe lange nach ebusd Konfigurationsdateien gesucht und bin irgendwann auf einer russischen Seite fündig geworden. Die Konfigurationsdateien habe ich noch etwas abgeändert (siehe Anhang). Die Datein ins /etc/ebusd Verzeichnis kopieren:
[ebus.zip](https://github.com/john30/ebusd/files/6873582/ebus.zip)

Ich verwende folgenden Konfiguration in der ebusd Datei:
/etc/default/ebusd
EBUSD_OPTS="--configpath=/etc/ebusd/ --device=/dev/ttyUSB_EBUS --sendretries=10 --receivetimeout=100000 --log=all:error --port=8888"

Den USB port am Raspberry habe ich fest auf den Namen ttyUSB_EBUS gelegt.

Ich kann mit Home Assistant die Wärmepumpe auslesen und z.B. auch Kühlung ein/ausschalten. Dazu habe ich folgendes Home Assiatant Plugin geschrieben (falls einer Home Assistant einsetzt und mit ebus Kommunizieren möchte):
https://github.com/Berlenz/Home-Assistant-ebus-access

Hier mal ein Bild der in Home Assistant ausgelesenen Ochsner WP Daten:
[image]

john30, 2021-10-03:
`--configpath=/etc/ebusd/22102.csv` funktioniert so sicher nicht, es wird hier ein Verzeichnis erwartet. Also wenn dann `--configpath=/etc/ebusd`

m4tze-smart, 2021-10-16:
Super Vielen Dank für die Nachrichten.

Hab mich nun noch einmal intensiv hingesetzt und dem neuen Adapter v3 direkt auch Erfolge erzielen können.
Vielen DANK John für die Zusendung und Einstellung. War im Prinzip Alles bereits Startklar eingestellt.

Ebus kann nun gelesen werden und meldet die Daten auch an Mqtt weiter, allerdings habe ich mit der CSV nicht ganz so viele Werte erhalten wie Berlenz.

EBUSD_OPTS=„-f --scanconfig=full --configpath=/etc/ebusd/ --accesslevel=* --latency=50000 -d enh:192.168.180.21:9999 --logareas=all --logfile=/var/log/ebusd.log --loglevel debug --loglevel=debug --lograwdata --enablehex --mqttport=1883 --mqtthost=192.168.180.10 --mqtttopic=ebusd/%circuit/%name --mqttclientid=ebusd“

version: ebusd 21.2.v21.2
access: *
signal: acquired
symbol rate: 52
max symbol rate: 94
min arbitration micros: 0
max arbitration micros: 89
min symbol latency: 9
max symbol latency: 55
reconnects: 0
masters: 5
messages: 138
conditional: 0
poll: 0
update: 9
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=TEM;ID=25440;SW=0113;HW=0000"
address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3632;HW=3030"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=TEM;ID=24849;SW=0605;HW=0102"
address 18: slave #12, scanned "MF=TEM;ID=WE_2 ;SW=3632;HW=3030"
address 31: master #8, ebusd
address 36: slave #8, ebusd

Bekomme im Grunde nur:

2021-10-17 01:21:48.524 [mqtt debug] publish ebusd/master/10u0503 {
     "status": {"value": 1},
     "state": {"value": 0},
     "maxmin": {"value": 0},
     "boiler": {"value": 0.0},
     "buffer1": {"value": 0},
     "buffer2": {"value": 1}}
2021-10-17 01:21:48.524 [mqtt debug] publish ebusd/21576/temperature.buffer.flow {
     "group": {"value": 96},
     "value": {"value": 0},
     "type": {"value": "0d"},
     "unit": {"value": "°C"},
     "max": {"value": 100.0},
     "min": {"value": 0.0},
     "temperature": {"value": 37.0}}
2021-10-17 01:21:48.524 [mqtt debug] publish ebusd/21576/status.heatpump {
     "group": {"value": 53},
     "value": {"value": 1},
     "type": {"value": "00"},
     "unit": {"value": 0},
     "max": {"value": 25.5},
     "min": {"value": 0.0},
     "status": {"value": "Switched off"}}
2021-10-17 01:21:48.524 [mqtt debug] publish ebusd/21576/unknown.02c60040 {
     "group": {"value": 70},
     "value": {"value": 65},
     "type": {"value": "04"},
     "unit": {"value": 40},
     "max": {"value": -0.1},
     "min": {"value": 0.0},
     "status": {"value": 197}}
2021-10-17 01:21:48.524 [mqtt debug] publish ebusd/21576/unknown.02c80040 {
     "group": {"value": 72},
     "value": {"value": 65},
     "type": {"value": "04"},
     "unit": {"value": 42},
     "max": {"value": 143.9},
     "min": {"value": 0.0},
     "status": {"value": 51}}

Was das für unknown werte sind keine Ahnung ...
Der unknown.02c80040 könnte mit 51% die Luftfeuchtigkeit Außen oder Innen sein.
Mir fehlen aber noch die wichtigen Werte wie Vorlautemp., tbo / tbm

Würde neben so einer schönen Visualisierung auch gerne noch den PV Überschuss in die Heizung blasen, dazu benötige ich eine Möglichkeit auf die Heizungsdaten zuzugreifen bspw. Vorlauftemperatur +10 Grad

Winny1904, 2021-10-31:
Hallo zusammen,

Ich wäre auch ein einer neu CSV interessiert. Wobei ich mit der aktuellen CSV gute werte erhalte aber ich denke nicht alle.

`
boiler nominal = 132;0;0d;02;1000;0;50.0
boiler status = 52;1;00;00;255;0;0
boiler tb = 4;0;0d;02;1000;0;45.3
broadcast datetime = 16.098;13:13:17;31.10.2021
buffer boiler_b = 41;1;0d;10;9990;-9990;12.0
buffer flow = 96;0;0d;02;1000;0;26.1
buffer heating = 40;1;0d;10;9990;-9990;12.0
buffer nominal = 224;0;0d;02;1000;0;24.1
buffer status_b = 54;1;00;00;255;0;1
buffer tpm = 17;0;0d;02;1000;0;25.3
buffer tpo = 15;0;0d;02;1000;0;26.1
heatpump cycles = 81;1;1d;00;2;4;68;14
heatpump heatkwh = 138;11;04;3c;-1;0;0.8
heatpump heatmwh = 134;11;0d;12;9999;0;9647
heatpump hours = 130;10;0d;34;1000;00;0;0
heatpump nominal = no data stored
heatpump status = no data stored
heatpump tqa = 71;0;0d;02;1000;0;20.8
heatpump tqe = 80;1;1d;00;1;8196;0.0
heatpump twr = 70;0;0d;02;1000;0;18.9
heatpump twv = 7;0;0d;02;1000;0;24.1
heatpump volumekwh = 218;10;0d;34;1000;0;0.0
heatpump volumemwh = 129;11;0d;12;9999;0;4250
heatpump waterkwh = -;31;00;00;0;0;0.0
heatpump watermwh = -;31;00;00;0;0;0
heatpump_m mode = 53;1;00;00;255;0;Abgeschaltet
master 10u0503 = 1;0;0;0.0;0;0
master data1 = 26.098;-;0;0;0;0;0;0;0;25.301
master data2 = 0.000;50.000;0;0;3
scan.06  = TEM;25440;0113;0000
scan.08  = TEM;WE_1 ;3632;3030
scan.15  = TEM;24849;0605;0102
scan.18  = TEM;WE_2 ;3632;3030
temperature boiler2 = -;31;00;00;0;0;0.0
temperature buffer1 = 96;0;0d;02;1000;0;26.1
temperature buffer2 = no data stored
temperature outside1 = no data stored
temperature outside2 = no data stored
temperature outside = 0;0;0d;02;500;-500;16.1
temperature tqa1 = 70;0;0d;02;1000;0;18.9
temperature tqe1 = 71;0;0d;02;1000;0;20.8
temperature twr1 = 8;0;0d;02;1000;0;21.7
temperature twv1 = 7;0;0d;02;1000;0;24.1
temperature water = 4;0;0d;02;1000;0;45.3
unknow 01940048 = no data stored
unknow 01960042 = no data stored
unknow 01cc0042 = no data stored
unknow 02c60040 = 70;65;04;28;-1;0;211
unknow 02c80040 = 72;65;04;2a;1439;0;25
` 

Von denen Ich keine Daten bekomme (leider alles was mit der FB Heizung zu tun hat) habe ich in der CSV mit # deaktiviert

john30, 2021-11-06:
da müsst ihr shcon selbst Hand anlegen und rausfinden, was in den messages drin steht.
der Adapter 3 kann kein UDP, nur TCP und das eben enhanced oder nicht.

m4tze-smart, 2021-11-15:
@Winny1904 
kannst Du Deine CSV zur Verfügung stellen ? 
Sind ja doch deutlich mehr Werte als meine aktuell aufzeigt.

Gibt es eine Vorgehensweise wie die Werte tiefer analysiert werden können ? 
Also wenn ich es richtig verstanden habe, dann gilt folgendes:

unknow 01940048 = no data stored --> CSV fragt einen Wert ab der auf dem ebus nicht gesendet wird
unknow 02c80040 = 72;65;04;2a;1439;0;25 --> CSV fragt einen Wert ab der auf dem ebus gesendet wird, aber wir wissen noch nicht was er konkret bedeutet 

**Hier muss ich nur rausfinden was der Wert 226 und 205 bedeuten könnte:**
2021-11-15 12:26:09.804 [update info] received MS cmd: 011506210402c60040 / 0a46410428ffff0000e2ad
2021-11-15 12:26:09.804 [update notice] received read 21576 unknown.02c60040 QQ=01: 70;65;04;40;-0.1;0.0;226
2021-11-15 12:26:11.054 [update info] received MS cmd: 011506210402c80040 / 0a4841042a9f050000cd02
2021-11-15 12:26:11.054 [update notice] received read 21576 unknown.02c80040 QQ=01: 72;65;04;42;143.9;0.0;205

**Hier weiß ich aber nicht weiter - offensichtlich sind die Werte gar nicht von der CSV interpretierbar**
2021-11-15 12:28:20.539 [bus notice] <011506210400e0004043000a60800d02e803000069015900
2021-11-15 12:28:20.868 [update info] received MM cmd: 1003050a00
2021-11-15 12:28:20.868 [update notice] received unknown MM cmd: 1003050a00
2021-11-15 12:28:20.874 [bus notice] <1003050a00f100
2021-11-15 12:28:21.430 [update info] received MM cmd: 1013050a00
2021-11-15 12:28:21.430 [update notice] received unknown MM cmd: 1013050a00
2021-11-15 12:28:21.435 [bus notice] <1013050a004b00

Winny1904, 2021-11-18:
[ochsner_csv.zip](https://github.com/john30/ebusd/files/7560943/ochsner_csv.zip)

Das ist die CSV die ich verwende.

ub99-muc, 2021-11-27:
Ich habe gerade eine Ochsner Air 11 C11A installiert bekommen mit Unifresh Speicher und mich treibt es nun such um die Daren auszulesen und zu protokolliere. Ein echtes HowTo gibt es wohl eher nicht :-( eBus ist Neuland für mich, komme aus der KNX Ecke mit etwas node-red auf Raspi Add-Ons

john30, 2021-11-28:
[hier](https://github.com/john30/ebusd/wiki/HowTos)

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30