john30/ebusd #442: Config files are not loaded (expect bai)

wgebis, 2021-08-07:
Hello,

Did you face with that issue?

```
$ ebusctl info
version: ebusd 21.2.v21.2
update check: OK
signal: acquired
symbol rate: 28
max symbol rate: 140
min arbitration micros: 366
max arbitration micros: 562
min symbol latency: 5
max symbol latency: 7
reconnects: 0
masters: 4
messages: 216
conditional: 3
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0608;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=47000;SW=0420;HW=1403"
address 26: slave
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Vaillant;ID=V8100;SW=0202;HW=2502"
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=V6100;SW=0418;HW=1902"
```

Files related to devices with addresses 15,35,50 are not loaded, due to this case i'm not able to e.g. read any param:

```
$ ebusctl read -f -c 470 OutsideTemp
ERR: element not found
```

wgebis, 2021-09-05:
Hello, after downloaded newest config files, I see that logs:
```
2021-09-05 16:00:41.944 [bus debug] start request 31
2021-09-05 16:00:41.944 [bus debug] arbitration start with 31
2021-09-05 16:00:42.040 [bus debug] arbitration lost
2021-09-05 16:00:42.040 [bus debug] ERR: arbitration lost during ready, retry
2021-09-05 16:00:42.105 [bus debug] ERR: read timeout during receive command ACK, switching to skip
2021-09-05 16:00:42.106 [bus debug] start request 31
2021-09-05 16:00:42.106 [bus debug] arbitration start with 31
2021-09-05 16:00:42.576 [bus debug] arbitration lost
2021-09-05 16:00:42.578 [bus debug] ERR: arbitration lost during ready, retry
```

wgebis, 2021-09-05:
and also: 
```
2021-09-05 16:02:15.503 [main error] scan config 15: ERR: arbitration lost
```

wgebis, 2021-09-05:
btw: when I switched to version 3.4 I got just after startup:
```
2021-09-05 16:41:32.687 [main debug] <<< done
2021-09-05 16:41:32.687 [network info] [00001] connection closed
2021-09-05 16:41:33.057 [update info] received MS cmd: 1050b505023001 / 0101
2021-09-05 16:41:33.057 [update notice] received unknown MS cmd: 1050b505023001 / 0101
2021-09-05 16:41:33.138 [bus notice] new master 30, master count 3
2021-09-05 16:41:33.189 [update info] received MS cmd: 3026b5040126 / 070500002c006a01
2021-09-05 16:41:33.189 [update notice] received unknown MS cmd: 3026b5040126 / 070500002c006a01
2021-09-05 16:41:33.690 [network debug] dead connection removed - 0
2021-09-05 16:41:35.095 [bus debug] ERR: CRC error during receive response CRC, switching to receive response ACK
2021-09-05 16:41:35.099 [bus debug] ERR: ACK error during receive response ACK, switching to skip
2021-09-05 16:41:37.186 [update info] received MS cmd: 1050b503020001 / 0affffffffffffffffffff
2021-09-05 16:41:37.187 [update notice] received unknown MS cmd: 1050b503020001 / 0affffffffffffffffffff
2021-09-05 16:41:37.427 [update info] received MS cmd: 1050b509040e010001 / 00
2021-09-05 16:41:37.427 [update notice] received unknown MS cmd: 1050b509040e010001 / 00
2021-09-05 16:41:37.668 [update info] received MS cmd: 1050b509040e200000 / 00
2021-09-05 16:41:37.668 [update notice] received unknown MS cmd: 1050b509040e200000 / 00
2021-09-05 16:41:37.687 [main debug] performing regular tasks
2021-09-05 16:41:37.687 [bus info] scan 15 cmd: 3115070400
2021-09-05 16:41:37.734 [bus debug] notify request: ERR: read timeout
2021-09-05 16:41:37.734 [main error] scan config 15: ERR: read timeout
2021-09-05 16:41:37.927 [bus notice] new master 03, master count 4
2021-09-05 16:41:37.927 [update info] received MS cmd: 1008b5110102 / 06033c96468c5c
2021-09-05 16:41:37.927 [update notice] received unknown MS cmd: 1008b5110102 / 06033c96468c5c
2021-09-05 16:41:38.027 [update info] received MS cmd: 3008b5110101 / 095c5a200fff520000ff
2021-09-05 16:41:38.027 [update notice] received unknown MS cmd: 3008b5110101 / 095c5a200fff520000ff
2021-09-05 16:41:38.320 [bus debug] ERR: read timeout during receive command, switching to skip
2021-09-05 16:41:39.125 [update info] received MS cmd: 1050b5040137 / 0200ff
2021-09-05 16:41:39.126 [update notice] received unknown MS cmd: 1050b5040137 / 0200ff
2021-09-05 16:41:39.398 [update info] received MS cmd: 1008b51009000000ffffff010000 / 0101
2021-09-05 16:41:39.399 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2021-09-05 16:41:39.735 [main debug] performing regular tasks
2021-09-05 16:41:39.735 [bus info] scan 26 cmd: 3126070400
2021-09-05 16:41:39.763 [bus debug] notify request: ERR: read timeout
2021-09-05 16:41:39.763 [main error] scan config 26: ERR: read timeout
2021-09-05 16:41:41.139 [update info] received MS cmd: 1050b505021800 / 00
2021-09-05 16:41:41.139 [update notice] received unknown MS cmd: 1050b505021800 / 00
2021-09-05 16:41:41.763 [main debug] performing regular tasks
2021-09-05 16:41:41.764 [bus info] scan 35 cmd: 3135070400
2021-09-05 16:41:41.806 [bus debug] notify request: ERR: read timeout
2021-09-05 16:41:41.806 [main error] scan config 35: ERR: read timeout
2021-09-05 16:41:42.507 [bus debug] ERR: read timeout during receive command, switching to skip
2021-09-05 16:41:42.863 [update info] received MS cmd: 3026b5040126 / 070500002c006a01
2021-09-05 16:41:42.863 [update notice] received unknown MS cmd: 3026b5040126 / 070500002c006a01
2021-09-05 16:41:43.229 [update info] received MS cmd: 1050b505023001 / 0101
2021-09-05 16:41:43.229 [update notice] received unknown MS cmd: 1050b505023001 / 0101
2021-09-05 16:41:43.808 [main debug] performing regular tasks
2021-09-05 16:41:43.808 [bus info] scan 50 cmd: 3150070400
2021-09-05 16:41:43.854 [bus debug] notify request: ERR: read timeout
2021-09-05 16:41:43.854 [main error] scan config 50: ERR: read timeout
2021-09-05 16:41:45.267 [update info] received MS cmd: 1008b5110101 / 095c5a200fff520000ff
2021-09-05 16:41:45.267 [update notice] received unknown MS cmd: 1008b5110101 / 095c5a200fff520000ff
2021-09-05 16:41:45.854 [main debug] performing regular tasks
```

john30, 2022-09-18:
closed due to inactiviy

closed 2022-09-18 by john30