john30/ebusd #446: Prevent/hide invalid message from logs

limad, 2021-09-02:
Hi
is there a way to prevent invalid message in logs ?
`
2021-09-02 23:54:52.914 [update notice] received write bai error03 QQ=10:
2021-09-02 23:54:53.149 [update notice] received write bai error04 QQ=10:
`
i'have defined messages like this: 
`
*wi,,,,,,"B503","00",,,,,,,,,
wi,,error03,error03,,,,03,,,,,,,,,
wi,,error04,error04,,,,04,,,,,,,,,
`
grabe detail:
`
1008b503020003 / 00 = 88: bai error03
1008b503020004 / 00 = 73: bai error04
`
Thanks

john30, 2021-10-03:
what do you mean with invalid messages? there are some options to adjust the logging, [see here](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#logging)

limad, 2021-10-09:
Hi
As there si no data "00"
dont know what thé message mean .

john30, 2021-10-11:
there is some data in the master part (the "03") and the response could just mean that e.g. there is currently no error. it could also mean a reset of error messages or something (see [service.inc](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/service.inc))

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30