john30/ebusd #449: Update the documentation

moskovskiy82, 2021-09-07:
Feeling really frustrated getting this to run. Installed bia the repos on ubuntu server 20.04
Took me around 30 minutes to understand which command to run first 
ebusd -d 172.16.xx.xx:9999 --scanconfig

Then following the instructions checkconfig?
Killed the daemon 
and then 
ebusd -d 172.16.xx.xx:9999 --checkconfig
just to get the following
[main error] invalid configpath without scanconfig

Where to dig next docs just dont say. The configuration path /ets/ebusd is empty

Commifreak, 2021-09-22:
I was wondering as well were to start.

/etc/ebusd is empty, yes, because per default, the configs are loaded from `http://ebusd.eu/config/`.

You could do a `ebuscl i` to view detected system so far. To interact with ebusd one have to send commands to HTTP-Port 8888 or read something via ebusctl.

You could also set `--httpport=PORT` to view a JSON output.

Iam still testing, since I got another problem preventing me from going further.

john30, 2021-10-03:
then please make a proposal by making a pull request.
A lot of info is in the [wiki](https://github.com/john30/ebusd/wiki).

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30