john30/ebusd #450: EBus doesn't bind to the port upon restart of the daemon

moskovskiy82, 2021-09-08:
Ubuntu Server 20.04
Followed the repo installation method.
UFW and apparmor are disabled.

 `$ sudo lsof -i -P -n | grep LISTEN
systemd-r  513 systemd-resolve   13u  IPv4  23824      0t0  TCP 127.0.0.53:53 (LISTEN)
ebusd      773            root    5u  IPv4  27472      0t0  TCP *:8888 (LISTEN)
sshd       774            root    3u  IPv4  29172      0t0  TCP *:22 (LISTEN)
sshd       774            root    4u  IPv6  29174      0t0  TCP *:22 (LISTEN)
docker-pr  999            root    4u  IPv4  30960      0t0  TCP *:4357 (LISTEN)

$ sudo systemctl restart ebusd

$ sudo lsof -i -P -n | grep 8888
$

$ sudo systemctl status ebusd
● ebusd.service - LSB: controls ebusd, the daemon for communication with eBUS heating systems.
     Loaded: loaded (/etc/init.d/ebusd; generated)
     Active: active (running) since Wed 2021-09-08 09:27:59 UTC; 1min 6s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 3564 ExecStart=/etc/init.d/ebusd start (code=exited, status=0/SUCCESS)
      Tasks: 3 (limit: 4079)
     Memory: 2.6M
     CGroup: /system.slice/ebusd.service
             └─3571 /usr/bin/ebusd --pidfile /var/run/ebusd.pid -d enh:172.16.xx.xx:9999 -l /var/log/ebusd.log --scanconfig

Sep 08 09:27:59  systemd[1]: Starting LSB: controls ebusd, the daemon for communication with eBUS heating systems....
Sep 08 09:27:59  ebusd[3564]:  * Starting ebusd ebusd
Sep 08 09:27:59  ebusd[3564]:    ...done.
Sep 08 09:27:59  systemd[1]: Started LSB: controls ebusd, the daemon for communication with eBUS heating systems..
`

ebus log

`2021-09-08 09:30:06.547 [bus notice] scan 08: ;21;20;38;0010023648;3100;013480;N1
2021-09-08 09:32:08.674 [main notice] update check: OK
2021-09-08 09:34:26.191 [main notice] SIGTERM received
2021-09-08 09:34:27.075 [bus error] signal lost
2021-09-08 09:34:29.038 [main notice] ebusd stopped
2021-09-08 09:35:49.926 [main notice] ebusd 21.2.v21.2 started with auto scan on enhanced device 172.16.2.26:9999
2021-09-08 09:35:50.468 [bus notice] bus started with own address 31/36
2021-09-08 09:35:50.490 [bus notice] device status: reset
2021-09-08 09:35:50.504 [bus notice] signal acquired
2021-09-08 09:36:38.261 [bus notice] new master 00, master count 2
`

john30, 2021-10-03:
I need extracts of the system log in order to find out what's going on

john30, 2022-03-27:
closed due to inactivity

closed 2022-03-27 by john30