john30/ebusd #452: New install: no automatic config reading / ERR: arbitration lost

Commifreak, 2021-09-22:
Hi! :)

```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

```
# dpkg -l | grep ebus
ii  ebusd                                21.2                                    armhf        eBUS daemon.
```

<details>
<summary>/etc/default/ebusd</summary>

```
# /etc/default/ebusd:
# config file for ebusd service.

# Options to pass to ebusd (run "ebusd -?" for more info):
EBUSD_OPTS="--scanconfig --httpport=8080"

# MULTIPLE EBUSD INSTANCES WITH SYSV
# In order to run multiple ebusd instances on a SysV enabled system, simply
# define several EBUSD_OPTS with a unique suffix for each. Recommended is to
# use a number as suffix for all EBUSD_OPTS settings. That number will then be
# taken as additional "instance" parameter to the init.d script in order to
# start/stop an individual ebusd instance instead of all instances.
# Example: (uncomment the EBUSD_OPTS above)
#EBUSD_OPTS1="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -p 8888 -l /var/log/ebusd1.log"
#EBUSD_OPTS2="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A900acTF-if00-port0 -p 8889 -l /var/log/ebusd2.log"
#EBUSD_OPTS3="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A900beCG-if00-port0 -p 8890 -l /var/log/ebusd3.log"

# MULTIPLE EBUSD INSTANCES WITH SYSTEMD
# In order to run muiltiple ebusd instances on a systemd enabled system, just
# copy the /usr/lib/systemd/system/ebusd.service file to /etc/systemd/system/
# with a different name (e.g. ebusd-2.service), remove the line starting with
# 'EnvironmentFile=', and replace the '$EBUSD_OPTS' with the options for that
# particular ebusd instance.
```
</details>

<details>
<summary>ebusd.log</summary>
```
2021-09-22 18:23:26.334 [main notice] ebusd 21.2.v21.2 started with auto scan on device /dev/ttyUSB0
2021-09-22 18:23:26.573 [bus notice] bus started with own address 31/36
2021-09-22 18:23:26.583 [bus notice] signal acquired
2021-09-22 18:23:26.777 [bus notice] new master 10, master count 2
2021-09-22 18:23:26.806 [bus notice] new master 03, master count 3
2021-09-22 18:23:26.806 [update notice] received unknown MS cmd: 1008b51009000063ffffff000000 / 0101
2021-09-22 18:23:27.065 [update notice] received unknown MS cmd: 1052b523080114ff0000ffff00 / 0101
2021-09-22 18:23:27.317 [update notice] received unknown MS cmd: 1052b5230402010139 / 02010a
2021-09-22 18:23:36.325 [update notice] received unknown MS cmd: 1008b5110101 / 09464570134d6d0000ff
2021-09-22 18:23:36.619 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080b601b101c87d00
2021-09-22 18:23:36.890 [update notice] received unknown MS cmd: 1008b51009000063ffffff000000 / 0101
2021-09-22 18:23:37.150 [update notice] received unknown MS cmd: 1052b523080114ff0000ffff00 / 0101
2021-09-22 18:23:37.397 [update notice] received unknown MS cmd: 1052b5230402010139 / 02010a
2021-09-22 18:23:46.283 [main error] scan config 08: ERR: arbitration lost
2021-09-22 18:23:46.370 [update notice] received unknown MS cmd: 1008b5110101 / 09464570134d6d0000ff
2021-09-22 18:23:46.663 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080b601b101c87d00
2021-09-22 18:23:46.930 [update notice] received unknown MS cmd: 1008b51009000063ffffff000000 / 0101
2021-09-22 18:23:47.187 [update notice] received unknown MS cmd: 1008b5110102 / 06033c96468270
2021-09-22 18:23:47.446 [update notice] received unknown MS cmd: 1052b523080114ff0000ffff00 / 0101
2021-09-22 18:23:47.694 [update notice] received unknown MS cmd: 1052b5230402010139 / 02010a
2021-09-22 18:23:56.449 [update notice] received unknown MS cmd: 1008b5110101 / 09464570134d6d0000ff
2021-09-22 18:23:56.745 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080b501b101c87d00
2021-09-22 18:23:57.015 [update notice] received unknown MS cmd: 1008b51009000063ffffff000000 / 0101
2021-09-22 18:23:57.196 [main error] scan config 15: ERR: arbitration lost
2021-09-22 18:23:57.275 [update notice] received unknown MS cmd: 1052b523080114ff0000ffff00 / 0101
2021-09-22 18:23:57.521 [update notice] received unknown MS cmd: 1052b5230402010139 / 02010a
2021-09-22 18:24:06.490 [update notice] received unknown MS cmd: 1008b5110101 / 09464570134d6d0000ff
2021-09-22 18:24:06.785 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080b501b101c87d00
2021-09-22 18:24:07.055 [update notice] received unknown MS cmd: 1008b51009000063ffffff000000 / 0101
2021-09-22 18:24:07.237 [main error] scan config 52: ERR: arbitration lost
2021-09-22 18:24:07.314 [update notice] received unknown MS cmd: 1052b523080114ff0000ffff00 / 0101
2021-09-22 18:24:07.562 [update notice] received unknown MS cmd: 1052b5230402010139 / 02010a
2021-09-22 18:24:16.570 [update notice] received unknown MS cmd: 1008b5110101 / 09464570134d6d0000ff
2021-09-22 18:24:16.866 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080b401b001c87d00
2021-09-22 18:24:17.136 [update notice] received unknown MS cmd: 1008b51009000063ffffff000000 / 0101
2021-09-22 18:24:17.393 [update notice] received unknown MS cmd: 1008b5110102 / 06033c96468270
2021-09-22 18:24:17.666 [update notice] received unknown MS cmd: 1008b5040100 / 0a03182418220903217013
2021-09-22 18:24:17.903 [update notice] received unknown BC cmd: 10feb516080017241822090321
2021-09-22 18:24:18.162 [update notice] received unknown MS cmd: 1052b523080114ff0000ffff00 / 0101
2021-09-22 18:24:18.425 [update notice] received unknown MS cmd: 1008b5110100 / 083402130008000083
2021-09-22 18:24:18.673 [update notice] received unknown MS cmd: 1052b5230402010139 / 02010a
2021-09-22 18:24:18.887 [update notice] received unknown BC cmd: 10feb51603017013
2021-09-22 18:24:26.642 [update notice] received unknown MS cmd: 1008b5110101 / 09464570134d6d0000ff
2021-09-22 18:24:26.936 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080b301b001c87d00
2021-09-22 18:24:27.207 [update notice] received unknown MS cmd: 1008b51009000063ffffff000000 / 0101
2021-09-22 18:24:27.389 [main error] scan config 08: ERR: arbitration lost
2021-09-22 18:24:27.466 [update notice] received unknown MS cmd: 1052b523080114ff0000ffff00 / 0101
2021-09-22 18:24:27.715 [update notice] received unknown MS cmd: 1052b5230402010139 / 02010a
2021-09-22 18:24:36.683 [update notice] received unknown MS cmd: 1008b5110101 / 09464570134d6d0000ff
2021-09-22 18:24:36.979 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080b301b001c87d00
2021-09-22 18:24:37.186 [main notice] SIGTERM received
2021-09-22 18:24:37.248 [update notice] received unknown MS cmd: 1008b51009000063ffffff000000 / 0101
2021-09-22 18:24:37.430 [main error] scan config 15: ERR: arbitration lost
2021-09-22 18:24:37.508 [update notice] received unknown MS cmd: 1052b523080114ff0000ffff00 / 0101
2021-09-22 18:24:37.557 [main notice] ebusd stopped
```
</details>

<details>
<summary>ebusctl i</summary>

```
# ebusctl i
version: ebusd 21.2.v21.2
signal: acquired
symbol rate: 81
max symbol rate: 85
reconnects: 0
masters: 3
messages: 13
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave
```
</details>

I wonder what I missed - the ebusctl i shows nothing detected (Its a Vaillant ecoCOMPACT system with VRC700 / VR71). The ebus coupler is the USB ESERA.

Any hints? :/

john30, 2021-10-03:
from your log it looks as if ebusd can't send at all on the bus. consequently, the ID query for each device fails which is necessary to identify the right CSV files to load.
you should [check if the poti on the couple is well aligned](https://github.com/john30/ebusd/wiki/6.-Hardware#adjusting-the-potentiometer). or maybe the coupler has a generic write problem.

Commifreak, 2021-10-03:
Thanks, let me check that. I'll post back then.

john30, 2022-09-18:
closed due to inactivity

closed 2022-09-18 by john30

arndtbraun, 2022-12-14:
Hi, i have the same issue.

Could it be a Problen that i have not installed a VRC controller (Heatingsystem only.
Master 11 is found but no slaves.
ebusctl i output:
`version: ebusd 22.4.v22.4-28-gbc3e6110
device: /dev/ttyUSB0
signal: acquired
symbol rate: 22
max symbol rate: 23
reconnects: 0
masters: 1
messages: 0
conditional: 0
poll: 0
update: 0
address 31: master #8, ebusd
address 36: slave #8, ebusd`

ebusd.log
`2022-12-14 09:08:12.007 [main notice] ebusd 22.4.v22.4-28-gbc3e6110 started with auto scan on device /dev/ttyUSB0
2022-12-14 09:08:12.023 [bus notice] bus started with own address 31/36
2022-12-14 09:08:12.053 [mqtt notice] connection established
2022-12-14 09:08:12.065 [bus notice] signal acquired
2022-12-14 09:10:19.224 [main notice] update check: revision v22.4 available
2022-12-14 09:12:02.559 [bus notice] new master 03, master count 2
2022-12-14 09:12:03.063 [bus notice] <0364b51202020066
2022-12-14 09:12:04.234 [bus notice] >31
2022-12-14 09:12:04.325 [bus notice] >31
2022-12-14 09:12:04.414 [bus notice] >31
2022-12-14 09:12:04.505 [bus notice] >31
2022-12-14 09:12:04.506 [main error] scan config 08: ERR: arbitration lost
2022-12-14 09:12:08.380 [bus notice] <0364b51202020066
2022-12-14 09:12:13.708 [bus notice] <0364b51202020066
2022-12-14 09:12:16.636 [bus notice] >31
2022-12-14 09:12:16.727 [bus notice] >31
2022-12-14 09:12:16.817 [bus notice] >31
2022-12-14 09:12:16.909 [bus notice] >31
2022-12-14 09:12:16.909 [main error] scan config 08: ERR: arbitration lost
2022-12-14 09:12:19.027 [bus notice] <0364b51202020066
2022-12-14 09:12:24.705 [bus notice] >31
2022-12-14 09:12:24.794 [bus notice] >31
2022-12-14 09:12:24.885 [bus notice] >31
2022-12-14 09:12:24.974 [bus notice] >31
2022-12-14 09:12:24.974 [main error] scan config 08: ERR: arbitration lost`