john30/ebusd #453: Write values through ebusd-http?

tinbert, 2021-09-22:
Hi John,
is it possible to write values by using the http-client? It should be possible, should'nt it?

With 
`http://10.0.1.48:8889/data/470/Hc1SFMode?write`
I got:
`{
"470":{
"messages":{
"Hc1SFMode":{
"name":"Hc1SFMode",
"passive":false,
"write":false,
"lastup":1632332853,
"zz":21,
"fields":{
"0":{
"name":"",
"value":"none",
}
}
},
"Hc1SFMode-w":{
"name":"Hc1SFMode",
"passive":false,
"write":true,
"lastup":0,
}
}
},...`

so I thought I could use "Hc1SFMode-w" to achieve this? But what's the syntax?

john30, 2021-10-03:
no. HTTP currently does not support issuing write messages. It only retrieves what was cached from the last write via the TCP client or from another device.

john30, 2022-11-13:
closed due to inactivity

closed 2022-11-13 by john30